cncjs/cncjs #88: Spindle Speed in GUI

AustinSaintAubin, 2016-10-22:
Spindle Speed is currently [set to zero by default](https://github.com/cheton/cnc/blob/master/src/web/widgets/Spindle/index.jsx) with no way to adjust it in the gui.

cheton, 2016-10-22:
In recent Grbl version (see https://github.com/grbl/grbl/issues/949#issuecomment-204726607), it made a change to require an 'S' spindle speed value that is non-zero to activate the spindle. 

How about adding a slider to set the spindle speed? (0-1000 with default build)

AustinSaintAubin, 2016-10-23:
I slider would be good. Might need to be able to set the max value in case of custom firmware (>1000).

As a side note: I was able to somewhat get around this issue by setting "$N1=S1250" so that the spindle speed is set on grbl startup.

cheton, 2016-11-03:
Updated in 1.7.0 release. Please let me know if this works for you.

closed 2016-11-03 by cheton