cncjs/cncjs #89: grbl 1.1 support

meikrantz, 2016-10-22:
grbl 1.1 (gnea/grbl) has several new additions that could be enabled in cnc.  Most grbl guis will require some rewriting to support.  Is this on the roadmap?

cheton, 2016-10-22:
Thank you for informing me. I will take a look shortly. 

https://github.com/grbl/grbl/issues/1107
https://github.com/gnea/grbl/

cheton, 2016-10-27:
@meikrantz
Grbl v1.1 support is now available in the master branch. I will publish v1.7 in the next few days if no issues found.

jarretluft, 2016-11-01:
Hey @cheton, It's been a while since I have worked with my machine, but just recently got back into some projects and have been checking out all the progress you've made since I last used your software.  CNC has progressed nicely, thanks for all the work you've done!

I've done some testing on your GRBL v1.1 changes:
- for the most part it looks like your changes to handle the new response message structure have worked, though I do not get the proper feedrates or spindle speeds showing in the GRBL widget, both continue to show 0 even after I execute gcode commands containing F and S variables
- The machine position no longer shows any value, as this has been removed from the GRBL response.  Not sure if you plan to keep this column in the axes widget for legacy GRBL users (whether anyone got much value out of it in the first place, i'm not sure) or maybe there's a way to toggle it on/off?
- Do you have plans to add buttons for realtime feedrate and spindle speed adjustments?  
The list of extended ascii realtime commands is [here](https://github.com/gnea/grbl/blob/edge/doc/markdown/realtime_cmds.md#extended-ascii-realtime-command-descriptions).  
Feed rate override buttons could probably go in the grbl widget (as that's where feedrate is reported), and the spindle speed overrides could go in the spindle widget.

Thanks again and keep up the good work!
-J.

cheton, 2016-11-01:
Hi @jarretluft,

Thank you for giving me your feedback. Please find my comments below:

> for the most part it looks like your changes to handle the new response message structure have worked, though I do not get the proper feedrates or spindle speeds showing in the GRBL widget, both continue to show 0 even after I execute gcode commands containing F and S variables. I notice the ? responses have changed to include both in a single FS:0,0 format, not sure if this is why it's not parsing properly...

The feedrate and spindle speed shown in the Grbl widget are parsed from the parser state command <b>$G</b>, which is the last state of Grbl parser, but it may not reflect real-time feedrate and spindle speed:
```
> $G
[GC:G1 G54 G17 G21 G90 G94 M0 M4 M9 T0 F250 S600]
```
[image]

To make it look more reasonable, I would consider changing the title from <b>Parser State</b> to <b>Status Reports</b>, and display realtime feedrate and spindle speed when <b>F</b> or <b>FS</b> variables are available from the <b>?</b> response.
It will fallback to parser state ($G) when using Grbl v0.9.

> Correction: It looks like depending on whether the $10 variable is 1 or 2, it could send machine or work position, and with either of those a WCO: value is sometimes returned which indicates what the current offset is. So really, using whichever (machine or work) position is being returned, as well as storing an WCO response you could still derive the missing position by either adding or subtracting the offset from the position value returned.

From my local testing, it should be able to work whether the $10 variable is either 1 or 2. Since the v1.7.0 release is still not published, may I know which branch or commits you've done for the test?
You can type `http://{ip-addr}:8000/api/controllers` in the browser, if you built the app from the latest changes, it should report `mpos`, `wpos`, and `wco` in the status field, like below:

[image]

> Do you have plans to add buttons for realtime feedrate and spindle speed adjustments?
> The list of extended ascii realtime commands is here.
> Feed rate override buttons could probably go in the grbl [...]

Yes, I do have plans to add override buttons to control realtime feedrate and spindle speed, it will probably be added on version later than v1.7.0, maybe in v1.7.x.

cheton, 2016-11-03:
The 1.7.0 release is available to download, it contains several improvements including Grbl 1.1 support. Please let me know if you have any problems.

closed 2016-11-03 by cheton

cheton, 2016-11-03:
Remaining TODO items:
* Real-time overrides
* New jogging command
* Extended ASCII commands

AustinSaintAubin, 2016-11-26:
Was Real-time overrides no addressed in [v1.7.2](https://github.com/cheton/cnc/releases/tag/v1.7.2)?

Is GRBL 1.1 safe to use with CNC.js?

cheton, 2016-11-27:
The support for real-time overrides was added in v1.7.2. This feature will be visible when Grbl 1.1 is detected. You can check if your API response `http://localhost:8000/api/controllers` returns the following result:
```js
{
    "controller": {
        "state": {
            "status": {
                "ov": [ // overrides
                    100,
                    100,
                    100
                ]
            }
        }
    }
}
```

I've been testing Grbl 1.1 with cnc v1.7 for a month and didn't see any communication problems. It should be safe to use.