cncjs/cncjs #90: Lessen Visibility of Completed Paths

AustinSaintAubin, 2016-10-26:
It is difficult to visually make out the completed paths (RED) from the preloaded paths (RED) and to an extent the to be completed paths (BLUE).

I propose that once the path has been completed, its color is changed to GREY with a High Transparency. This would make it easy to see the preloaded paths (RED) and the to be completed paths (BLUE).

Picture bellow shows the issues... lot of RED, not sure which cut is coming next.
[image]

cheton, 2016-10-26:
Thanks you for your suggestions. I will put the enhancements in v1.7.

cheton, 2016-11-03:
Resolved in 1.7.0

closed 2016-11-03 by cheton