cncjs/cncjs #94: Smoothieware support

tritao, 2016-11-23:
Hey, first of all thanks for your work on this nice piece of software.

Do you have any plans on adding support for Smoothieware to the software?

cheton, 2016-11-23:
Smoothieware is similar to Grbl, I think that is not difficult to add Smoothieware support to the cnc software. I will get a 4XC SmoothieBoard for protocol testing some other time.

Ref.
https://github.com/vlachoudis/bCNC/issues/143
https://github.com/winder/Universal-G-Code-Sender/issues/204

tritao, 2016-11-23:
I wonder if @arthurwolf could help out here with a board to help you test.

If not, given your location, you could check out the MKS SBase, which is compatible with Smoothieware.

arthurwolf, 2016-11-23:
Hey !

Smoothie is very similar to grbl if you use the CNC build, and  add in config : 
grbl_mode true

I can give you a discount, or send you a broken board ( that will still be able to communicate ) if you want.

Cheers :)

cheton, 2016-11-24:
Thank you for your information. I will make a purchase for the SmoothieBoard (or MKS SBase v1.3 from China) and then start to work out.

tritao, 2016-12-01:
Just tried to connect to Smoothieware as suggested by @arthurwolf, with the firmware compiled with CNC=1 which puts it in Grbl-compatible mode.

Doesn't seem to work too well with cnc though.

```
2016-12-01T17:30:13.838Z - debug: [cncserver] New connection from 192.168.0.174: id=1p4flM5-1LXzY-kLAAAA, token.id=bce85834-fd3e-46ad-97e4-1be600159a83, token.name=joao     at Namespace.<anonymous> (/Users/joao/Dev/cnc/dist/cnc/app/index.js:3406:28)
Started the server at http://192.168.0.174:8000/.
2016-12-01T17:30:15.275Z - debug: [cncserver] socket.open("/dev/cu.usbmodem1421", {"controllerType":"Grbl","baudrate":115200}): id=1p4flM5-1LXzY-kLAAAA     at Socket.<anonymous> (/Users/joao/Dev/cnc/dist/cnc/app/index.js:3457:32)
2016-12-01T17:30:15.288Z - debug: [Grbl] Connected to serial port "/dev/cu.usbmodem1421"     at SerialPort.<anonymous> (/Users/joao/Dev/cnc/dist/cnc/app/index.js:5175:28)
2016-12-01T17:30:15.289Z - silly: [Grbl] > 
2016-12-01T17:30:15.292Z - silly: [Grbl] < ALARM:Abort during cycle
i4aR5O9GGE_RjwSzUujnywqa9VBtNNQd GET /api/controllers 200 8.366 ms
2016-12-01T17:30:17.065Z - debug: [cncserver] socket.command("/dev/cu.usbmodem1421", "homing"): id=1p4flM5-1LXzY-kLAAAA, args=[]     at Socket.<anonymous> (/Users/joao/Dev/cnc/dist/cnc/app/index.js:3508:32)
2016-12-01T17:30:17.066Z - silly: [Grbl] > $H

2016-12-01T17:30:18.072Z - silly: [Grbl] < ok
2016-12-01T17:30:18.701Z - debug: [cncserver] socket.command("/dev/cu.usbmodem1421", "unlock"): id=1p4flM5-1LXzY-kLAAAA, args=[]     at Socket.<anonymous> (/Users/joao/Dev/cnc/dist/cnc/app/index.js:3508:32)
2016-12-01T17:30:18.702Z - silly: [Grbl] > $X
```

One issue is that the Axes widget never unlocks itself even after the homing command, which homes the machine correctly. But the unlock button, which sends `$X`, never gets any reply from the board.

cheton, 2016-12-02:
Thank you for the information. Just received my SmoothieBoard, I will give it a try in the next few days.

tritao, 2016-12-02:
Thanks for the help.

After some more testing, the board does send the response to `$X`, it's some problem processing the responses.

cheton, 2016-12-02:
The command execution will be blocked by a readiness check until it received a Grbl startup line
`Grbl 1.1d ['$' for help]`:

https://github.com/cheton/cnc/blob/v1.8.5/src/app/controllers/Grbl/GrblController.js#L233-L252
```js
this.grbl.on('startup', (res) => {
    this.emitAll('serialport:read', res.raw);

    if (!this.ready) {
        // View Grbl settings
        this.feeder.feed({ line: '$$' });

        // View startup blocks
        this.feeder.feed({ line: '$N' });

        if (!this.feeder.isPending()) {
            this.feeder.next();
        }
    }

    this.ready = true;
    this.queryResponse.status = false;
    this.queryResponse.parserstate = false;
    this.queryResponse.parserstateEnd = false;
});
```

I can add a piece of code to detect Smoothie that can work with current Grbl controller.

The full support for Smoothieware, including a Smoothie widget, might be available in the 1.9.0 release.

cheton, 2016-12-02:
Added preliminary support for Smoothieware in the [1.8.6](https://github.com/cheton/cnc/releases/tag/v1.8.6) release. Stability is not guaranteed, but it's worth a try!

tritao, 2016-12-02:
Awesome, will do some testing and let you know.

cheton, 2016-12-05:
Hi @arthurwolf,

I found an issue while trying to stop streaming a G-code program to Smoothie, the status report (?) may report unusual work position with uncertain offsets.

The steps to reproduce:
Step 1. Streaming a G-code program to Smoothie
Step 2. Send `!` and `ctrl-x` to feed hold and reset
Step 3. Send $X to unlock an Alarm state
Step 4. Run step 1 again

After taking several tries, I found that Smoothieware lost its work position in status report, but this never happen in Grbl.

May I know if I made something wrong while trying to recover an alarm state?

arthurwolf, 2016-12-05:
Hey.

As far as I know, once Smoothie goes into Alarm state, it's not expected to
do anything right, and should be fully reset before being used again
@wolfmanjm might have more details.

Cheers.

wolfmanjm, 2016-12-05:
$X or M999 will reset from Alarm mode, but if positioning is lost then you also need to home after $X.. with $H. It is impossible to guarantee that position is not lost after an abrupt abort. Even Grbl does not guarantee that.

cheton, 2016-12-08:
I used to pause and resume a G-code program immediately using `!` and `~` commands. What's your recommended way to abort from a hold state?

Here are my steps to abort a G-code program against a hold:
1. Clear out the queue
2. Send `~` to resume
3. Send ctrl-x to abort
4. Unlock Alarm ($X)
5. Retract Z to a clearance height
6. Home ($H)

tritao, 2016-12-09:
Been testing the new Smoothie support with a laser cutter today and I can report that it works.

There seemed to be some problems pausing the job, sometimes it worked but most of the time the job just continued until I pressed the stop button. I will do some more checking tomorrow.

cheton, 2016-12-10:
Hi @tritao,

I'm still fine tuning the workflow for pausing and resuming a job, I will publish a new release once I confirmed a correct usage.

Regarding your question, I got some responses here:  https://github.com/LaserWeb/LaserWeb3/commit/ab3ffa6c26ab968eeef98734a4f43bcc331302fd#commitcomment-20137765

There is a difference between Grbl and Smoothie, both <b>!</b> and <b>M600</b> cannot pause a job immediately in Smoothie, the pause (!) will take effect until the planner queue is finished.

wolfmanjm, 2016-12-11:
FYI I am removing the ! and ~ from smoothie, as it doe snot really do anything. You can continue to send them but they will be ignored. The control-X will continue to work as immediate abort.
If you want to pause you should simply stop sending gcode to smoothie, and then continue to send when you resume. this is basically what happens now in smoothie anyway and I just realized that the ! was not doing anything effective.
So there is no immediate pause/resume in smoothie, and given how hard it is to implement I suspect there will not be any time soon.
M600/M601 are there for 3D printers to replace filament and should not be used by CNC hosts as it requires the host to honor the reprap style feedback commands like ` // command:pause`

cheton, 2016-12-11:
@wolfmanjm Thank you very much for your detailed explanation. I will list it as a known issue in FAQ,  and remove `!` and `~` from <kbd>Pause</kbd> and <kbd>Resume</kbd> buttons as well.

cheton, 2016-12-11:
Note. Both `!` and `~` are removed in https://github.com/Smoothieware/Smoothieware/commit/716c7597d0993a6f0a270d5dfd6477477c1264a3

wolfmanjm, 2016-12-11:
Yes just did that, so you can continue to send ! so when/if it is ever implemented it will just work :) right now it'll just be ignored.

cheton, 2016-12-11:
Sure! Thank you for your guidance. I just upgraded the firmware to the latest build version <b>edge-fecb4cc</b>, I will do more tests with this version.

cheton, 2017-02-13:
Added in v1.9.0-alpha.0

closed 2017-02-13 by cheton

cheton, 2017-02-13:
@tritao, @arthurwolf, @wolfmanjm 

Thank you so much for making it happen.

arthurwolf, 2017-02-13:
Thank you so much for all the work to get smoothie supported ! I asked the smoothie community to test this so we can find any problem that potentially exists.