cncjs/cncjs #96: Firefox css issue on settings page (screenshot inside)

slimline33, 2016-11-26:
Hi, there is an css error on firefox on the settings page...

[image]

It is not importent but I wanted to share it :)

slimline33, 2016-11-26:
Found another view issue on the probe widget:

[image]

cheton, 2016-11-27:
I just resolved the css problem in the <b>Settings</b> page, but I cannot reproduce the issue on the probe widget. May I know which cnc and firefox versions you're running on what platform? Thanks.

slimline33, 2016-11-27:
mhmm... after clearing browser cache the widget is o.k

[image]

thanks!

Firefox 50.0 on Windows7 64bit

closed 2016-11-27 by slimline33