cncjs/cncjs #97: Update the raspberry kernel to beta in installation guide

slimline33, 2016-11-27:
Hi!

I think it is not good to tell the user to update the raspberry kernel with sudo rpi-update.
Thats always nightly kernels and there could be the one or the other bug in.

Here: https://github.com/cheton/cnc/wiki/Raspberry-Pi-Setup-Guide#maintaining-your-software-stack-w-updates--upgrades

Just use apt-get update / upgrade. That will also update the kernel to a stable version!

Source: https://www.elektronik-kompendium.de/sites/raspberry-pi/2006061.htm (german only)

Thanks

Christoph

cheton, 2016-11-27:
Loop @AustinSaintAubin

AustinSaintAubin, 2016-11-27:
Updated wiki documentation regarding Raspberry Pi setup and updates.
github.com/cheton/cnc/wiki/Raspberry-Pi-Setup-Guide

cheton, 2016-11-27:
Great! Thanks.

closed 2016-11-27 by cheton

slimline33, 2016-11-27:
Nice work! Thanks!