cncjs/cncjs #99: Packaged interface

arthurwolf, 2016-12-02:
Hello.

What do you use NodeJS for ?
Smoothieboard has an ethernet interface with a http interface for sending Gcode.
If this project added support for this, would it be possible to simply package this application as a html page, without needing nodejs ?

Cheers.

cheton, 2016-12-03:
Hi @arthurwolf,

The Node.js server backend plays an important role in this software, it is used to deal with serial communication with CNC milling controllers, and also provides AJAX/WebSocket API calls to define a abstract communication layer between frontend and backend. Furthermore, access control, server configuration settings, and user account management are also included.

I haven't tried the ethernet interface for sending G-code via HTTP, but I would put the communication to backend rather than frontend. Running a server instance can ensure proper stability when sending G-code, it won't be affected by CPU loads in a desktop environment, and it can continue currently running job even accidentally closing user's browser. A similar issue has been discussed in https://github.com/LaserWeb/lw.comms/issues/1. LW4 are moving forward to the same thing.

To make installation easier for desktop users, especially for those who don't have experiences with Node.js, the prebuilt Mac / Windows / Linux apps generated by CI are provided with each release (download links: https://github.com/cheton/cnc/releases). It behaves like a native app, so users don't need to worry about Node.js installation.

arthurwolf, 2016-12-03:
That's ok.

The idea here was essentially turning the app into a single html page that you can run on an android tablet. That way you can just glue the tablet to your machine and use that to control it.
It would be nice if it were possible, but if it's not I understand why.

Cheers.

cheton, 2016-12-03:
Thank you for your explanation. It make sense to me to host a web interface on Smoothieboard without extra installation. I just took a look at the Smoothie implementation (e.g. [public/js/smoothie.js](https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/smoothie.js
) and [public/js/socketcomms.js](https://github.com/LaserWeb/LaserWeb3/blob/master/public/js/socketcomms.js)) on LaserWeb3, it is clear to me now.

In the beginning, the support for ethernet communication will be added on the server side that treats Smoothie as another backend service. Meanwhile, I will study how to make a simplified web interface  (w/o Node.js dependency) as an add-on to the Smoothieboard, that should bring conveniences for Smoothieboard users.

arthurwolf, 2016-12-03:
That is really awesome news.
Once you have looked into it, if you can measure the amount of work, and
give me your hourly price, I can also look into finding funding to pay you
for this work ( the direct ethernet link without going through the backend )

cheton, 2017-02-28:
Hi @arthurwolf,

May I confirm whether the query status command "?" is supported in telnet mode?

The Grbl-like status command `?` is available when connecting from USB port, but I got no response when typing `?` in telnet mode. I can run `get status` command to query current status instead of `?`, but it always returns an extra empty line which seems to be unnecessary.

```sh
$ telnet 192.168.1.250
Trying 192.168.1.250...
Connected to 192.168.1.250.
Escape character is '^]'.
Smoothie command shell
> get status
<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.0000>

> G0X10Y10
ok
> get status
<Idle,MPos:10.0000,10.0000,0.0000,WPos:10.0000,10.0000,0.0000>

>
```

I just traced the code in SimpleShell.cpp, the return string of `THEKERNEL->get_query_string().c_str()` already contains "\r\n" at the end, so it's not necessary to send another "\n" that caused an empty line. Is it right?

https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/utils/simpleshell/SimpleShell.cpp#L847
```cpp
    } else if (what == "status") {
        // also ? on serial and usb
        stream->printf("%s\n", THEKERNEL->get_query_string().c_str());

    } else {
```

Thanks!

arthurwolf, 2017-02-28:
> simpleshell/SimpleShell.cpp#L847

Hey.

Yes, ? is one of the "grbl" commands, and they are supposed to be executed
at a higher priority than every other command, but that has only been
implemented for USB. Unfortunately this means that Ethernet doesn't have
the command at all.
I think for now you w%ant to be using "get status". I'm not sure what the
extra line is about, I'll ask.

cheton, 2017-02-28:
This is what I'd expect:

```
> get status
<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.0000>
>
```

but actually it returns two consecutive "\n" when calling "get status":
```
> get status
<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.0000>

>
```

That's just a minor issue, but would like to get fixed. :)

fishpepper, 2018-08-03:
Is there a cncjs branch somewhere that uses the ethernet port to talk to a smoothieboard?

Where is the best spot to add this feature? Inside the cncjs-controller (https://github.com/cncjs/cncjs-controller) ?
I might code this soon, my smoothieboard just started to randomly loose the serial port connection while cutting :(

nesquik011, 2018-08-03:
I asked and got the same problem they told me this function will be available in new version milestone and i am waiting

arthurwolf, 2018-08-03:
If you need help/info on how to talk to the Smoothieboard, don't hesitate
to ask me. This should be easy to add.

fishpepper, 2018-08-03:
The Smoothieboard part looks quite easy. Sending data via a network connection should be straight forward as well (api very similar to serial port).
I am fighting with finding the right place to add this feature...

There is some serial port access going on in cncjs-controller (e.g. listPorts()).
On the other Hand there is some serial stuff happening inside the individual controllers.

nesquik011, 2018-08-03:
Well i dont know , i did try to edit cncjs many time but when in install it works with default and i cant even add widget someone made so if you would make video on how to do it , i would be so happy or pictures

fishpepper, 2018-08-06:
I hacked together a first version. You can find my fork here: https://github.com/fishpepper/cncjs

See SmoothieController.js to set your IP and Port:
```
this.connection = new SocketSerialConnection({
            host: 'cnc', //'192.168.178.222',
            port: 23, //23,
            writeFilter: (data) => {
                return data;
            }
        });
```

WARNING: DO NOT USE THIS ON A RUNNING MACHINE!

It seems to work, but there are some issues:
1) no selection of ethernet in gui possible, just select smoothie any port and connect
2) no clean connection. you have to click reset and unlock to get things going
3) playing a gcode seems very slow
4) the status is only updated every now and then (? has no prio on ethernet)
5) playing a gcode file locks up every now and then, clicking pause and resume lets it continue

I think I will give up here. In order to get this running properly one would have to modify the smoothie firmware. 
Maybe having two sockets could help, one for commands and one for status and one for hi-level tasks (?!~ commands etc).

I think I will buy a better USB cable and try again... The machine used to be working fine the last 20+ hours. It just started to produce random hangs on USB :(

nesquik011, 2018-08-06:
whatever you do it will still the same even if it started to look fine , because with time the wire get dust and weak blah blah  and the drop too has some reasons with power supply of the computer tons of rezons so to fix that ether and its the best , if it was not the huge paper printers was going to be just usb not ether support , coz they know it makes problems , and try to pass a high volt next the usb it will drop  it just sucks :P

nesquik011, 2018-08-06:
so i will keep waiting for the new cncjs  version  milstone