cncjs/cncjs #100: Strange Behviour - 1.8.6

jarretluft, 2016-12-04:
Hi @cheton,
I'm getting some odd behaviour and errors in the latest version (1.8.6).

I've tried running some pretty standard gcode twice now and had errors and erratic machine movement both times.
I get a mess of errors printing out on the server side, and a non-standard gcode error (grbl Error:20).  Additionally, after the error the machine movement no longer matches what should be happening.  I only get a couple lines into the gcode until things start going wrong and I have to abort.

Any ideas what's going on?  The only change i've made to my system since it was working last night was updating to 1.8.6.  
I'm using grbl version ($i):  [VER:1.1d.20161023:]

**Gcode:**
```
T1M6
G17
G21
G0 Z3.2004
G0 X0.0000 Y0.0000 S12000 M3
G0 X88.9358 Y-3.7040 Z3.1750
G1   Z-2.5000 F508.0
G1  Y-63.7040  F762.0
G0   Z3.1750
G0  Y-3.7040 
```

**Server Side Errors:**
```
2016-12-03T23:39:25.021Z - info: Started the server at http://127.0.1.1:8000
2016-12-03T23:43:39.105Z - error: uncaughtException: Uncaught, unspecified "error" event. ([object Object]) date=Sat Dec 03 2016 16:43:39 GMT-0700 (MST), pid=1621, uid=1000, gid=1000, cwd=/usr/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nodejs, version=v4.6.0, argv=[/usr/bin/nodejs, /usr/bin/cnc], rss=57937920, heapTotal=35438188, heapUsed=31746412, loadavg=[0.02490234375, 0.044921875, 0.00244140625], uptime=64794, trace=[column=17, file=events.js, function=Smoothie.emit, line=144, method=emit, native=false, column=19, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Smoothie.parse, line=6265, method=parse, native=false, column=29, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index.js, function=, line=5177, method=null, native=false, column=13, file=events.js, function=emitOne, line=77, method=null, native=false, column=7, file=events.js, function=SerialPort.emit, line=169, method=emit, native=false, column=17, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=null, line=24, method=null, native=false, column=null, file=null, function=Array.forEach, line=null, method=forEach, native=true, column=13, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=, line=23, method=null, native=false, column=18, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=SerialPort._emitData, line=313, method=_emitData, native=false, column=14, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=293, method=null, native=false, column=7, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=306, method=null, native=false, column=17, file=fs.js, function=FSReqWrap.wrapper [as oncomplete], line=576, method=wrapper [as oncomplete], native=false], stack=[Error: Uncaught, unspecified "error" event. ([object Object]),     at Smoothie.emit (events.js:144:17),     at Smoothie.parse (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:6265:19),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:5177:29),     at emitOne (events.js:77:13),     at SerialPort.emit (events.js:169:7),     at /usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:24:17,     at Array.forEach (native),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:23:13),     at SerialPort._emitData (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:313:18),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:293:14),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:306:7),     at FSReqWrap.wrapper [as oncomplete] (fs.js:576:17)]
2016-12-03T23:43:39.139Z - error: uncaughtException: Uncaught, unspecified "error" event. ([object Object]) date=Sat Dec 03 2016 16:43:39 GMT-0700 (MST), pid=1621, uid=1000, gid=1000, cwd=/usr/lib/node_modules/cncjs/dist/cnc/app, execPath=/usr/bin/nodejs, version=v4.6.0, argv=[/usr/bin/nodejs, /usr/bin/cnc], rss=58200064, heapTotal=35438188, heapUsed=31961980, loadavg=[0.02490234375, 0.044921875, 0.00244140625], uptime=64794, trace=[column=17, file=events.js, function=Smoothie.emit, line=144, method=emit, native=false, column=19, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index.js, function=Smoothie.parse, line=6265, method=parse, native=false, column=29, file=/usr/lib/node_modules/cncjs/dist/cnc/app/index.js, function=, line=5177, method=null, native=false, column=13, file=events.js, function=emitOne, line=77, method=null, native=false, column=7, file=events.js, function=SerialPort.emit, line=169, method=emit, native=false, column=17, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=null, line=24, method=null, native=false, column=null, file=null, function=Array.forEach, line=null, method=forEach, native=true, column=13, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js, function=, line=23, method=null, native=false, column=18, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=SerialPort._emitData, line=313, method=_emitData, native=false, column=14, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=293, method=null, native=false, column=7, file=/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js, function=, line=306, method=null, native=false, column=17, file=fs.js, function=FSReqWrap.wrapper [as oncomplete], line=576, method=wrapper [as oncomplete], native=false], stack=[Error: Uncaught, unspecified "error" event. ([object Object]),     at Smoothie.emit (events.js:144:17),     at Smoothie.parse (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:6265:19),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/dist/cnc/app/index.js:5177:29),     at emitOne (events.js:77:13),     at SerialPort.emit (events.js:169:7),     at /usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:24:17,     at Array.forEach (native),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/parsers.js:23:13),     at SerialPort._emitData (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:313:18),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:293:14),     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/node_modules/serialport/lib/serialport.js:306:7),     at FSReqWrap.wrapper [as oncomplete] (fs.js:576:17)]
```

**CNC Console Error:**
```
> (9) G0 Z3.175
error:20
```

cheton, 2016-12-04:
I didn't see any errors in 1.8.5, but it happened in 1.8.6. It might be related to the code of adding Smoothie support. You can rollback your version to 1.8.5 (`npm install cncjs@1.8.5`), I will take a look into this issue soon.

cheton, 2016-12-04:
The root cause is the first line, I guess your Grbl 1.1d didn't have M6 enabled, and it reported an error like this:
```
> T1M6
error:20
```

The Smoothie parser in the 1.8.6 release didn't handle errors properly, so it blocked the execution. I will fix it by today.

cheton, 2016-12-04:
"error" is a reserved event in socket.io: https://github.com/socketio/socket.io/issues/2285

I will change this event name to another one so that it won't conflict with the reserved event.

Files:
https://github.com/cheton/cnc/blob/v1.8.6/src/app/controllers/Grbl/GrblController.js#L196
https://github.com/cheton/cnc/blob/v1.8.6/src/app/controllers/Grbl/Grbl.js#L658
https://github.com/cheton/cnc/blob/v1.8.6/src/app/controllers/Smoothie/Smoothie.js#L523

cheton, 2016-12-05:
Fixed in 1.8.7

closed 2016-12-05 by cheton

cheton, 2016-12-05:
Just to let you know I changed the streaming protocol since [1.8.3](https://github.com/cheton/cnc/releases/tag/v1.8.3). The so called <b>character-counting streaming protocol</b> can help maximize Grbl performance, but the downside is that it doesn't tell the exact line of error. I will open an issue on https://github.com/grbl/grbl for clarification.

jarretluft, 2016-12-05:
> <https://github.com/cheton/cnc/releases/tag/v1.8.3>. The so called *character-counting
> streaming protocol* can help maximize Grbl performance, but the downside

Thanks @cheton, I will test it out tonight!

A thought on error handling:
This was accomplished in Chilipeppr by maintaining a second array of the
gcode commands sent into grbl as a Queue.  Each command sent to grbl would
also get pushed into the queue, and each OK/Error response would pop the
oldest command off the queue. This way, no matter how many additional gcode
lines had been sent in, when an error was encountered the next line to pop
off the queue would be the cause.

cheton, 2016-12-05:
yeah! I just figured it out what I need to fix for the error handling. Thank you for the guidance.

cheton, 2016-12-05:
Now it's able to display the exact line of an error occurred. Thank you for your suggestion.
[image]