cncjs/cncjs #101: Error using npm install

dru5412, 2016-12-04:
tried npm install cncjs -g today and got lots of errors relating to serialport.

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.6/serialport-v4.0.6-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.6 and node@4.5.0 (node-v46 ABI) (falling back to source compile with node-gyp)
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/4.5.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/cncjs/node_modules/serialport/.node-gyp"
make: Entering directory '/usr/local/lib/node_modules/cncjs/node_modules/serialport/build'
make: *** No rule to make target '../.node-gyp/4.5.0/include/node/common.gypi', needed by 'Makefile'.  Stop.
make: Leaving directory '/usr/local/lib/node_modules/cncjs/node_modules/serialport/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.30+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/usr/local/lib/node_modules/cncjs/node_modules/                                       serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/local/lib/node_modules/cncjs/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/local/lib/node_modules/cncjs/node_modules/serialport
gyp ERR! node -v v4.5.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/usr/local/lib/node_mod                                       ules/cncjs/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/local/lib/node_modules/cncjs/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/cncjs/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:829:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.4.30+
node-pre-gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/cncjs/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/local/lib/node_modules/cncjs/node_modules/serialport
 install: info lifecycle serialport@4.0.6~install: Failed to exec install script

node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok

Seems to just time out on the next step...

Something wrong or something I am doing wrong?

cheton, 2016-12-04:
Try the following if you're using a root account:
```
$ sudo npm uninstall -g cncjs
$ sudo npm install -g cncjs --unsafe-perm
```

dru5412, 2016-12-05:
Thanks for tip, but this happens after the uninstall/install commands too.

Even tried npm install serialport and that throws the same 404 error looking for this file.

https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.6/serialport-v4.0.6-node-v46-linux-arm.tar.gz

Ill try again today to see if problem still there and report back. 

Thanks

cheton, 2016-12-05:
Are you trying to install cncjs on a Raspberry Pi? I suggest you use `nvm` to manage Node.js versions, then you don't need to sudo as root. This will resolve a lot of installation problems.
https://github.com/cheton/cnc#nodejs-installation

dru5412, 2016-12-05:
Managed to resolve the issue, it was due to being root sudo when installing. 

Installing as a user with sudo prefix seemed to resolve it..

Not sure if that was the main cause but fixed now, thanks for help.

Will close issue off.

closed 2016-12-05 by dru5412

mobalized, 2017-02-16:
Having the same issue. Have not been able to find a solution. However if you go to the link that it is trying to pull the serialport tarball from it is throwing a 404... So is it possible that link has changed because I don't see any arm builds on their github page.

cheton, 2017-02-16:
@mobalized

Are you using the root account to install cncjs?

mobalized, 2017-02-16:
@cheton

I have tried as user Pi with sudo, user Pi without sudo, and after sudo -s.

cheton, 2017-02-19:
May I know which gcc/g++ version you're using in your raspberry pi? I saw some people encountered installation problems because they are using wheezy. gcc/g++ 4.8 or above is required for Node.js v4 or higher.