cncjs/cncjs #102: Win64 - Not Starting Up

GeerGuy, 2016-12-04:
I can't get Win64 version of 1.8.5 or 1.8.6 to start up on Windows 10.  Install seems to work, but the form never opens.  "cnc.exe" is shown twice in the Task Manager.

Going back to 1.8.4 works fine.

Request:
Would like to see the gcode file name shown some place.  My gcode file names contain information about the job, like order, max depth, speed and bit selection, so it would be nice to easily see this.  EX. Widget_1_0.75D_2S_EM250

tritao, 2016-12-04:
Same problem here, had to compile from source to get the latest version to open on Windows

cheton, 2016-12-05:
Thank you for reporting this issue. I removed the installer from both 1.8.5 and 1.8.6 releases. I will try to find out what's going wrong with Win32/64 builds.

cheton, 2016-12-05:
It's side effects of commit https://github.com/cheton/cnc/commit/c9c78d9409c45e369dd7c92929788915c88a4e55. The main entry to launch Electron app was accidentally removed, so the window will never open. I will fix it in the 1.8.7 release.

closed 2016-12-05 by cheton

cheton, 2016-12-05:
Fixed in 1.8.7

commit 8a1a94be (2016-12-05): Fix an issue that failed to start Electron app due to a side effect of commit c9c78d9 (resolves #102)