cncjs/cncjs #103: Idea: Remote pendant (Playstation 3 Dualshock Controller / SIXAXIS Controller)

AustinSaintAubin, 2016-12-04:
# Gamepad Control
Goal:Use Use [Playstation 3 Controller](https://www.playstation.com/en-us/explore/accessories/dualshock-3-ps3/) wirelessly over bluetooth to control CNC.

Using a wireless game controller (like a PS3 controller) seems to be one of the lowest cost & simplest solution methods.
Integrating Node.js libraries for support looks like the best option. I also looked at 

### Resources Options:
- Node.js
	- https://www.npmjs.com/package/node-gamepad [Lots of supported controllers, looks most promissing, Uses node-hid]
	- https://www.npmjs.com/package/node-hid [All Human Interface Devices]
	- https://github.com/KenanSulayman/PS3.js
- Python
	- https://pythonhosted.org/triangula/sixaxis.html
	- https://www.piborg.org/rpi-ps3-help
	- http://www.ev3dev.org/docs/tutorials/using-ps3-sixaxis/
	- https://wiki.gentoo.org/wiki/Sony_DualShock
	- https://github.com/ApproxEng/triangula

-----

# Playstation Controller Setup ( general guide to connect hardware )

Here is what I have figured out so far for PS3 on Raspberry PI 3 w/ integrated bluetooth.
The bellow just shows how to get PS3 controller connected.

### Install
```
# Install & Enable Bluetooth Tools
sudo apt-get install -y bluetooth libbluetooth3 libusb-dev
sudo systemctl enable bluetooth.service

# Add pi user to bluetooth group
sudo usermod -G bluetooth -a pi
```

### Pairing Tools
```
# Get and build the command line pairing tool (sixpair)
wget http://www.pabr.org/sixlinux/sixpair.c
gcc -o sixpair sixpair.c -lusb

### Connect PS3 over USB
# Get PS3 DS 
sudo ./sixpair

### Disonnect PS3 over USB
bluetoothctl
### Connect PS3 over USB
devices
agent on
trust MAC # Replace "MAC" with MAC of "Device 64:D4:BD:B3:9E:66 PLAYSTATION(R)3 Controller"
trust 64:D4:BD:B3:9E:66 
quit
### Disonnect PS3 over USB, you should now be able to connect wirelessly. To check this, first list everything in /dev/input:
```

### Test Controller Connectivity
```
# List Devices
ls /dev/input

### PS3 Controller: press the PS button, the lights on the front of the controller should flash for a couple of seconds then stop, leaving a single light on. If you now look again at the contents of /dev/input you should see a new device, probably called something like ‘js0’:

# List Devices
ls /dev/input
```

### Get Battery Level
`cat "/sys/class/power_supply/sony_controller_battery_64:d4:bd:b3:9e:66/capacity"`

### Joystick Application
```
# Install
sudo apt-get install joystick

# Usage / Test
jstest /dev/input/js0
```

AustinSaintAubin, 2016-12-09:
I got it work. Its fucking bad ass. I can control my CNC with PS3 DS3 controller over bluetooth from raspberry pi... its everything i hoped it would be. Still have some minor bugs. But I have work in the morning.
Alpha version:

# Node.js DS3 Controller Setup
 - https://www.npmjs.com/package/node-hid
 - https://www.npmjs.com/package/dualshock-controller

```
# Install Tools
sudo apt-get install -y libudev-dev libusb-1.0-0-dev
sudo apt-get install -y build-essential git
sudo apt-get install -y gcc-4.8 g++-4.8 && export CXX=g++-4.8

# Set access to /usr/lib/node_modules
ls -ld /usr/lib/node_modules; stat --format '%a' /usr/lib/node_modules
sudo chmod a+w /usr/lib/node_modules  # chmod 777
ls -ld /usr/lib/node_modules; stat --format '%a' /usr/lib/node_modules

# Install Node-HID
#########sudo npm install -g node-hid --unsafe-perm
-- OR --
# Compile HID
cd ~/
git clone https://github.com/node-hid/node-hid.git
cd node-hid                                        # must change into node-hid directory
git submodule update --init                        # done on publish automatically
npm install -g  # npm install                      # rebuilds the module
node-pre-gyp rebuild                               # rebuilds the C code
node-pre-gyp install --fallback-to-build

# Install Node-dualshock-controller
#########sudo npm install -g dualshock-controller--unsafe-perm
-- OR --
# Compile HID
cd ~/
git clone https://github.com/rdepena/node-dualshock-controller.git
cd node-dualshock-controller                       # must change into node-hid directory
git submodule update --init                        # done on publish automatically
npm install -g  # npm install                      # rebuilds the module

# Set access to /usr/lib/node_modules
sudo chmod g-w,o-w /usr/lib/node_modules  # chmod 755
ls -ld /usr/lib/node_modules; stat --format '%a' /usr/lib/node_modules

# Test
node ~/node-dualshock-controller/examples/consolePrintDualShock3.js

```

# Socket.io Client
```
sudo npm install -g socket.io-client
```

# The Code
Usage: node test.js

test.js [MIT License](https://opensource.org/licenses/MIT)
```
var io = require('socket.io-client');
var token = 'YOUR_API_KEY'

var socket = io.connect('ws://localhost:8000', {
  'query': 'token=' + token
});

socket.on('connect', function() {
	 console.log('Socket IO Connected!');

	 // List ports
[... 353 lines cut ...]
});
//up-down movement
controller.on('upDown:motion', function (data) {
	 //...doStuff();
});

//controller status
//as of version 0.6.2 you can get the battery %, if the controller is connected and if the controller is charging
controller.on('battery:change', function (value) {
	console.log('battery:change:' + value);
});
controller.on('connection:change', function (value) {
	console.log('connection:change:' + value);
});
controller.on('charging:change', function (value) {
	console.log('connection:change:' + value);
});

//connect the controller
controller.connect();
```

That was a lot to get done in a day... 

[ ] Still need a way to detect when controller connects, then load this script.. also need to close script on connectivity loss with controller... then it should load on connect / reconnect.

cheton, 2016-12-09:
Hi @AustinSaintAubin,

This is pretty cool. Would you like to continue your work in the <b>examples</b> directory or somewhere else? Just let me know if you'd like to become a collaborator.

AustinSaintAubin, 2016-12-10:
I would love to become a collaborator!
I think after I get this flushed out it would like a nice addition to the CNC tools set.

cheton, 2016-12-10:
Thank you. I just sent you an invitation.

You can push your code changes to the [dev](https://github.com/cheton/cnc/tree/dev) branch or create a new branch for development, then merge it to the [master](https://github.com/cheton/cnc/tree/dev) branch once you completed it.

AustinSaintAubin, 2016-12-10:
Will do once have a working beta.

I need some help getting the token.
Right now I have to grad it from another session...
How do I handshake and then get the token?

Would also be nice for uses using authentication to be able to specify a username & password if authentication is enable.

Current test code with static token:
```
var io = require('socket.io-client');
var token = 'YOUR_TOKEN'

var socket = io.connect('ws://localhost:8000', {
  'query': 'token=' + token
});

socket.on('connect', function() {
	 console.log('Socket IO Connected!');

	 // List ports
	socket.emit('list', function(data) {
		console.log('LIST: ' + data);
	});

	// Open port
	socket.emit('open', '/dev/ttyUSB0', { baudrate: Number(115200) });
});
```

cheton, 2016-12-10:
You can try below example to generate an access token using a secret stored in `~/.cncrc`. The access token lifetime seems to be a required field for jsonwebtoken, it can be expressed in seconds (e.g. 60) or a time span string like '7 days', '3d', '1h'. 

```js
var fs = require('fs');
var path = require('path');
var jwt = require('jsonwebtoken');
var io = require('socket.io-client');

// https://github.com/auth0/node-jsonwebtoken#jwtsignpayload-secretorprivatekey-options-callback
var generateToken = function(payload, secret) {
    var token = jwt.sign(payload, secret, {
        expiresIn: '30d' // 30 days
    });

    return token;
};

var getUserHome = function () {
    return process.env[(process.platform === 'win32') ? 'USERPROFILE' : 'HOME'];
};
var cncrc = path.resolve(getUserHome(), '.cncrc');
var config = JSON.parse(fs.readFileSync(cncrc, 'utf8'));
var token = generateToken({ id: '', user: '' }, config.secret);
var socket = io.connect('ws://localhost:8000', {
    'query': 'token=' + token
});

socket.on('connect', function() {
    console.log('Socket IO Connected!');

    // List ports
    socket.emit('list', function(data) {
        console.log('LIST: ' + data);
    });

    // Open port
    socket.emit('open', '/dev/ttyUSB0', { baudrate: Number(115200) });
});

socket.on('error', function() {
    console.log('error');
});
```

AustinSaintAubin, 2016-12-10:
Looks like your code is working well. Thanks for the help.

AustinSaintAubin, 2016-12-12:
BETA v0.1 [2016/12/11]

NOTE: Use [pm2](https://github.com/Unitech/pm2) to run, so it can auto restart the script
```
pm2 start cnc-pendant_ps3.js
```

cnc-pendant_ps3.js
```
// Node.js Playstation 3 / DS3 Controller for CNC.js
// by Austin St. Aubin <austinsaintaubin@gmail.com>
// v0.1 BETA [2016/12/11]
// https://github.com/cheton/cnc/issues/103

// [Dependacies]
var io = require('socket.io-client');  // Socket.io connection to CNC
var jwt = require('jsonwebtoken');
var fs = require('fs');
var path = require('path');
var dualShock = require('dualshock-controller');  // ttps://www.npmjs.com/package/dualshock-controller

// [Varables]
var controller_serial_port = '/dev/ttyUSB0'
var controller_serial_baud = 115200

var socket_address = 'localhost'
	socket_port = '8000'
	socket_token_expiration = '30d' // The access token lifetime is a required field for jsonwebtoken, it can be expressed in seconds (e.g. 60) or a time span string like '7 days', '3d', '1h'.

var socket, controller;

// =====================================================
// Check/Wait for Controller to Connect
setInterval(checkController, 3*1000);
var controler_started = false;

// [Function] check for controller to conect (show up in devices), then start services. Kill services on disconect.
function checkController(socket, controller) {
	console.log('Checkign Controller Status');

	// Check if Controller Exist
	fs.stat('/dev/input/js0', function(err, stat) {
		if(err == null) {
			// Controller Conected
			console.log('js0 exists');

			if (!controler_started) {
				console.log('Starting Services');

				// Start Socket Connection & Controller Conection
				run();

				// Set Startup Varable
				controler_started = true;
			}

		} else if(err.code == 'ENOENT') {
			// Controller Disconected / Not Conected
			console.log('js0 NOT exists');

			if (controler_started) {
				console.log('Stopping Services');

				// Kill Socket Connection
				//socket.destroy();

				// Kill  Controller Conection
				//controller.destroy();

				// !!!!!!!!!!!!!!! Exit Script (to be relauched by pm2)
[... 537 lines cut ...]
		 //...doStuff();
	});

	//forward-back movement
	controller.on('forwardBackward:motion', function (data) {
		 //...doStuff();
	});
	//up-down movement
	controller.on('upDown:motion', function (data) {
		 //...doStuff();
	});

	//controller status
	//as of version 0.6.2 you can get the battery %, if the controller is connected and if the controller is charging
	controller.on('battery:change', function (value) {
		console.log('battery:change:' + value);
	});
	controller.on('connection:change', function (value) {
		console.log('connection:change:' + value);
	});
	controller.on('charging:change', function (value) {
		console.log('connection:change:' + value);
	});
}
```

AustinSaintAubin, 2016-12-15:
Updated above script... still have a few small issues to resolve.
But the above beta script is very good for right now.

AustinSaintAubin, 2016-12-16:
[PS3 CNC Control Button Map](https://docs.google.com/drawings/d/1DMzfBk5DSvjJ082FrerrfmpL19-pYAOcvcmTbZJJsvs/edit?usp=sharing) Work in Progress.

AustinSaintAubin, 2016-12-22:
FYI, new baby today. Going to be busy for the next week.
Will pick this backup once things settle down.

cheton, 2016-12-22:
@AustinSaintAubin  Congratulation! 👪

cheton, 2017-01-09:
Hi @AustinSaintAubin,

Would you like to add your file "cnc-pendant-ps3.js" as a main entry in the package.json file? For example:
```json
"bin": {
  "cnc": "./bin/cnc",
  "cnc-pendant-ps3": "./bin/cnc-pendant-ps3"
},
```

You can add the file to [bin](https://github.com/cheton/cnc/tree/dev/bin), I will publish a new version once it's ready!

AustinSaintAubin, 2017-01-11:
That would be good, but I cant get past "GH006: Protected branch update failed for refs/heads/master.        
Required status check "continuous-integration/travis-ci" is expected. At least one approved review is required."

cheton, 2017-01-11:
Let me check it.

AustinSaintAubin, 2017-01-11:
I think it might be best to branch plugins off into their own repository... but I am still thinking on that... I still need to work on a few things in regard to this script before it is ready for prime time... hope to work on it more this weekend. For now I have commited changes to the [pendant-ps3](https://github.com/cheton/cnc/tree/pendant-ps3) branch.

cheton, 2017-01-11:
Sure! The branch is back, you can push your changes to the pendant-ps3 branch. Thanks.

AustinSaintAubin, 2017-01-25:
Still working on this...

I have some food for thought and need input.
I have two options:

1. Make a separate node js & npm named `cncjs-pendant-ps3` that users can add like any other node.js application. Allows pendants to run anywhere and connect to cncjs.
2. Work to integrate HID & pendant functionality into cnc js. Allows for configuration in cncjs settings and maintenance that occurs within the cncjs project.

What are your thoughts?

cheton, 2017-01-25:
I vote the first one. A separate repository will be convenient for source management and version control.

cheton, 2017-01-26:
I created an organization for managing multiple projects: https://github.com/cncjs. An invitation has already been sent to your mailbox. I will rename and move this project to the organization after 1.9 release, you can create a new repository for <strong>cncjs-pendant-ps3</strong> and push your changes to the repository. 😃

AustinSaintAubin, 2017-01-27:
Going to create `cncjs-pendant-ps3` this Friday or Saturday. Looks like the cnc -> cncjs moveover went well. Still have some documentation to update. Will try to help as time permits.

f0m3, 2017-01-31:
This feature sounds very cool. Will it be possible to use it on a mac?

AustinSaintAubin, 2017-01-31:
I have designed it to run on the raspberry pi, but it should be able to run on Mac OS w/ NodeJS and all dependencies installed. 

I run CNCjs and my pendent on a Raspberry Pi 3. This is ideal for me because end user OS's sleep, auto update, etc. I use my MacBook to connect to the CNCjs web interface running on the Raspberry Pi. Basically everything runs on the Raspberry Pi. The end user devices (MacBook, Chromebook) are then free to do other things. Additionally, because I run the pendent software on the Raspberry Pi, the PS3 controller connects to the Raspberry Pi in my shop. Allowing my to only have to take my PS3 into my dirty areas, no need to bring my MacBook into the shops dust.

AustinSaintAubin, 2017-02-24:
Closing. Any new issues / discussions should be posted to [cncjs/cncjs-pendant-ps3](https://github.com/cncjs/cncjs-pendant-ps3) or [cncjs/cncjs-pendant-boilerplate](https://github.com/cncjs/cncjs-pendant-boilerplate)

closed 2017-02-24 by AustinSaintAubin

spankurmonkey, 2019-03-06:
Is there a specific type of PS3 controller and/or Bluetooth dongle ?  I have a generic Bluetooth PS3 controller and have yet to get it to pair.