cncjs/cncjs #105: cnc 1.8.5 login action lasts about 70 seconds

thymjan, 2016-12-05:
I'm running cnc 1.8.5 on a raspberry pi 1 model B+. I know that's not ideal, but had worked for the last versions. Now I tried the login feature. The login screen shows up immediately, but after hitting the login-Button it lasts about 70 seconds till the GUI appears. Should this be normal behavior?cnc

`cnc -vv` shows errors:
`2016-12-04T15:59:09.199Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/boCLjlqtL_-3fJFmMILYD___h9QwMJD0.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:3482:42)`

cheton, 2016-12-05:
It shall be completed within 1~2 seconds. May you try to open <b>Developer Tools</b> to check if there are any console errors or network connection issues?

[image]

thymjan, 2016-12-05:
Mmmh... don't know what I have to do exactly. Do I have to start the developer tools on the server or the client? Do I have to install them? How do I call them? Sorry, I'm a newbie to this, may you help me?

cheton, 2016-12-05:
You can use the following keyboard shortcuts to open Developer Tools in Chrome browser:

Access DevTools | On Windows | On Mac
------ | -------- | ------
Open Developer Tools | F12, Ctrl + Shift + I | Cmd + Opt + I

BTW, I'm not sure if the 70s delay is related to a dns lookup by this [line](https://github.com/cheton/cnc/blob/v1.8.7/src/app/index.js#L119), but a asynchronous call shall not block program execution. Are you able to ping your hostname like below?
```
$ ping `hostname`
PING cheton-mba.local (192.168.1.100): 56 data bytes
64 bytes from 192.168.1.100: icmp_seq=0 ttl=64 time=0.097 ms
64 bytes from 192.168.1.100: icmp_seq=1 ttl=64 time=0.146 ms
```

cheton, 2016-12-05:
How about the second time opening the web interface after a successful login, are you still encountered a long delay?

thymjan, 2016-12-05:
ping cnc.local
PING cnc.local (192.168.1.13): 56 data bytes
64 bytes from 192.168.1.13: icmp_seq=0 ttl=64 time=37.091 ms
64 bytes from 192.168.1.13: icmp_seq=1 ttl=64 time=55.698 ms
64 bytes from 192.168.1.13: icmp_seq=2 ttl=64 time=16.540 ms
-> This should be no problem, am I right?

Logging in:
[image]

Calling GUI when already logged in:
[image]

cheton, 2016-12-05:
Hmmm. If you try to <b>Sign Out</b> from the nav dropdown, will the long delay happen again?

[image]

thymjan, 2016-12-05:
No. I can this with "normal" behavior.

cheton, 2016-12-05:
No idea how this will happen. You can try cleaning browser cache, removing server config file `~/.cncrc`, and then restart the server again to check which step caused this issue.

thymjan, 2016-12-05:
What about this errors?
```
2016-12-05T16:29:39.679Z - info: Started the server at http://127.0.1.1:8000     at /home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:7702:32
2016-12-05T16:29:44.297Z - debug: [cncengine] New connection from 192.168.1.3: id=rfSaJlRc7WR4tDmrAAAA, token.id=87e60200-fd65-43ac-9a5f-5464fdbceffb, token.name=pi     at Namespace.<anonymous> (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:8138:28)
2016-12-05T16:29:55.774Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/I-H3Jznw6s65OvdGesa6emseusoAl8DR.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:3538:42)
2016-12-05T16:29:55.821Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/I-H3Jznw6s65OvdGesa6emseusoAl8DR.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:3538:42)
2016-12-05T16:29:55.873Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/I-H3Jznw6s65OvdGesa6emseusoAl8DR.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:3538:42)
2016-12-05T16:29:55.924Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/I-H3Jznw6s65OvdGesa6emseusoAl8DR.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:3538:42)
2016-12-05T16:29:55.977Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/I-H3Jznw6s65OvdGesa6emseusoAl8DR.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.4.7/lib/node_modules/cncjs/dist/cnc/app/index.js:3538:42)
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /?lang=de 200 613.575 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/app.css?2ac2e49f43f0ef0444f3b3831e1f7803 200 168.720 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/app-2.css?78485cfa0e2dcd255fb7a415de688cd9 200 179.881 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/polyfill.5ee83d8f43ef736a911d.js 200 127.791 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/app.9c3e101dbd382a2bd503.bundle.js 200 146.037 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/app-1.css?78485cfa0e2dcd255fb7a415de688cd9 200 123.228 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/images/32x32/loading.gif 200 22.246 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/vendor.dc3c09a14f3d76b883f6.js 200 159.765 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU POST /api/signin 401 87.655 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/favicon.ico 200 26.509 ms
p93w3WekhDe083JycXTxQi-YwAeWFPuU GET /58d1d0c2/af7ae505a9eed503f8b8e6982036873e.woff2 200 12.508 ms
```

cheton, 2016-12-05:
The session-file-store error is just fine. It displayed the error message when the server has been restarted, but a opening browser window still kept previous session id, so an error reported.

thymjan, 2016-12-05:
Now I have updated to 1.8.7
Same behavior. I deleted the .cncrc file.
Now if I add a new user there is also this long delay. I hit the "OK"-Button in the "Add new account"-window and again: about 70 seconds till the window disappears.

Maybe the encryption-algorithm is to heavy for the raspberry pi 1 model B+?
`top` shows up nearly 100% CPU-usage for node while I'm waiting.

Same high CPU-load  when logging in.

cheton, 2016-12-05:
Maybe it's caused by the `generateToken` function
https://github.com/cheton/cnc/blob/v1.8.7/src/app/api/api.users.js#L18-L24

```js
const generateToken = (payload, secret = settings.secret) => {
    const token = jwt.sign(payload, secret, {
        expiresIn: 60 * 60 * 24 // expires in 24 hours
    });

    return token;
};
```

I will take a look at <b>jsonwebtoken</b> again and check if there are some lightweight encryption algorithms that can be used as jwt.sign() options.
https://github.com/auth0/node-jsonwebtoken

cheton, 2016-12-08:
I have no RPi 1, but it also took about 3~4 seconds delay on my RPi2. Sorry for that I didn't find a faster solution so far, however, current implementation will re-validate expired access token after 24 hours. How about to extend the expiration time to a longer period like 30 days as a workaround, or adding a configurable value?

AustinSaintAubin, 2016-12-15:
configurable value might be best for anyone wanting to still have it expire after 24hrs.

cheton, 2016-12-15:
A new `--access-token-lifetime` option is now available in the master release:
https://github.com/cheton/cnc/blob/master/src/cnc.js#L40
```
--access-token-lifetime <lifetime>', 'access token lifetime in seconds or a time span string (default: 30d)
```

It will be published with 1.8.8 release.

cheton, 2016-12-22:
Hi @thymjan,

You can install the latest version (v1.8.8), and pass access token lifetime as a command-line option or save this configuration permanently to `~/.cncrc` file.

Check out the usage at https://github.com/cheton/cnc#usage

closed 2016-12-22 by cheton