cncjs/cncjs #106: G2 arc code freeze

gendebeat, 2016-12-08:
Hello. i think this programs is great!
I use it for a year and I like it more than the web of chilipeperr.com,It is comfortable and visually pleasing
I use now tinyg2, de 100.x branch.
i use vectric aspire and when i generate gcode with arcs, g2 or g3, the tinyg2 freeze or make bad arcs.
First i  think was a tinyg2 issue, may be the configuration. i try other earlier versions but not resolve the problem. 
But yesterday i try to send the same gcode through chilipepperr.com, and woilaa, it works correct. 

When i use this program to send gcode like this:
> G17
> G21
> G0 Z2.0010
> G0 X0.0000 Y0.0000
> G0 X5.0000 Y0.0000 Z2.0000
> G1   Z-1.0000 F10000.0
> G2 X0.0000 Y5.0000 I0.0000 J5.0000 F1000.0
> G2 X5.0000 Y10.0000 I5.0000 J0.0000
> G2 X10.0000 Y5.0000 I0.0000 J-5.0000
> G2 X5.0000 Y0.0000 I-5.0000 J0.0000
> G0   Z2.0000
> G0 Z2.0010
> G0 X0.0000 Y0.0000
> M30
the program stops o make bad desings (like straigth instead of arcs).
some times in one point. sometimes in other. 
may be the problem is with the J besause, 
when i send code like this:
> G17 
> G21 
> G2 I50 F2000
it makes the correct circle. 

Can it be a problem sending the gcode through the json server?

thanks

cheton, 2016-12-09:
It works well if I send commands (in JSON format) line by line from the Console widget. You can try it manually:
```
{gc: "G17"}
{gc: "G21"}
{gc: "G0 Z2.0010"}
{gc: "G0 X0.0000 Y0.0000"}
{gc: "G0 X5.0000 Y0.0000 Z2.0000"}
{gc: "G1 Z-1.0000 F10000.0"}
{gc: "G2 X0.0000 Y5.0000 I0.0000 J5.0000 F1000.0"}
{gc: "G2 X5.0000 Y10.0000 I5.0000 J0.0000"}
{gc: "G2 X10.0000 Y5.0000 I0.0000 J-5.0000"}
{gc: "G2 X5.0000 Y0.0000 I-5.0000 J0.0000"}
{gc: "G0 Z2.0000"}
{gc: "G0 Z2.0010"}
{gc: "G0 X0.0000 Y0.0000"}
{gc: "M30"}
```

However, if I send the whole in a batch, things goes wrong just like you said - straight instead arcs. I will do a test on Chillipeppr and check for differences.

cheton, 2016-12-09:
Here is my output with `cnc -vvv`, I removed the last line M30 in this test:
```
2016-12-09T02:44:19.961Z - silly: [TinyG2] < {"r":{},"f":[1,0,12]}
2016-12-09T02:44:19.962Z - silly: [TinyG2] < {"r":{},"f":[1,0,12]}
2016-12-09T02:44:19.966Z - silly: [TinyG2] < {"r":{},"f":[1,0,18]}
2016-12-09T02:44:19.966Z - silly: [TinyG2] < {"qr":27,"qi":1,"qo":0}
2016-12-09T02:44:19.967Z - silly: [TinyG2] < {"r":{},"f":[1,0,17]}
2016-12-09T02:44:19.968Z - silly: [TinyG2] < {"r":{},"f":[1,0,20]}
2016-12-09T02:44:19.969Z - silly: [TinyG2] < {"qr":26,"qi":1,"qo":0}
2016-12-09T02:44:19.970Z - silly: [TinyG2] < {"r":{},"f":[1,0,22]}
2016-12-09T02:44:19.970Z - silly: [TinyG2] < {"qr":25,"qi":1,"qo":0}
2016-12-09T02:44:19.972Z - silly: [TinyG2] < {"r":{},"f":[1,0,23]}
2016-12-09T02:44:19.973Z - silly: [TinyG2] < {"r":{},"f":[1,0,24]}
2016-12-09T02:44:19.975Z - silly: [TinyG2] < {"r":{},"f":[1,0,25]}
2016-12-09T02:44:19.978Z - silly: [TinyG2] < {"r":{},"f":[1,0,24]}
2016-12-09T02:44:19.980Z - silly: [TinyG2] < {"r":{},"f":[1,0,14]}
2016-12-09T02:44:19.982Z - silly: [TinyG2] < {"r":{},"f":[1,0,18]}
2016-12-09T02:44:19.984Z - silly: [TinyG2] < {"r":{},"f":[1,0,17]}
2016-12-09T02:44:19.988Z - silly: [TinyG2] < {"qr":5,"qi":20,"qo":0}
2016-12-09T02:44:20.213Z - silly: [TinyG2] < {"sr":{"vel":70.18,"stat":5,"cycs":1,"mots":1,"momo":0,"posz":0.069,"mpoz":0.069}}
2016-12-09T02:44:20.463Z - silly: [TinyG2] < {"sr":{"vel":228.65,"posz":0.784,"mpoz":0.784}}
2016-12-09T02:44:20.718Z - silly: [TinyG2] < {"sr":{"vel":156.70,"posz":1.665,"mpoz":1.665}}
2016-12-09T02:44:20.877Z - silly: [TinyG2] < {"qr":6,"qi":0,"qo":1}
2016-12-09T02:44:20.971Z - silly: [TinyG2] < {"sr":{"vel":118.62,"posx":0.168,"posz":2.001,"mpox":0.168,"mpoz":2.001}}
2016-12-09T02:44:21.224Z - silly: [TinyG2] < {"sr":{"vel":321.49,"posx":1.041,"posz":2.001,"mpox":1.041,"mpoz":2.001}}
2016-12-09T02:44:21.478Z - silly: [TinyG2] < {"sr":{"vel":418.91,"posx":2.708,"posz":2.000,"mpox":2.708,"mpoz":2.000}}
2016-12-09T02:44:21.733Z - silly: [TinyG2] < {"sr":{"vel":267.10,"posx":4.252,"posz":2.000,"mpox":4.252,"mpoz":2.000}}
2016-12-09T02:44:21.986Z - silly: [TinyG2] < {"sr":{"vel":110.41,"posx":4.945,"posz":2.000,"mpox":4.945,"mpoz":2.000}}
2016-12-09T02:44:22.011Z - silly: [TinyG2] < {"qr":7,"qi":0,"qo":1}
2016-12-09T02:44:22.241Z - silly: [TinyG2] < {"sr":{"vel":203.76,"feed":10000.00,"momo":1,"posx":5.000,"posz":1.479,"mpox":5.000,"mpoz":1.479}}
2016-12-09T02:44:22.491Z - silly: [TinyG2] < {"sr":{"vel":296.97,"posz":0.328,"mpoz":0.328}}
2016-12-09T02:44:22.746Z - silly: [TinyG2] < {"sr":{"vel":155.18,"posz":-0.686,"mpoz":-0.686}}
2016-12-09T02:44:22.896Z - silly: [TinyG2] < {"qr":8,"qi":0,"qo":1}
2016-12-09T02:44:23.002Z - silly: [TinyG2] < {"sr":{"vel":154.17,"momo":2,"posx":4.795,"posy":0.013,"posz":-1.000,"mpox":4.795,"mpoy":0.013,"mpoz":-1.000}}
2016-12-09T02:44:23.125Z - silly: [TinyG2] < {"qr":9,"qi":0,"qo":1}
2016-12-09T02:44:23.253Z - silly: [TinyG2] < {"sr":{"vel":254.47,"posx":4.052,"posy":0.428,"mpox":4.052,"mpoy":0.428}}
2016-12-09T02:44:23.508Z - silly: [TinyG2] < {"sr":{"vel":516.69,"posx":3.066,"posy":1.713,"mpox":3.066,"mpoy":1.713}}
2016-12-09T02:44:23.762Z - silly: [TinyG2] < {"sr":{"vel":554.75,"posx":1.623,"posy":3.593,"mpox":1.623,"mpoy":3.593}}
2016-12-09T02:44:24.016Z - silly: [TinyG2] < {"sr":{"vel":302.31,"posx":0.492,"posy":5.067,"mpox":0.492,"mpoy":5.067}}
2016-12-09T02:44:24.191Z - silly: [TinyG2] < {"qr":10,"qi":0,"qo":1}
2016-12-09T02:44:24.270Z - silly: [TinyG2] < {"sr":{"vel":236.26,"posx":0.098,"posy":5.932,"mpox":0.098,"mpoy":5.932}}
2016-12-09T02:44:24.524Z - silly: [TinyG2] < {"sr":{"vel":302.96,"posx":0.770,"posy":6.695,"mpox":0.770,"mpoy":6.695}}
2016-12-09T02:44:24.779Z - silly: [TinyG2] < {"sr":{"vel":528.01,"posx":2.281,"posy":7.705,"mpox":2.281,"mpoy":7.705}}
2016-12-09T02:44:25.033Z - silly: [TinyG2] < {"sr":{"vel":478.75,"posx":4.136,"posy":8.944,"mpox":4.136,"mpoy":8.944}}
2016-12-09T02:44:25.287Z - silly: [TinyG2] < {"sr":{"vel":256.20,"posx":5.378,"posy":9.774,"mpox":5.378,"mpoy":9.774}}
2016-12-09T02:44:25.363Z - silly: [TinyG2] < {"qr":12,"qi":0,"qo":2}
2016-12-09T02:44:25.541Z - silly: [TinyG2] < {"sr":{"vel":246.39,"posx":6.328,"posy":9.779,"mpox":6.328,"mpoy":9.779}}
2016-12-09T02:44:25.794Z - silly: [TinyG2] < {"sr":{"vel":409.25,"posx":7.040,"posy":8.713,"mpox":7.040,"mpoy":8.713}}
2016-12-09T02:44:26.048Z - silly: [TinyG2] < {"sr":{"vel":539.46,"posx":8.225,"posy":6.939,"mpox":8.225,"mpoy":6.939}}
2016-12-09T02:44:26.303Z - silly: [TinyG2] < {"sr":{"vel":371.28,"posx":9.355,"posy":5.248,"mpox":9.355,"mpoy":5.248}}
2016-12-09T02:44:26.528Z - silly: [TinyG2] < {"qr":14,"qi":0,"qo":2}
2016-12-09T02:44:26.557Z - silly: [TinyG2] < {"sr":{"vel":247.19,"posx":9.938,"posy":4.248,"mpox":9.938,"mpoy":4.248}}
2016-12-09T02:44:26.813Z - silly: [TinyG2] < {"sr":{"vel":258.41,"momo":0,"posx":9.503,"posy":3.456,"posz":-0.797,"mpox":9.503,"mpoy":3.444,"mpoz":-0.788}}
2016-12-09T02:44:27.068Z - silly: [TinyG2] < {"sr":{"vel":531.56,"posx":8.317,"posy":2.545,"posz":-0.060,"mpox":8.317,"mpoy":2.545,"mpoz":-0.040}}
2016-12-09T02:44:27.323Z - silly: [TinyG2] < {"sr":{"vel":555.64,"posx":6.574,"posy":1.208,"posz":1.022,"mpox":6.574,"mpoy":1.208,"mpoz":1.022}}
2016-12-09T02:44:27.578Z - silly: [TinyG2] < {"sr":{"vel":253.73,"posx":5.298,"posy":0.229,"posz":1.815,"mpox":5.298,"mpoy":0.229,"mpoz":1.815}}
[... cut ...]
2016-12-09T02:44:36.398Z - debug: [cncengine] socket.command("/dev/cu.usbmodemFD121", "stop"): id=mqe6Iz2J_8JpQRO8AAAB, args=[]     at Socket.<anonymous> (/Users/cheton/.nvm/versions/node/v4.5.0/lib/node_modules/cncjs/dist/cnc/app/index.js:8240:32)
2016-12-09T02:44:36.399Z - silly: [TinyG2] > !%
2016-12-09T02:44:36.650Z - silly: [TinyG2] > {clear:null}
2016-12-09T02:44:36.650Z - silly: [TinyG2] > {"qr":""}
2016-12-09T02:44:36.651Z - silly: [TinyG2] < {"r":{"clear":null},"f":[1,0,12]}
2016-12-09T02:44:36.652Z - silly: [TinyG2] < {"r":{"qr":28},"f":[1,0,9]}
```

cheton, 2016-12-09:
My firmware build is 78.02. It didn't work for me when running this G-code program on Chilipeppr's TinyG workspace as well. Sometimes G2 goes straight line with random moves and unexpected result, not sure if it's related to [these issues](https://github.com/synthetos/TinyG/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20g2%20g3).

Hi @jarretluft,

Do you have chances to test the G-code program with TinyG2 on Chilipeppr?
```
G17
G21
G0 Z2.0010
G0 X0.0000 Y0.0000
G0 X5.0000 Y0.0000 Z2.0000
G1 Z-1.0000 F10000.0
G2 X0.0000 Y5.0000 I0.0000 J5.0000 F1000.0
G2 X5.0000 Y10.0000 I5.0000 J0.0000
G2 X10.0000 Y5.0000 I0.0000 J-5.0000
G2 X5.0000 Y0.0000 I-5.0000 J0.0000
G0 Z2.0000
G0 Z2.0010
G0 X0.0000 Y0.0000
M30
```

cheton, 2016-12-09:
I made a firmware build 100.12 from [G2 edge branch](https://github.com/synthetos/g2/tree/edge), and it still cannot work correctly on both cnc and Chilipeppr.

gendebeat, 2016-12-10:
??¡ I with Chilipeppr have not problems. 
Summarizing:
I have g2 firmware 100.12 too. I have cnc 1.8.7, and Chilipeppr from web.
Now i try this gcode. Its a circle with radius 10mm. I use laser gcode to see the results.

```
G17
G21
G0 X0.0000 Y0.0000 M03 S0
G0 X10.0000 Y0.0000 M03 S0
G1   F10000.0 M03 S50 
G2 X0.0000 Y10.0000 I0.0000 J10.0000 F2000.0 M03 S50
G2 X10.0000 Y20.0000 I10.0000 J0.0000
G2 X20.0000 Y10.0000 I0.0000 J-10.0000
G2 X10.0000 Y0.0000 I-10.0000 J0.0000
S0
M05
G0 X0.0000 Y0.0000
M30 
```

With Chilipeppr it make a fluid circle, not problem. It do a good round corners in squares too. 
not problem using g2 or g3 code with I and J parameter.
With CNC 1.8.7 some software freeze: CNC, g2core?  may be can laser a quartes of circle and then stop
(danger, laser stays ON) I need to reset arduino due, reset serial json server and reconect CNC sofware. 
May be a problem with serial comunication CNC<->JSON ??

amx1, 2016-12-29:
Dear Cheton, dear Gendebeat,

I observed the same behavior using arcs (G2 and G3 commands), even with the latest firmware 100.15. The problem seems to be the handshake protocol of the TinyG2 controller. G2 and G3 commands may lead to a buffer overflow of the planner buffer, because these commands fill the planner buffer directly. I tried to modify the code due to the description of the communication protocol. 1) send a command to the controler 2) wait for a receive (ack -  send as r: 3) send next command. This didn't work with arc commands, because these commands fill the planner buffer with a lot more than one value and the qr response is to slow. If you send more commands when the planner buffer gets filled by the G2/G3 commands, the controler crashes (??) I tried to program around this issue, but I don't know if it is the best implementation using the status report of the controler when a motion command is finished. (sr: line: #) The commands will be checked before sending and generate line numbers. If a G2/G3 command is sent, the sender will wait until the planner buffer has some free space and motion command is completed. I tested it with some different programs and looks quite good. You will find the modified files as attachment.

Best

Axel

[TinyG2.zip](https://github.com/cheton/cnc/files/678029/TinyG2.zip)

cheton, 2016-12-30:
@amx1 I will do some tests for your code. Thank you so much for your help.

amx1, 2017-01-07:
Hi,
I am still working on the TinyG2 problem. (Even the latest version of CNC has problems with arc commands on TinyG2). The update of the firmware V100.17 for TinyG2 arduino due based hardware  changed the handshake behavoir of the controller pretty much. Alden Hart is working on it. The handshake is very close to the proposed protocol now. If you read the TinyG2 forum, he is still fighting with the input buffer overflow issue.  I changed my old approach - which tried to overcome the problems of V100.16 and earlier. As attachment you will find my actual version, which is pretty much straight forward now. Dont forget to update the controller to V100.17. It doesnt run on 100.16 or earlier. It seems to run pretty smooth and you can really take advantage of the large planner queue. 

I tested the codes of the example directory, e.g. jsdc.gcode, g2-arc.gcode, ... Some examples like the helical and the 3D print codes don't work, because the commands (E... and 3D arc) are not supported by TinyG2.

Best

Axel

[TinyG2_V100.17.zip](https://github.com/cheton/cnc/files/691764/TinyG2_V100.17.zip)

cheton, 2017-01-08:
Thank you for the information. I did experience some problems with v100.15 and cannot figure out what went wrong. I will try your solution with v100.17 and keep you updated.

amx1, 2017-01-08:
> I think we're seeing a lot over over-complication on the sender side. A lot
> of this is due to historical protocols getting mixed up with the current
> protocol. Let me explain.
>
> The simplest way to send to a g2core machine is to simply send lines. You
> have to honor when to OS blocks, but that's usually just a setting when you
> open the buffer. We'll call this the "blasting" technique.
>
> The main down side of that "blasting data" technique is that the buffer on
> the host side becomes large and you can't send a feed hold. Also, some
> hosts (small Linux machines in particular) don't handle large send buffers
> well and use more processor than they should (probably in paging the buffer
> to/from RAM).
>
> So, the next simplest way to send to a g2core machine, and the simplest
> with a very responsive feed hold, is to:
>
> 1) send a line
> 2) wait for the {r} response
> 3) go to 1 if there are more lines to send
> 4) done
>
> We'll call this "simple line mode."
>
> There is one problem with that technique in that some moves are extremely
> [... 27 lines cut ...]
>
> We deprecated queue reports (qr) just for what amx1 was saying about how
> some gcode lines (like an arc) can fill an empty queue all by themselves.
> There's no clear relationship between a "line" and a "queue entry," and it
> was way too hard to keep track of them.

Dear Cheton,

 

i contacted the developers of TinyG2.  It is very simple now with V100.17 and „simple linemode“ it works like a charm. I attached the latest version. BTW it is not working with V100.15.

 

Best

 

Axel

cheton, 2017-01-09:
Note that I made some code changes to the controller interface in v1.8.13. If you need to merge new changes, you need to update your code with new initialization parameter and the 'load' command callback.

https://github.com/cheton/cnc/blob/83377954b77ac84cd5c8b1aad6e11f47967ceaec/src/app/controllers/TinyG2/TinyG2Controller.js#L105
```js
this.sender = new Sender(STREAMING_PROTOCOL_SEND_RESPONSE);
```

https://github.com/cheton/cnc/blob/83377954b77ac84cd5c8b1aad6e11f47967ceaec/src/app/controllers/TinyG2/TinyG2Controller.js#L491
```js
'load': () => {
    const [name, gcode, callback = noop] = args;

    const ok = this.sender.load(name, gcode);
    if (!ok) {
        callback(new Error(`Invalid G-code: name=${name}`));
        return;
    }

    log.debug(`[TinyG2] Load G-code: name="${this.sender.name}", size=${this.sender.gcode.length}, total=${this.sender.total}`);

    this.workflowState = WORKFLOW_STATE_IDLE;
    callback(null, { name: name, gcode: gcode });
},
```

cheton, 2017-01-09:
I upgraded my Arduino Due firmware to g2core v100.19 and tested several gcode files (e.g. "jsdc.gcode") with your code, but it didn't report machine position and all stopped from the beginning.

[image]

May I have your terminal output to compare the differences? You can run `cnc -vvv` with verbose output and try loading jsdc.gcode as shown below:

[image]

cheton, 2017-01-09:
I created a g2core branch (https://github.com/cheton/cnc/tree/g2core). You can try it with v100.17 on an Arduino due board.

amx1, 2017-01-09:
Ok, we worked in parallel. There were a lot of left overs in my old
version. I cleaned it up and made an update to 1.8.13. You might try it. I
included a diff as patch input to see what I changed. The jsdc example runs
smooth.  I attached the log. I am on nvm version 7.3.0 and npm 4.0.5.

After some exchange with aldenhart the main issue was using JSON stringify
for sending gcodes. This is not forseen and leads to an error (104) and
bypasses the planner queue - but executes the commands.

There is one open point left: running codes in imperial coordinates, e.g.
the circle-inch.gcode The controller reports the work coordinates in inch
but the machine position still in mm. I did not find the position in the
cnc code where to fix it. The log is attached.

aldenhart, 2017-01-09:
The machine position is always reported in MM, and in absolute (machine) coordinates with no offsets. This is to provide UIs with the ability to always display a graphic position without having to track the units more or state of the offsets (which can get complicated).

cheton, 2017-01-09:
Hi amx1,

I cannot see your attachment, could you upload files again?

BTW, please let me know if you are interested in becoming a project collaborator, I can add you to the collaborator list, and you can push commits to the g2core branch for verification.

amx1, 2017-01-09:
Hi Cheton,

ok,

cheton, 2017-01-10:
Invitation sent

amx1, 2017-02-01:
The firmware TinyG2 for arduino due had some issues suddenly stopping the job, but the problems should be solved now. Aldenhart and Giseburt will include the changes in the next few days in the edge branch. You might test the FW branch dev-209b...  https://github.com/synthetos/g2/issues/209. I had no errors since then.

I also updated the TinyG controller part of cnc. Thank you cheton for inviting me and setting up a separate branch, but I hope it is fixed now and I will leave to you to include the changes into your dev branches. I think we don't need a separate branch for this now. There is one point left: The performance of the GUI depends very much on the  power and the OS of the machine running the browser. My setup: the controller is connected to a X86 PC running Ubuntu 16.04 64bit. The front end is a dual boot laptop dell x86 Windows 8.1 or Ubuntu 16.04 64bit, Wifi. The GUI is quite slow and the updates of the positions of the machine are stuttering on Windows 8.1 either with chrome or firefox. On linux it is very smooth in both cases even on the same Laptop! The controller is not loosing any commands or steps in all cases. It is mainly a GUI problem.

Attached you will find the updated version based on v1.8.15. It implements the simple linemode protocol with a low / high water threshold to throttle the sender in case of very fast jobs. Otherwise you might get not updates of the position. See discussions with fw developers https://github.com/synthetos/g2/issues/209

[CNC_TinyG2_g2core_V1.zip](https://github.com/cncjs/cncjs/files/744568/CNC_TinyG2_g2core_V1.zip)

amx1, 2017-02-01:
I just patched the 1.9.0 alpha version to support the latest g2core FW on Arduino Due

@cheton I had to change the init sequence, because you can't split the sr sequence in two parts. The first one will be overwritten on Arduino Due. There is a buffer limit of 254 chars, so I implemented a compression hack. 

[TinyG2_g2core_1.9.0.zip](https://github.com/cncjs/cncjs/files/745369/TinyG2_g2core_1.9.0.zip)

cheton, 2017-02-01:
Thank you for your efforts! I will make a test by tomorrow.

cheton, 2017-02-02:
@amx1,

Thanks for the information. I took a look at https://github.com/synthetos/g2/issues/209 and made several tests, the result of my experiment confirms the same thing as @giseburt said in https://github.com/synthetos/g2/issues/209#issuecomment-271165948, sending `{"gc":...}` will bypass queue throttling and caused unexpected behaviour with G2 or G3 arcs. Change the sending routine to  send simple command string instead of `{gc:...}` can resolve this issue with TinyG firmware version 0.97 and 0.98. 🥂 

In addition, I'll also remove the help query (?) from my init commands, as it will turn subsequent status reports to standard text output, not in JSON format. That might be the actual reason why I chose to use `{"gc":...}` to send g-code lines from the beginning, since sending `{"gc":...}` will enforce JSON reporting.

Hi @giseburt,

After upgrading to g2core firmware build 100.19 for my Arduino Due, I faced two issues as listed below:

* Both work and machine positions are not updated in status reports. You can check the output, after sending initialization commands, I manually send "G0X50Y50", "G0X20Y20Z50", "{sr:null}", and "G0X0Y0" as shown below, but couldn't get updated result for machine and work positions.

```
{"r":{"fv":0.99,"fb":100.19,"fbs":"100.19-12-gb6995","fbc":"settings_default.h","hp":3,"hv":0,"id":"0084-7bd6-29c6-7bd","msg":"SYSTEM READY"},"f":[1,0,1]}
> {"ej":1}
{"r":{"ej":1},"f":[1,0,10]}
> {"jv":4}
{"r":{"jv":4},"f":[1,0,10]}
> {"qv":2}
{"r":{"qv":2},"f":[1,0,10]}
> {"sv":1}
{"r":{"sv":1},"f":[1,0,10]}
> {"si":250}
{"r":{"si":250},"f":[1,0,12]}
{"sr":{"stat":3}}
> {sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
{"r":{"sr":{"line":7,"vel":0,"feed":500,"stat":3,"cycs":0,"mots":0,"hold":0,"momo":0,"coor":1,"plan":0,"unit":1,"dist":0,"frmo":1,"path":2,"posx":0,"posy":0,"posz":0,"posa":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0}},"f":[1,0,161]}
> {"hp":null}
> {"fb":null}
{"r":{"hp":3},"f":[1,0,13]}
{"r":{"fb":100.19},"f":[1,0,13]}
> {"mt":null}
> G0X50Y50
{"r":{},"f":[1,0,9]}
{"sr":{"line":0,"stat":5,"cycs":1}}
> G0X20Y20Z50
{"r":{},"f":[1,0,12]}
> {sr:null}
{"r":{"sr":{"line":0,"vel":0,"feed":500,"stat":5,"cycs":1,"mots":0,"hold":0,"momo":0,"coor":1,"plan":0,"unit":1,"dist":0,"frmo":1,"path":2,"posx":0,"posy":0,"posz":0,"posa":0,"mpox":0,"mpoy":0,"mpoz":0,"mpoa":0}},"f":[1,0,11]}
> G0X0Y0
{"r":{},"f":[1,0,7]}
```

* With g2core firmware build 100.19, this gcode file "circle.nc"
```
G0 X-5 Y0 Z0 F200
G2 X0 Y5 I5 J0 F200
G2 X5 Y0 I0 J-5
G2 X0 Y-5 I-5 J0
G2 X-5 Y0 I0 J5
G1 Z1 F500
G0 X0 Y0 Z5
```

  reported an error of "159: Arc specification error - endpoint is starting point"
  ```
  > G2 X5 Y0 I0 J-5
  {"r":{"n":2},"f":[1,159,17]}
  ```

This error doesn't happen with firmware version 0.98 (firmware build: 87.01).

amx1, 2017-02-02:
@cheton 
I made a fast check with my modifications. I see no errors with your short example. It is running without problems. I attached the full log (-vvv) of my actual version.

I saw also the effect of the missing reports of line, feed, etc. The problem is, that  you split the configuration of the report in 2 parts from v1.8.15 to 1.9.0 during init. All switches not set to true, will be set to false by default. Therefore the first part is ignored. You can overcome this and the limitation of 254 character by compressing the commands using t instead of true. It will give you some extra space. With this option the whole configuration could be send in one line. If you take a look at my version, you can see that I put everything in one command and used a .replace after JSON.stringify. It was the fastest way to implement it. Take a look at the sent commands in the log marked as: "[TinyG2] > Init: ...."

[Test.zip](https://github.com/cncjs/cncjs/files/748116/Test.zip)

cheton, 2017-02-02:
I've reduced the init command as below within the limit of 254 characters, so it should not be a  problem of serial buffer overrun:
```
{sr:{line:t,vel:t,feed:t,stat:t,cycs:t,mots:t,hold:t,momo:t,coor:t,plan:t,unit:t,dist:t,frmo:t,path:t,posx:t,posy:t,posz:t,posa:t,mpox:t,mpoy:t,mpoz:t,mpoa:t}}
```

The strange thing is that my g2core firmware build 100.19 reported the error code 159 for the line 2:
```
> G2 X5 Y0 I0 J-5
{"r":{"n":2},"f":[1,159,17]}
```

I also noticed that we have one minor difference in the <b>fbs</b> string: "100.19-16-g80bf-dirty" vs. "100.19-12-gb6995"

 
May I tried using your built firmware? Either .elf or .bin is fine. Thus we can make a test on the same basis. Thank you!

amx1, 2017-02-02:
@cheton 
for shure, btw did checkout the Motate branch too?

[Test_g2core.zip](https://github.com/cncjs/cncjs/files/748261/Test_g2core.zip)

amx1, 2017-02-02:
@cheton 

looking at {"r":{"n":2},"f":[1,159,17]}, the controller takes this command as line number 2  because of "n":2.  Which sender routine are you using? The original one or mine, because the original one checks only the f statement which means no error, my version waits for the correct line number. The  {"r":{"n":... This is sent when the command is processed.

cheton, 2017-02-02:
I just tested your firmware build and it ran without problems. I remembered that I checkout edge branch for g2core, but forgot to checkout the edge branch for the Motate submodule. Would it be the problem?

amx1, 2017-02-02:
@cheton 
I used the marked branches: dev-209b-random-pauses and dev-5-usb-interrupts these are the latest modifications of aldenhart and giseburt and will be merged to edge/master in a few days. check your versions (git branch -l marked with *)

am@PC-LAB:~/workspace/git/TinyG2/g2core$ git branch -l
* dev-209b-random-pauses
  edge
am@PC-LAB:~/workspace/git/TinyG2/g2core$ cd ../Motate
am@PC-LAB:~/workspace/git/TinyG2/Motate$ git branch -l
* dev-5-usb-interrupts
  master
am@PC-LAB:~/workspace/git/TinyG2/Motate$

cheton, 2017-02-02:
May I confirm how did you create your build?

My steps: `cd g2core; make BOARD=gShield`

amx1, 2017-02-02:
I think you should also specify the CONFIG parameter, I defined my own values based on the Othermill settings. I started with:

cd g2core; make clean; make CONFIG=Othermill BOARD=gshield

You may take a look at boards.mk to find the right combination for you, because the CONFIG parameter may overwrite the BOARD definition and selects the wrong ARM platform

cheton, 2017-02-03:
The CONFIG parameter works! You saved my day. Thank you so much. 👍

amx1, 2017-02-03:
Perfect! 

There is one thing left for jobs with imperial coordinates. The controller return the wpos, vel and feed  in inch but the machine pos always in  mmi dont know where to implement this conversion

cheton, 2017-02-03:
I also noticed that yesterday, it's fixed in this commit:
https://github.com/cncjs/cncjs/commit/c64edd0b1e7223433de3c2f48433abb8d6f01109

giseburt, 2017-02-03:
"pos (IOW posx, posy, etc) are always in mm.

"mpo" (mpox, mpoy, etc) are in the current units.

Also, the current unit is in the "unit" value. 0 is inches (G20), and 1 is
mm (G21).

These can all be put into the SR. (Note you can't add a group to the SR, so
you have to add mpox, mpoy, not mpo.)

-Rob

aldenhart, 2017-02-03:
Actually, it's the other way around. `pos_` is returned in in work coordinates - current units with all offsets applied, and `mpo_` is returned in internal machine coordinates - absolute coordinates (G53 with no offsets) and mm units. (MPO means machine position).

The reason for this is that DROs usually need pos responses, and graphics displays usually need mpo responses.

amx1, 2017-02-03:
@cheton 
I will check this at the weekend

cheton, 2017-02-04:
@amx1

For the GUI performance issue you mentioned earlier, I will remove the list view from the G-code widget in 1.9. Each time when receiving new state update from server, it will re-render the list view, and it will cause significant performance drop on auto scrolling. 

[image]

amx1, 2017-02-04:
@cheton 

I checked the latest master branch, sorry to say, the G2 error is back again. I tried both parameters: complete line mode and simple line mode. see test and log files.

I think the complete linemode is not needed, because the USB connection is fast enough. The idea of CLM is to "burst" the first 4 lines to fill the planner buffer and then loop in simple linemode, but it has the risk that even the first 4 lines scramble the input buffer. 

I will check the code. I think I saw the problem already.

I had a different issue with GUI performance. It is related to the update of the position of the tool in the 3D graph. It is not due to your 3D implementation. It is due the concept of the software of the controller for very fast moves. It is clear that the controller has to prioritize the different tasks. The control of the steppers has to have the absolute priority. Therefore status reports are delayed or postponed in the case of very fast jobs (series of very fast moves). The only feedback is the ACK of a received command, {r: You can see this effect even in the attached log files. You send a lot of commands without receiving a status update of pos, vel, feed, ... You will have no information to update the position of the tool in the 3D widget. The tool is "jumping" with very large steps. It doesn't influence the correct function of the controller. The cnc machine is running properly, but the display doesn't look nice.

My approach was to implement the simple linemode, with a throttling when the planner buffer is full. I stop the sender if qr =4 and restart the sender when e.g. qr=6. This will give some headroom to receive an update of pos, vel and also line. You can see the different behaviour in the log files. I attached also the log for my actual version for the same test case

[Test_20170204.zip](https://github.com/cncjs/cncjs/files/752218/Test_20170204.zip)
. (test_amx1.log)

cheton, 2017-02-04:
Thank you for your detailed and clear explanation.

I'm not aware the effect of delayed status reports for a very fast job. The reason why I tried not to watch queue reports since it will be deprecated as mentioned earlier by @giseburt.

G2 and G3 arcs did work when I tested yesterday, but now it happened again after I made several code changes. I think the G2 error might be caused by refactoring the workflow state (https://github.com/cncjs/cncjs/commit/877b674955a7e85e8bbf3c667ecf3fdb0223fbd3). I will take a closer look at your attached logs for further investigation.

amx1, 2017-02-04:
> refactoring the workflow state (877b674

I tried to implement my first idea. It is running, but I did not finish
checking yet.

amx1

amx1, 2017-02-04:
@cheton

the sending routine seems to work, but the control of the workflow: start, pause, stop, resume is still buggy, I implemented the renumbering of gc lines also, because I like the display of lines sent and compare it with lines processed. you will find errors of the gcode a bit faster

[Test_2017.02.041.zip](https://github.com/cncjs/cncjs/files/752445/Test_2017.02.041.zip)

amx1, 2017-02-04:
@cheton 

quick and dirty solution with line numbers and throttling for fast jobs
[Test_20170204_V1.zip](https://github.com/cncjs/cncjs/files/752545/Test_20170204_V1.zip)

amx1, 2017-02-05:
@cheton 

thank you for fixing the imperial coordinates for 3D view, it is ok now. There is still a difference in the axes widget. The machine position is ok, the work position is to small by the conversion factor  

[image]

cheton, 2017-02-05:
I changed a bit to your work, most were some name changes and refining if-else statements, you can checkout the master branch and do tests again. BTW, the work position was corrected in this commit 2bac497ecce3f187537efc11538eef0286862f7f.

amx1, 2017-02-05:
I did a quick check. It looks really good now. I didn't find any critical issues yet. Thank you very much! We should thank also @aldenhart  and @giseburt for their quick responses.

amx1, 2017-02-06:
@cheton 

the "pulldown" options of the TinyG widget are really cool, what do think of implementing this also for the G-code widget to show the gcodes, because in some error cases it might be helpful to see the actual gcode line.

cheton, 2017-02-06:
How about adding a button that opens a new window to show the gcode?

amx1, 2017-02-06:
Dear cheton,

 

yes, perfect, this is the best idea, because it will be needed only for debugging.

 

BTW: it runs perfect now. I am thinking about a soft reset sequence, because in case of an gcode error, it is quite inconvinient to find the right sequence of commands to stop the workflow, sender and clear the controller.

amx1, 2017-02-07:
@cheton 

I checked the actual master branch, with the following errors

/home/am/workspace/git/cnc-master/showdown.js
   1:1   warning  Unexpected var, use let or const instead       no-var
   2:1   warning  Unexpected var, use let or const instead       no-var
   3:1   warning  Unexpected var, use let or const instead       no-var
   4:1   warning  Unexpected var, use let or const instead       no-var
   5:1   warning  Unexpected var, use let or const instead       no-var
  81:70  error    Trailing spaces not allowed                    no-trailing-spaces
  88:28  error    Calls to require() should use string literals  import/no-dynamic-require

cheton, 2017-02-08:
I applied https://cnc.js.org yesterday and integrated showdown into the post build process to generate a html page under /docs. It's fixed in the master branch.

closed 2017-02-28 by cheton