cncjs/cncjs #107: Laser widget

tritao, 2016-12-09:
So we've been testing cnc with our laser cutter and it seems to work!

It would be nice to have a laser widget, similar to the one already provided for spindle, with laser intensity control, and a way to fire on/off the laser using the "fire" commands provided by Smoothie.

cheton, 2016-12-10:
http://smoothieware.org/laser#toc4

Laser Intensity Control:
- <kbd>M221 Snnn</kbd> globally scales the laser power provided by G0/G1 by nnn percent. So M221 S75 will scale the laser power to 75%.

Laser Test:
- <kbd>fire nnn</kbd> where nnn is 0-100 percentage of power (example fire 10 will turn on laser at 10%)
- <kbd>fire off</kbd> turn off the test fire and return to automatic mode.

cheton, 2017-03-09:
Added in 1.9.0-alpha.5

[image]

closed 2017-03-09 by cheton

briankb, 2017-05-06:
I'm trying to connect a PWM controlled diode laser to a TinyG, what pins are controlled with the Laser widget controls? I have searched the repo and forums but can't figure it out. I see the reference to smoothieboard laser mode but I'm not sure how that relates to wiring up the TinyG board.

cheton, 2017-05-07:
Ooops. I forgot adding support for feed, rapid, and spindle override commands in the TinyG controller. 
```js
// https://github.com/cncjs/cncjs/blob/v1.9.4/src/app/controllers/TinyG/TinyGController.js#L934

'feedOverride': () => {
    // Not supported
},
'spindleOverride': () => {
    // Not supported
},
'rapidOverride': () => {
    // Not supported
}
```

This will be added in the next release. Before it's implemented, you can follow below steps to control feed, rapid, and spindle (for laser) overrides.

1. Type <b>$sys</b> from the console widget to show [all system settings](https://github.com/synthetos/g2/wiki/Text-Mode#displaying-settings-and-groups).
https://github.com/synthetos/g2/wiki/Text-Mode#displaying-settings-and-groups

    ```
    > $sys
    {
        "r": {
            "sys": {
                "fb": 100.19,
                "fbs": "100.19-17-g129b",
                "fbc": "settings_othermill.h",
                "fv": 0.99,
                "hp": 3,
                "hv": 0,
                "id": "0084-7bd6-29c6-7bd",
                "jt": 0.75,
                "ct": 0.01,
                "sl": 0,
                "lim": 1,
                "saf": 1,
                "m48e": 1,
                "mfoe": 0, // manual feed override enable
                "mfo": 1, // manual feedrate override: 1.00 [from 0.05 (5%) to 2.00 (200%)]
                "mtoe": 0, // manual traverse over enable
                "mto": 1, // manual traverse override: 1.00 [from 0.05 (5%) to 2.00 (200%)]
                "mt": 2,
                "spep": 1,
                "spdp": 0,
                "spph": 1,
                "spdw": 1.5,
                "ssoe": 0, // spindle speed override enable
                "sso": 1, // spindle speed override: 1.000 [from 0.05 (5%) to 2.00 (200%)]
                "cofp": 1,
                "comp": 1,
                "coph": 1,
                "tv": 1,
                "ej": 1,
                "jv": 4,
                "qv": 1,
                "sv": 1,
                "si": 250,
                "gpl": 0,
                "gun": 1,
                "gco": 2,
                "gpa": 2,
                "gdi": 0
            }
        },
        "f": [1, 0, 5]
    }
    ```

2. Enable spindle speed override `{ssoe:1}` if it's not enabled
    ```
    > {ssoe:1}
    {"r":{"ssoe":1},"f":[1,0,10]}
    ```

3. Change spindle speed override setting using`{sso:n}`
    **5%**
    ```
    > {sso:0.05}
    {"r":{"sso":0.05},"f":[1,0,12]}
    ```
    **100%**
    ```
    > {sso:1.00}
    {"r":{"sso":1},"f":[1,0,12]}
    ```
    **200%**
    ```
    > {sso:2.00}
    {"r":{"sso":2},"f":[1,0,12]}
    ```

Please let me know if it works for you. Thank you.

briankb, 2017-05-07:
I tried that and now the laser intensity when running gcode is 0%, the test still works but intensity when connected to the SPIN pin doesn't change laser intensity. I tried {ssoe:0} (once changed to json mode in tinyg) to disable but I must have done something else to the system.

briankb, 2017-05-07:
Verified with a digital probe (saleae logic) that their is no difference between a Laser Test pulse at 100%, 50% and 0% when using the Laser widget, at least on the SPIN output on a TinyG v8 board. Using 1000 ms Laser test duration. The frequency of the pulse was the same for each. I'm NOT an expert on using a logic analyzer or oscope or any of this really so take my testing as is.

cheton, 2017-05-08:
Hi @ril3y

I'm trying to test TinyG v8 board with firmware version v0.97. Connecting to the SPIN pin does not change laser intensity. I also tried connecting to the PWM pin, but the voltage is still the same when switching to a different spindle speed (e.g. `M3 S1000` -> `M3 S2000`). Do you know what's the right way for controlling laser intensity? Thanks.

SirGeekALot, 2017-05-08:
I do not have a laser hooked up to my TinyG v8, but I do have it on my Openbuilds Ox CNC router with a variable speed spindle. However, I have a spindle speed controller connected, and IIRC, it takes the signal from the TinyG and send it to the spindle speed controller (which outputs the appropriate PWM duty cycle). I'll check later when I'm home to tell you which pins go to the spindle speed controller.