cncjs/cncjs #111: Engraving files are not loaded when greater than 10Mb

paulusjacobus, 2017-01-01:
I am using cnc for laser engraving via an arduino uno with grbl. I noticed that cnc does not load bigger files than 10Mb. Is there a way to increase the size?

mayhem2408, 2017-01-02:
What platform are you running? I haven't gone as large as 10Mb, but 5-6Mb files are very slow with my raspberry Pi struggling to keep up the data flow. Smaller files have no problem.

paulusjacobus, 2017-01-02:
Hi @mayhem2408 i use a windows 10 machine and i can not upload beyond 10 Mb files. They are big because of 10 bits pixel and 256 gray shades. I wonder if this is something that can be configured? Thanks

cheton, 2017-01-04:
Hi @paulusjacobus,

Could you attach a zip file containing your 10MB gcode examples? That will be useful for troubleshooting.

paulusjacobus, 2017-01-04:
Hi @cheton  will do tonight when I get home. They are very simple rows of G1 Xn.n Yn.n Sn commands.

paulusjacobus, 2017-01-04:
Hi I have attached a g code example file that does not load

cheton, 2017-01-04:
You can send the file to my email: cheton@gmail.com

cheton, 2017-01-07:
Hi @paulusjacobus,

The maximum upload file size is limited by the body-parser middleware:
https://github.com/cheton/cnc/blob/master/src/app/config/settings.base.js#L54
```js
// https://github.com/expressjs/body-parser
'body-parser': {
    'json': {
        // maximum request body size. (default: <100kb>)
        limit: '10mb'
    },
    'urlencoded': {
        extended: true,
        // maximum request body size. (default: <100kb>)
        limit: '10mb'
    }
}
```

I will increase the upload file limit from 10mb to 100mb in the next version. That should resolve your issue.

BTW, I'm doing performance enhancements for loading large G-code files. This can significantly reduce the total loading time from 40+ seconds to 5~8 seconds using your 10MB example.

paulusjacobus, 2017-01-07:
> settings.base.js#L54

Thanks Cheton that's great news. Looking forward to the improved size
handling of files.

Thanks Paul

cheton, 2017-01-09:
Fixed in v1.8.13

closed 2017-01-09 by cheton