cncjs/cncjs #112: Error codes

MoonshineSG, 2017-01-02:
Starting with version 1 grbl does not send descriptions for error messages and settings. It would be useful if there is some sort of "translation" done by cnc.js... 
All the codes descriptions are available at https://github.com/gnea/grbl-Mega/tree/edge/doc/csv

cheton, 2017-01-12:
Added error messages in 1.8.14

[image]

closed 2017-01-12 by cheton

MoonshineSG, 2017-01-13:
Yes, it works as shown above. I see that ALARM and Settings, not yet implemented, but will be as well?

Would maybe be nicer if the error is displayed in a more "human" way (not that we don't read json) ?

Smth like:
```
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
> M6
Error 9: G-code lock
```

or even

```
Grbl 1.1e ['$' for help]
[MSG:'$H'|'$X' to unlock]
> M6
G-code lock (error 9)
```

Maybe, a mouse over long description ("G-code commands are locked out during alarm or jog state.") which would benefit those starting to explore cnc ?

cheton, 2017-01-13:
Detailed descriptions for ALARM and $$ will be implemented in a later version.

In my example above, I displayed additional error message in a well-formed json format `{"err":{...}}` that's because I want to differentiate translated result from Grbl's error message, especially showing the line number that caused an error while sending a g-code file.

[image]

Use a human readable format is fine with me for your suggestions, I will try to find a better format to display errors.