cncjs/cncjs #114: simulation mode

brychanrobot, 2017-01-03:
Would it be possible to have a mode that allows a simulated CNC machine? This would mostly be useful for trying the interface, especially for loading g-code, which currently isn't possible without a machine connected. It could possibly be done by writing a grbl controller simulator in node which communicates over a websocket, and then allowing the user to select a simulator websocket rather than a serial port. I had some success with github.com/grbl/grbl-sim, but that one isn't very full-fledged or cross-platform. It would come up as a valid machine, and you could change settings, but as soon as you try to run gcode it would fail. Plus, I could really only get it to work in Linux and it's tricky because it spoofs a serial port. I'd be interested in helping with development if you think this would be worthwhile. The simpler option is to just have a switch in the settings that enables a simulated mode, which would enable G-Code uploading and exploring the interface.

mayhem2408, 2017-01-03:
I simply have a second Arduino Uno flashed with GRBL 1.1e with nothing connected to it and plugged into my USB port. I can test any sender software with it without having to use my real CNC machine, A simple, cheap and effective simulator.

cheton, 2017-01-09:
Hi @brychanrobot,

Just like what @mayhem2408 said, having an Arduino is a simple and effective way rather than simulation. I had tried grbl-sim, but it usually became unstable when running a gcode file, so I gave up the way of doing simulation.

I'd be really happy if you're interested in developing a simple Grbl simulator that can be communicated over a serial connection, that will benefit novice users if they don't have an Arduino to explore the interface.

mayhem2408, 2017-01-09:
@brychanrobot, @cheton and even better, it is OS independent. I use this method to test Windows and Linux streaming software and I know that any problems I find will be with the sender or GRBL itself and is as close to a real world simulation as possible. I don't have to worry about a third variable in the testing equation. And once I've finished testing and simulating, I just plug my Arduino into my CNC shield and the simulation become real world without any modification. You can pickup a cheap Uno or Nano on ebay for just a few dollars. I bought 3x Nanos for $8 last year. Never can have to many Arduinos lying around.

cheton, 2017-01-11:
I will close this issue now since there are features of necessarily higher priorities over simulation.

closed 2017-01-11 by cheton

AustinSaintAubin, 2017-01-13:
@brychanrobot you can order a low cost [Generic Arduino board](http://www.banggood.com/ATmega328P-Nano-V3-Controller-Board-Compatible-Arduino-p-940937.html) for testing (_just for testing, I would not trust one of these boards to control my CNC router, but it works well to test sending commands to_).

arsdragonfly, 2019-08-23:
See [this grbl simulator](https://github.com/grbl/grbl-sim)

MitchBradley, 2019-08-23:
grbl-sim does not work very well.  It tries to simulate the Arduino UART at too low a level which causes timing problems.  Several of us have tried it and given up.  I considered modifying the grbl-sim code to make it work better, but it turned out not to be worth my effort compared to spending $3 on an Arduino Nano for testing.