cncjs/cncjs #115: GRBL-Mega V1.1 and Buffer Size

mayhem2408, 2017-01-09:
@cheton I run running a Arduino Mega and GRBL-Mega V1.1. There are a number of reason of this, but I have noticed that the larger serial buffer is not being fully utilised in CNC.js. The Standard GRBL has 15-16 Planner Buffers, and 127 Serial Buffers. My GRBL-Mega has 64 Planner buffers and 255 Serial buffers. The planner buffer is being fully utilised, but the serial buffer is still only using up to 127. I'm guessing CNC.js is hardwired for a max of 127. Is this something you can increase or detect dynamically and connection, or have an option to select the buffer size?

cheton, 2017-01-09:
Thank you for the information. I'll add a configurable option to change the default buffer size.

mayhem2408, 2017-01-09:
@cheton In the mean time I manually changed the src/app/lib/gcode-sender.js and the src/app/controllers/Grbl/GrblController.js to reflect the larger buffer it is worked fine. Ther serial buffer is now hovering around 30, instead of 160. Having that option though would be great. 
You could have CNC auto detect the buffer if GRBL1.1 is set to report the buffer, BF:63,255 in my case. You could automatically set the buffer to by using the BF: value.

cheton, 2017-01-09:
Right now the buffer size is limited to 120 as shown below. I subtract 7 from 127 to ensure it won't overflow Grbl's receive buffer since because are two commands (`?` and `$G\n`) sending to Grbl regularly.
```js
this.sender = new Sender(STREAMING_PROTOCOL_CHAR_COUNTING, {
    // Grbl has a 127 character serial receive buffer.
    // Use a lower value to deduct the length of regular commands:
    // - parser state command ($G\n)
    // - current status command: (?)
    bufferSize: 120
});
```

It's doable to automatically detect BF report and enlarge the buffer size on the fly. For example:
```js
bufferSize = Math.max(BF[1] - n, bufferSize);
// where BF[1] is the RX buffer, n is the amount that needs to subtract from the RX buffer
```

cheton, 2017-01-12:
Added auto detection in 1.8.14. Please let me know if it works with grbl-Mega v1.1.

closed 2017-01-12 by cheton

mayhem2408, 2017-01-12:
@cheton I've tested 1.8.14 with both my Nano and Mega2560 and the buffers are worked as expected. Great work.

cheton, 2017-01-13:
I found I didn't count the length of new line character (\n) which will consume Grbl's RX buffer space. I will fix that bug later.