cncjs/cncjs #118: Manual tool change for GRBL

mandrav, 2017-01-12:
Hi,

any plans to add "support" for M6 command (tool change) ?
It could be as "easy" as replacing M6 line when encountered with a call to a user-specified macro (which would move to a pre-defined location to allow for the manual tool change and then probe to reset the z-height before resuming).

To be honest, I 've been spoiled by this functionality which is offered by bCNC :-)
But having moved my control to a RPi, it is no longer convenient for me to use bCNC and I 've started using your software more (great work btw!).

Anyway, thanks for taking this into consideration.

cheton, 2017-01-14:
#### Tool Change in bCNC
https://github.com/vlachoudis/bCNC/wiki/Tool-Change

Hi @mandrav,

Do you have an example of "tool change" macro as described in bCNC's wiki page? Thanks.

mandrav, 2017-01-14:
Hi @cheton ,

the following macro assumes that there's a touch plate (or other means of Z-probing) at [X-300 Y-300 Z-50] in machine coordinates (G53).

here's the program that would replace the M6 command to GRBL:

`; [1]: keep a backup of current state ($G) as well as current XY`
`M5 ; stop spindle`
`G90 ; absolute coordinates`
`G53 G0 Z-1 ; raise the spindle up to (almost) the top of the travel`
`G53 G0 X-300 Y-300 ; go to tool change position`
`M0 ; pause the program to allow the manual tool change`
`G53 G0 Z-40 ; move above the touch plate`
`G91 G38.... ; probe command`
`; [2]: now calculate the offset and adjust WCS`
`G53 G0 Z-1 ; raise spindle up again to safe height`
`M0 ; optional extra pause (e.g. to remove probe wires if required)`
`; [3]: reset state ($G) to what was saved on [1] (also restart spindle etc)`
`G0 X[saved] Y[saved] ; [4]: move to XY that contained M6`
`... continue with program`

Keep in mind that the above are NOT a macro in bCNC but instead provided by the program at runtime. The reason is, as you understand, that there's a couple of places that need some simple calculations (as well as the saving/restoring of the current state).
The way it works is like this:

1. Zero out your workpiece (Z=0 on top of it)
2. Calibrate the tool: it's a similar macro which just does the probing part of the above so it knows the probe offset in respect to your workpiece Z=0
3. Run the job
4. On each tool change, run the above code and at point [2] adjust the WCS Z=0 by adding to Z the difference of the current probe location (with the new tool) minus the previous tool's probe location.
5. Continue the job

Having written all that, I understand it maybe doesn't quite match the philosophy of your "program player" and would need quite a few modifications.
That's why I proposed to just run a user-defined macro when M6 is encountered. It will require some knowledge about the program by the user but, at least, will allow for using M6 in your code without GRBL throwing an error that it's not supported.
Such a user macro could look like this (for use with a touch plate that you put on your workpiece to zero it out):

`M5 ; stop spindle`
`G53 G0 Z-1 ; raise spindle to the top`
`G53 G0 Xx Yy ; go where it's convenient to change the tool (machine coordinates!)`
`M0 ; pause for the tool change`
`; also put touch plate on the workpiece, on X0Y0`
`G0 X0 Y0 ; go to workpiece X0Y0`
`G0 Zprobe ; go to Z=touch_plate_height+10 give room for height diff. of new tool)`
`G91 G38 ... ; run the probe command as you like`
`G10 L20 P1 Zplate... ; adjust Z=0 by adding the touch plate's height`
`G91 G0 Z5 ; move up`
`G90 ; absolute coordinates`
`M0 ; pause for cleanup by the user (remove touch plate, wires, etc)`
`M3 ; restart spindle and go on`

Let me know if you think its feasible.

cheton, 2017-01-14:
Thank you for the detailed explanation. Do you see any problems for the steps below?

1. Pause sending g-code to the controller when M6 is encountered
2. Save current modal state ($G) as well as machine position
3. Run a user-defined macro for tool changing
    - Grbl will enter <b>Hold</b> state on M0, click <kbd>Play</kbd> or <kbd>Cycle Start</kbd> button to resume
4. Restore previously saved modal state and machine position
5. Continue sending g-code

---

One more question here: Do you expect to manually choose a user-defined macro when M6 is encountered? This will display a pop-up window before <b>Step 3</b>.

mandrav, 2017-01-14:
Hi @cheton ,

yes, I think that would work fine :-)
As for choosing the macro, either ask to select one from a list of macros or make it a setting (like "Tool change macro").

Finally, it'd be nice if a message was shown so the user knows it's time for a tool change and doesn't wonder why the program stopped ;-)

cheton, 2017-01-14:
I raised a question in https://github.com/gnea/grbl/issues/91. Tool number is also useful for automatically choosing Tool Change macros.

mandrav, 2017-01-14:
Nice catch!

xpix, 2017-04-07:
If this eimplemented now and can i get a example macro for a Toolchange?

AustinSaintAubin, 2017-04-07:
It is not released yet. Still part of the #139 1.9 Release Backlog.

cheton, 2017-05-02:
v1.9.4 enables support for evaluating expression after the % symbol (similar to bCNC). You can manually replace M6 command with below expressions. A Manual Tool Change UI will be added in a later release.
```
%x0=posx,y0=posy,z0=posz  ; Keep a backup of current XYZ
   :   :   :
   <Your manual tool change commands>
   :   :   :
G0 X[x0] Y[y0]  ; Move to XY that contained M6
```

mandrav, 2017-05-03:
Some great news!
Will try it out and let you know.

AustinSaintAubin, 2017-05-05:
@mandrav Please post your macro once you have had a chance to test it out. Have you had a change to test?

mandrav, 2017-05-05:
@AustinSaintAubin no, I haven't had the time to test this yet but, according to @cheton , from what I understand you have to manually replace the M6 commands (no support for auto-replacing with macros or otherwise).

cheton, 2017-05-05:
The macro that I am going to use will be similar to bCNC's manual tool change code. You can check out CNC.py as shown below:
https://github.com/vlachoudis/bCNC/blob/master/CNC.py
[image]

neilferreri, 2017-08-03:
Any updates on this?
@mandrav Do you have a macro to share?

neilferreri, 2017-08-14:
@cheton I've tried a macro to do a tool change and it comes close, but I always get errors.  Sometimes the errors are for invalid statements, and sometimes the machine goes idle and freezes in the middle of a job.  It seems that anytime I put a `%wait` in the gcode, it causes problems.  The same simple macro worked without issue in BCNC (not using the built in tool change).  I just replaced my M6 commands with:
`%wait
G53 Z-15
M5
G53 X-220 Y-220

M0
G91
G38.2 F50 Z-50
G10 L20 P1 Z1.2	

M0
G90`

I'm not sure what else to share with you to help.
Thanks...I'd love to switch to your GUI on all of our machines.

cheton, 2017-08-14:
@neilferreri 

The `%wait` is just a dwell command that waits for the planner queue to empty. It will be replaced with `G4 P0.5 (%wait)` as shown below:
https://github.com/cncjs/cncjs/blob/v1.9.7/src/app/controllers/Grbl/GrblController.js#L219-L230
```js
dataFilter: (line, context) => {
    if (line === WAIT) {
        const { sent, received } = this.sender.state;
        log.debug(`Wait for the planner queue to empty: line=${sent + 1}, sent=${sent}, received=${received}`);

        this.sender.hold();

        return `G4 P0.5 (${WAIT})`; // dwell
    }

    return this.dataFilter(line, context);
}
```

Could you try below program and check if it works for you? 
```
G4 P0.5
G53 Z-15
M5
G53 X-220 Y-220

M0
G91
G38.2 F50 Z-50
G10 L20 P1 Z1.2

M0
G90
```

neilferreri, 2017-08-15:
@cheton 
I'm still getting all kinds of errors.  I get to the probe with no problem, the probe works fine, but it throws errors coming out of the probe.  Sometimes immediately, sometimes a few lines down.  Have you tried a tool change macro?

neilferreri, 2017-08-15:
@cheton After reading what you discovered in issue #186 , I reverted to 1.8.17 and the tool change macro worked with no errors.

cheton, 2017-08-15:
@neilferreri 
Thank you for your findings.

neilferreri, 2017-08-24:
@cheton The find / replace macro works with the new 1.9.8 version.  Seems that the issues were the same as those in issue #186.  Thank you.

cheton, 2017-08-25:
Release Notes of CNCjs 1.9.8
https://github.com/cncjs/cncjs/releases/tag/v1.9.8

cheton, 2017-10-01:
The preliminary support for pausing G-code execution with M0, M1, M2, M6, and M30 commands is now available in the master branch. You can run any macros or jog during a pause. I will make a thorough test in the next few days.

cheton, 2017-10-07:
#### Remaining Todo Items
* [ ] Restore work position after completion
   (1) Move to Z-safe height defined by user
   (2) Move to the previous position along the X and Y axis
   (3) Move to the previous position along the Z axis

* [ ] Restore modal state after completion

    Modal Group | Member Words
    :----- | :-----
    Coordinate System Select | G54, G55, G56, G57, G58, G59
    Plane Select | G17, G18, G19
    Distance Mode | G90, G91
    Feed Rate Mode | G93, G94
    Units Mode | G20, G21
    Spindle State | M3, M4, M5
    Coolant State | M7, M8, M9

* [ ] Update M6 Tool Change instructions on wiki

cheton, 2017-10-13:
https://github.com/cncjs/cncjs/releases/tag/v1.9.11

CNCjs 1.9.11 is now available with M6 tool change support. You can upgrade to 1.9.11 and read the wiki page at https://github.com/cncjs/cncjs/wiki/Tool-Change to learn how to create a tool change macro.

closed 2017-10-13 by cheton