cncjs/cncjs #122: A-Axis - Navigation Buttons

ichbinscheisse, 2017-01-15:
Hi cheton,
I really appreciate having an offline alternative for controlling my g2core/arduino due setup.

What I'm missing are some navigation buttons for the A-axis (+keyboard shortcuts and a 'zero out' option would be nice.)
Are there any plans for this?

cheton, 2017-01-15:
You can click on the [image] icon to open <b>Axes Settings</b>, and enable DRO for the A-axis.

[image]

ichbinscheisse, 2017-01-15:
Hi,
that I already discovered. 
But I was looking for A+ and A- buttons, the same as with x, y and z.

Another problem with the already existing A-Axis display is that it doesn't update its position when changing the Axis via command line.
When I enter G1 A22 for example, the display will still show 0.000mm instead of 22mm/degrees.

cheton, 2017-01-15:
There is not enough room to add <kbd>A+</kbd> and <kbd>A-</kbd> buttons within the Axes widget, I may do something like this to save more space:

[image]

For your DRO problem, are you running TinyG on an Arduino Due? It seems I forgot to turn on <b>posa</b> and <b>mpoa</b> for status reports ([source code](https://github.com/cheton/cnc/blob/master/src/app/controllers/TinyG2/TinyG2Controller.js#L307)). I will fix it in the 1.9.0 release.

ichbinscheisse, 2017-01-15:
That little + and - helps alot.
Personally I don't use the diagonal arrows much, so there might be some room for the A-Axis there.
But that's probably just me.

Waiting for the next release.
Thank you :)

cheton, 2017-01-16:
The movement for the A-axis should use units of <b>degrees</b> (°), neither <b>mm</b> nor <b>inches</b>. I need to correct the display units as well.

amx1, 2017-02-01:
what do you think about including the A axis as a row (A+, A, A-) next to the Z row? The buttons predefined pos, work pos and machine pos could be smaller from point of view

cheton, 2017-02-01:
That's a good suggestion. I can try to use 3 different icons (incl. tooltips) or dropdown menu to render predefined pos, work pos, and machine pos, that will save a lot of space to include navigation buttons like A+, A, A-.

cheton, 2017-08-25:
Release Notes of CNCjs 1.9.8
https://github.com/cncjs/cncjs/releases/tag/v1.9.8

closed 2017-08-25 by cheton