cncjs/cncjs #123: Job duration clears out after job done

mandrav, 2017-01-22:
Hi,

I ve noticed that the "job duration" field in GCode widget is cleared after the job is done.
It would be nice if it displayed the job duration until a new job starts or a new gcode file is loaded.

Keep up the great work.

f0m3, 2017-01-28:
Yes. Thats a thing i noticed too. BTW... estimated duration would be cool too :)

cheton, 2017-01-28:
The duration estimation is one of the missing features from the very beginning. May I know if there are some algorithms or sample code that can be referenced for estimating duration? It will be great if it can also compute the acceleration and deceleration for better precision.

f0m3, 2017-01-28:
I know that universal gcode sender does it. Not extremely precise but ok... 
Unfortunately i am new to git and java... i didn't find the function in the sources... only this:
[https://github.com/winder/Universal-G-Code-Sender/search?utf8=%E2%9C%93&q=estimatedTimeRemaining](https://github.com/winder/Universal-G-Code-Sender/search?utf8=%E2%9C%93&q=estimatedTimeRemaining) Maybe i can learn this way how to find such a function :-)

cheton, 2017-01-29:
Looks like UGS has commented out  GrblSimulator and only makes a crude estimation by calculating timePerCode in current versions. I can add similar thing in v1.9, and try to fine tune the estimation function in later versions.

[image]
[image]

f0m3, 2017-01-29:
I think thats a quite slim way to estimate the remaining time when you dont want to parse the whole gcode file and calculate the duration for each Line from way and speed.

cheton, 2017-02-08:
Fixed in 1.9.0-alpha

closed 2017-02-08 by cheton