cncjs/cncjs #124: tool path not repected

htaxil, 2017-01-23:
Hi Cheton,

I encountered a problem with cnc 1.8.10 : The tool path is not respected.
I generate the gcode with fusion 360, check with CAMotics and everything seems correct. When I import the gcode into cnc, the path drawn corresponds to the desired job, but when the job runs, the tool takes another path...
[Photo 1](https://drive.google.com/file/d/0B9EHDiX9bmlKTnF4cHRMUzB1YzA/view?usp=sharing)
[Photo 2](https://drive.google.com/file/d/0B9EHDiX9bmlKakViZmdDQ180ZDQ/view?usp=sharin)

The CNC firmware is :
GRBL_VERSION "1.1e"
GRBL_VERSION_BUILD "20161219"

cheton, 2017-01-23:
Ooops. I saw a lot of errors reported on the console widget. Could you attach your g-code file and paste Grbl settings ($$)?

Not sure if your problem is resulted from buffer overrun, you can take a trial run with cnc 1.8.15 before starting a job, it fixed a potential buffer overrun issue since 1.8.3 (https://github.com/cheton/cnc/releases/tag/v1.8.15).

cheton, 2017-02-19:
Hi @htaxil,

May I know if there have been any updates regarding this issue?

See #133

andrewismoody, 2017-02-21:
Based on issue #133 I would venture a guess that this is hardware related - either interference or faulty cabling/components. I connected my Arduino directly (no USB hub) to my Raspberry Pi with a shielded ferrite-core USB cable (Tripp-Lite U023-006) and all of my similar issues went away.
I also had flashed my Arduino according to the links listed here:
https://github.com/grbl/grbl/wiki/Known-Bugs
I think some of the issues were the Atmel serial bus on the Arduino itself.

cheton, 2017-02-21:
Hi @htaxil,

I'm going to close this issue. You can re-open it again if you have other findings.

closed 2017-02-21 by cheton

htaxil, 2017-02-21:
Hi cheton, than's for your answere and sorry for my late response.
Every thing work well after upgrading to grbl-1.1e.20170114 and cnc 1.8.15.
However I have to disable the soft limits because they generate an alarm every time work starts (i think it's a grbl issue).