cncjs/cncjs #125: Suggestion: Name of the software

f0m3, 2017-01-24:
Hi,
first: I love this software! Its just what i wanted to have and it changed my entire life :1st_place_medal: 
But: I fear that the name "cnc" is the problem why i didn't find it earlier. Its not what one might suggest to be the name of a software to drive a cnc machine. Its like when some asks in a forum: "What gcode sending software should i use?" And somebody answeres: "You should use cnc". The asker might think the one that answers didnt understand the question :)
just my 2 cents. as i said: i love it!

AustinSaintAubin, 2017-01-25:
I agree, the name CNCjs which was the name originally is much better for search and reference. It is hard to refence the CNC node js project. I tell people to search cncjs and they are redirected to this 'Cnc' repo from the old CNCjs repo. I think the name needs to go back to being cncjs.

cheton, 2017-01-25:
Thank you for reporting this. Yes, this has troubled me and most of you for a long time. The original purpose of renaming the project from cncjs to cnc was to keep a consistent naming for the executable binary ([bin/cnc](https://github.com/cheton/cnc/blob/master/bin/cnc)) and project name, but it's hard to tell others whether it refers to the cncjs software or cnc mills.

I'd consider changing it back to use cncjs in v2.0 for both project name and navigation bar, but still keep the executable name as `bin/cnc`. How do you think about it?

AustinSaintAubin, 2017-01-25:
I recommend changing the executable / bin to `cncjs` as well. It keeps with the naming convention.
I think that changing the name and repo to `cncjs` with 2.0 would be a good idea.

I know how you feel. But I think this would be a wise move long term.

Smuk3D, 2017-01-25:
> thread [2].
> Links:

Hi,

i had found this project also by accident :-D

i had build a small CNC for PCBs. I used bCNC but i want a WebSolution. 
The GUI from bCNC is ugly. And i found much stuff and freaky links and 
one GitHub Autoleveling Z axis Issue .. And over this i found this 
Project.

I think if you want much more user it is better to have an other name as 
cnc .. cncjs is a good option i think :-)

greetings from germany

andrewismoody, 2017-02-22:
I would like to propose that we have a mini contest to suggest some more marketable names.  I would like to start promoting this software on forums, as I think there are a lot of people our there like me who are looking for this solution but can't seem to locate it.  Maybe give it a catchy name to spark interest.
My suggestion (with an obligatory tag line, of course):
Vive CNC - breathe life into your CNC projects

Also, a very simple one-liner to describe the software would be useful:
Full-cycle, autonomous, headless CNC control

cheton, 2018-02-03:
Thanks @MitchBradley, we finally came up with the following schema:

* The name for the computer running Grbl or whatever is the "CNC controller"
* The Node.js-based software is "the CNCjs server". Its filename is "cncjs-server". In prose documentation, it should be referred to initially as "the CNCjs server program", then shortened to either "the server" or "cncjs-server".
* The computer that runs the server is "the server computer".
* The user-facing set of controls is initially "the CNCjs user interface", then shortened to "the CNCjs UI" or "the UI".
* The computer (laptop, desktop, tablet, or smartphone) that displays the UI is "the screen device".
* The Electron bundle is initially "the CNCjs app", shortened to "the app". Its filename is "cncjs-app". The problem with using the word "desktop" is that laptops are much more common these days than desktops. The word "app" is consistent with common usage for modern software that can be accessed either from a browser or a dedicated app. For example you can access Facebook from a browser via a URL, or you can download the Facebook app which takes you directly there.

Later on we will provide a terminology page on the wiki to explain those terms and definitions.

closed 2018-02-03 by cheton