cncjs/cncjs #126: Stream webcam

f0m3, 2017-01-28:
Is there a simple way to network stream the webcam? Like when i have my macbook attached to my GRBL and point the cam to my cnc-router ic could go to other rooms and watch my machine on my smartphone?

cheton, 2017-02-05:
The webcam widget only supports M-JPEG stream, you can use [ffmpeg](https://ffmpeg.org/) to stream audio and video over network. If you need RTSP support, you can see #46 or check out the FAQ below:
* [Webcam Streaming with Raspberry Pi](https://github.com/cncjs/cncjs/wiki/FAQ#webcam-streaming-with-raspberry-pi)
* [Restream RTSP to M-JPEG](https://github.com/cncjs/cncjs/wiki/FAQ#restream-rtsp-to-m-jpeg)

cheton, 2017-02-13:
I'm closing this issue since it's not a bug and no active discussions here. You can continue your discussions here or re-open a closed issue if you found something needs for improvements.

closed 2017-02-13 by cheton