cncjs/cncjs #127: Firmware auto detection

cheton, 2017-02-05:
So far Grbl, Smoothieware, and TinyG (incl. g2core) are supported by cncjs, it's time to provide an auto detection mechanism from the server side so users don't need to explicitly choose the firmware from the web interface.

I'm also considering removing the connection widget and put it on the top navigation bar, that will save more space for the UI display.

liqdfire, 2017-02-05:
If you remove the choice of the firmware, are you going to auto control the display of the Grbl or TinyG widgets?

cheton, 2017-02-06:
The choices of firmware are still there, I will add an <kbd>Auto</kbd> option for determining firmware automatically. It's also doable to control the display of widgets according to the selected or detected firmware.

neilferreri, 2017-08-14:
@cheton I'm not sure if this is where I should post this, but I noticed that on a machine with custom OEM grbl firmware cncjs 1.9.7 does not perform correctly.  On the same machine, I was able to use 1.8.something with no issues.