cncjs/cncjs #129: Connection problems

Hill180, 2017-02-09:
connection problems

Hi 

I try to move my homemade CNC from a serial based interfase til USB and Audrino

I have connected a Audrino with GRBL 1.1 loaded onto. Then i have started cncjs and connected the audrino board. I get the status "sucess! all checksums match!"

cool i think. Some functions is gryed out and i can see on the wiki it perhaps i due to connection problems. 

I have tryed to send manuel G codes. they get send but no responce.. 

I have used the windows x64 installer. and have not made any other setup.

Hope some one can help in the right direction thanks 

br
Steen

cheton, 2017-02-09:
May you check if you're able to see Grbl initialization message after connecting to Arduino. You can check the message from Arduino's serial monitor or from the console widget in cncjs. For example:

```
Grbl 1.1e ['$' for help]
```

Arduino
[image]

cncjs
[image]

Hill180, 2017-02-09:
Problem solved, after reinstalling GRBL on Arduino  and the Arduino IDE it worked for some reason. :-)

Cool 

Thanks for the response !!

closed 2017-02-10 by cheton