cncjs/cncjs #130: Recommend Changing GUI Feed & Speed Override Order

AustinSaintAubin, 2017-02-10:
I recommend reordering the Feed & Speed Override order.
(for the people who don't like to play [where's waldo](https://www.google.com/search?q=where%27s+waldo&source=lnms&tbm=isch), look at "F" & "S" overrides "1%", "10%")

From: 
[image]

To:
[image]

This groups the increase & decrease together, putting the lowest denominations next to each other.

cheton, 2017-02-10:
Sure. I will change it by today.

cheton, 2017-02-11:
I use a smaller icon size for 1% and -1%, that will be easy to differentiate between 1% and 10% at first glance.

[image]

cheton, 2017-02-13:
Fixed in 1.9.0-alpha.1

closed 2017-02-13 by cheton