cncjs/cncjs #132: File Management

andrewismoody, 2017-02-17:
First off - this is absolutely THE BEST software for headless hobby-grade CNC control.  I spent the better part of two weeks scouring the web for something like this and happened on it by accident (this is covered in the naming issue, I think).
Only thing I would really like to see is a file management widget.  Would love to be able to do full-cycle job management inside the GUI.  The Upload GCode button is great and viewing the watch dir is also very good, but if we could just tie those two together, it would close the loop for me.
I was really looking for a way to make 'Repetier-Server' work for my CNC the way it does for my 3D printer.  It would be nice to have "one tool to rule them all", but this project comes dang close to filling that need.
I might poke around a bit in the code and see if I can help contribute this feature.  I have zero experience with nodejs, but I have dabbled in JavaScript, HTML, and such, so I might be able to get a prototype going at the very least.

cheton, 2017-02-17:
Either some screenshots or UI prototypes will help a lot for getting the whole picture. Your contributions are welcome. 😄

andrewismoody, 2017-02-17:
In the simplest form: when you upload, just store that gcode file in the watch dir and provide a delete button for each file listed.  I don't think it would require a separate widget after my first pass at the code.  It would be just two extra functions in the api.  When you call the 'upload file' function, call the newly created 'store file' which writes the collected bytes to a file in the watch dir with the same name.  Then, a simple button added to the watch dir pop-out for each file that calls a newly created 'delete file' api function.  I may be over simplifying, but it seems very feasible in my mind with the limited exposure to the technology that I have.
Basically, the goal would be to remove the need to use another tool for file management (currently connecting with WinSCP to copy new gcode files over).

Here is a very crude UI mock-up:
[image]

Here is a screenshot from Repetier-Server.  Not suggesting that we rip them off completely, but this works really well in that app.  Maybe a stretch goal to wrap it all into it's own widget.
[image]

Nandox7, 2018-03-22:
+1 to this.
I use similar tools to control/manage 3D printers and is one of the most usefull features.
As you can redo old jobs or just use it for multiple consecutive runs.

bradanlane, 2019-04-08:
I too have interest in a lightweight integrated file management.

As a short term solution, I installed BrowsePy on the Raspberry Pi which is running CNCjs. I have BrowsePy pointing to the watch directory for CNCjs. This keeps everything browser based.

_I have no affiliation with BrowsePy. I just wanted to share one option for an interim solution._

dieterrosch, 2020-01-02:
+1 for this.

DanielYWoo, 2021-01-04:
+1
octoPrint did this too, very convenient

MitchBradley, 2021-01-04:
Given the fact that there has been little CNCjs development recently, I recommend using some workaround like the BrowsePy that was mentioned in an earlier comment.

laetic, 2021-10-08:
Setting up BrowsePy and [the watchDirectory as suggested here](https://github.com/cncjs/cncjs-shopfloor-tablet/issues/14#issuecomment-488099732) took me longer than I'd like to admit, but it works great!

I'm using pm2 to launch browsePy along with cnc.js and you can even use the custom widget to just point to your browsepy address and it shows up on the main cncjs page.

[image]

Pastitas, 2023-02-14:
> Setting up BrowsePy and [the watchDirectory as suggested here](https://github.com/cncjs/cncjs-shopfloor-tablet/issues/14#issuecomment-488099732) took me longer than I'd like to admit, but it works [...]

How did you set up the custom widget to get a url? mine won't take it