cncjs/cncjs #133: Buffer Overflow / Unexpected Responses

andrewismoody, 2017-02-17:
I'm not 100% sure on this, but I've been troubleshooting an issue with my gcode files being sent to GRBL incorrectly.  I get weird responses back about missing words and incorrect number fomats, which leads me to believe the buffer is overflowing.
I poked around quite a bit today trying to understand what's going on, and the best I can tell, the issue is with the periodic status queries that are sent on a timer outside of the sender queue, written directly to the serial port.  I'm looking at GrblContoller.js, line 319 and 324.  I don't think, the way it's written, that these commands are being counted toward the buffer count.  I think it would be better to inject them into the sender queue and allow the buffer counter to include those bytes to prevent buffer overflow.
But I might not be reading it correctly.

cheton, 2017-02-18:
```js
// https://github.com/cncjs/cncjs/blob/master/src/app/controllers/Grbl/GrblController.js#L112
this.sender = new Sender(SP_TYPE_CHAR_COUNTING, {
    // Consider periodic commands ('$G\n', '?') to make sure the buffer doesn't overflow
    bufferSize: (128 - 4) // The default buffer size is 128 bytes
});
```

Grbl has 128 bytes of buffer space by default. I deducted 4 bytes from the buffer size to exclude periodic commands (i.e. len('$G\n') + len('?') = 4), but I'm afraid that may not be safe enough to prevent from buffer overflow.

Could you provide me the following information? Thanks.
1. Which version of cncjs you're currently using
2. Your Grbl version ($I) and settings ($$)
3. Attach some of your g-code file files that can even sometimes reproduce the buffer overrun issue
4. Try change the buffer size lower than 124 and test again. That said, a buffer size of 120 bytes should be safe enough. You have to find actual file location in this way:

    ```
    $ dirname `which cncjs`
    /Users/cheton/.nvm/versions/node/v6.9.4/bin
    $ cd /Users/cheton/.nvm/versions/node/v6.9.4/bin
    $ cd ../lib/node_modules/cncjs/dist/cnc/app/
    $ vim index.js
    // Find the line of "bufferSize: 128 - 4"
    ```

andrewismoody, 2017-02-18:
I flashed to Grbl 1.1f today.  I was running 0.9i, which I thought might have been the culprit, but the problem persisted with the latest Grbl build.
I'm running the latest stable release of cncjs, 1.8.16.
I'll try to change the buffer size, but I think this will be overridden by the status query itself, which resets the buffer size to match what is reported by grbl, right? (GrblController.js line 156)  This can be changed as well, I suppose, but I think this is just working around the problem.
Also, I've seen these timer commands bunch up at the end of the job rather than running interspersed.  I think that maybe the fake threading model of javascript (does this carry over to node?) might be thwarting the injection attempts.  This is part of the reason I recommended injecting them periodically into the sender queue, as it would prevent any sort of race conditions.  If I'm understanding the situation correctly, the timer function isn't being executed specifically when expected, and multiple calls are stacking up, then flooding the serial port.
I was worried about trying to fork the build in order to perform troubleshooting, but taking your hint of editing the files directly on my Raspberry Pi, I'll try to hammer out the details and report back with my findings.
I think this log output will support this hypothesis:
[image]

cheton, 2017-02-18:
Yes, I forgot the line you mentioned (GrblController.js line 156), you have to update the line as well while editing the index.js file. That was purposed to support GRBL-Mega, which has buffer space of 255 bytes, see https://github.com/cncjs/cncjs/issues/115.

I do agree with your suggestion, the sender queue, feeder queue, and periodic commands should be combined into one using priority queues to prevent from race conditions. I may try to refactor this part in a near future.

cheton, 2017-02-18:
I guessed it's probably caused by dynamically changing the buffer size during gcode sending:

```js
// Detect the buffer size if Grbl is set to report the rx buffer (#115)
if (res && res.buf && res.buf.rx) {
    const rx = Number(res.buf.rx) || 0;
    // Deduct the length of periodic commands ('$G\n', '?') to prevent from buffer overrun
    const bufferSize = (rx - 8);
    if (bufferSize > this.sender.sp.bufferSize) {
        this.sender.sp.bufferSize = bufferSize;
    }
}
```

It should be corrected as below.
```js
//  Update buffer size only when the workflow state is idle
if (this.workflow.state === WORKFLOW_STATE_IDLE && res && res.buf && res.buf.rx) {
    :   :   :
}
```

cheton, 2017-02-18:
I will provide a hot fix update later today for both 1.8.x release and 1.9.x pre-release.

cheton, 2017-02-18:
@andrewismoody 

You can check 1.8.17 for new updates.

andrewismoody, 2017-02-18:
I'm installing v1.8.17 now to test.  I still think there is some benefit to removing the timers on the status update commands and embed them into the queue logic.  I have some ideas on how to accomplish this, I'll try to do some testing on my local copy and see if I can get a handle on it.  My experience with JavaScript makes me very leery of timers in general.  JavaScript has a very unpredictable threading model.
My initial thought is to define an array of status update commands in the sender.  Then each controller can populate that array appropriately.  When the next() function kicks off to fill the buffer, we place those status update commands at the front of the buffer.
Some pseudo code here (basically just copied the logic from the while loop), going to attempt these changes on my local copy.  Will need some additional constructor and class definition code as well as adjustment to GrblController to inject the Status Commands into the array.  I think this is worth a shot to remove the timer logic, anyway.  I'll let you know how my testing goes.
Maybe we can add a timestamp variable (lastStatusRequest) in the class definition to track when the last status was requested.  Then check that timestamp in the next() function just before the call to process() and if the elapsed time is greater than or equal to our predetermined status interval, we inject the status commands.

var elapsedStatusDelta = now - lastStatusRequest;
if (elapsedStatusDelta >= allowedStatusDelta)
{
    for (let statusCommand of statusCommands) {
        if ((statusCommand.length > 0) && ((sp.dataLength + statusCommand.length) >= sp.bufferSize)) {
            break;
        }
        sp.dataLength += statusCommand.length;
        sp.queue.push(statusCommand.length);
        this.emit('data', statusCommand);
    }
    lastStatusRequest = now;
}

I may be way off on this, so I'll let you know if I have any success.

andrewismoody, 2017-02-18:
The 1.8.17 patch didn't seem to have any impact on the issue.  I may have misdiagnosed it.  I don't see any status commands being written to the log during the gcode send, but I'm still seeing lots of errors reported by Grbl.  When I stopped the job, the flood of status commands is now being written to the log file.  The tool position is not being updated in the console during the job, so that leads me to believe that the status commands aren't actually being queued until the job ends.  Screenshots below - first is paused job, second is stopped job.  I'll keep poking around and see if I can make any discoveries.
I really love this project, you have some amazing features here.  Hoping to get this going!
[image]
[image]

andrewismoody, 2017-02-18:
I'm going to switch to the v1.9.0-alpha.3 release, as it has a lot of good fixes in it and will make testing much easier.

cheton, 2017-02-18:
Would you mind to paste your gcode file and Grbl settings? That will help identify the issue from my side.

andrewismoody, 2017-02-18:
here is my current Grbl settings.  I can't post the gcode I'm testing, I'll try make a similar one that isn't proprietary and see if I can reproduce the issue.
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=7 (dir port invert mask:00000111)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=15 (this returns everything)
$11=0.010 (this makes it corner more slowly for accuracy)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=7 (homing dir invert mask:00000111)
$24=25.000 (homing feed, mm/min)
$25=250.000 (homing seek, mm/min)
$26=30 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=94.474 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=250.000 (x accel, mm/sec^2)
$121=250.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=290.000 (y max travel, mm)
$132=100.000 (z max travel, mm)

andrewismoody, 2017-02-18:
Ok, thanks for being patient with me.  I think there's some other issue going on here that has actually made it look like the status loop was the culprit, but in reality, it seems to be an issue with the serial port communication.  I'm running 1.9.0-alpha.3 now.  I'm watching the status updates in the log just sitting idle and they're running smoothly until I suddenly see an 'error:2' in the console.  At that point, the status updates stop.  I send a ? command from the console widget, it is successful, then the status updates stat flowing again in the logs until another random 'error:2' occurs and stops everything.
I feel kinda silly that I dug so deep into the code and now it looks like something completely different is going on. 
I'm running Raspbian Wheezy 4.4.38-v7+ on a Raspberry Pi 2 attached to an Arduino Uno via USB running Grbl v1.1f.
Node version is v4.0.0
I saw a 'warning' during install of cncjs about building the serial port component due to not finding the right version, but didn't think anything of it originally.
I did have some trouble getting node installed, now that I recall, as well.  Couldn't find an apt-get package of v4 for Wheezy, so I installed it manually from here: https://nodejs.org/dist/v4.0.0/node-v4.0.0-linux-armv7l.tar.gz
Maybe I just need to bump up to Jessie and try again.
I'm sorry for wasting your time on this.  I'll do some more troubleshooting on the serial port and see if I can get it going.

andrewismoody, 2017-02-18:
I bumped node up to v6.9.5, removed cncjs, and reinstalled alpha.3.  I got a similar serial port warning about building 4.0.7 since there was no supported binary for 6.9.5.  Lots of other warnings during that build as well.
Afterward, everything pretty much acted identically to before.  So, I did some more searching and I found this link:
https://github.com/grbl/grbl/wiki/Known-Bugs

It seems there is a 'known issue' with Arduino firmware and USB to Serial communication.  I don't remember having this issue with Universal GCode Sender, but maybe it was working around it.

I'll try to flash the Arduino sometime later on and see if that fixes it.

andrewismoody, 2017-02-18:
I ran this same job via Universal GCode Sender and I don't see all of the error:2 and error:25 issues, but I do see occasional error:1 responses (I don't think I looked this closely before).  I think this confirms that at least part of this issue is related to my Arduino Atmel firmware version.

cheton, 2017-02-19:
Could you try upgrading serialport to "5.0.0-beta3"?
https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/tag/5.0.0-beta3

If you can make the build in your environment, just replace the line in package.json and re-install again:
```
"dependencies": {
    "serialport": "5.0.0-beta3"
}
```

Install dependencies and create a production build:
```
$ node -v
6.9.4
$ npm install -g npm  # Make sure your npm version is up-to-date
$ npm -v
4.2.0
$ git clone git@github.com:cncjs/cncjs.git
$ cd cncjs
$ npm install
$ npm run prepublish
$ ./bin/cnc -vv
```

cheton, 2017-02-19:
> 'm watching the status updates in the log just sitting idle and they're running smoothly until I suddenly see an 'error:2' in the console. At that point, the status updates stop. I send a ? command [...]

This is similar to #134 (UI stopped updating in the middle of job). I might need to add a timestamp to check if has been last for long time of not receiving status report, and then send '?' again to trigger updates.

andrewismoody, 2017-02-19:
I believe it is related.  From what I can tell, it's stopping when it receives an error:1 from Grbl.  So it might be an issue in the error receive logic that doesn't kick the timer back off.  Just a hunch, though, I haven't dug into this part of the code.  Turning on verbose logging
I'm going to try to update the serial port version this morning.
Some updates on other fronts, I added a shielded USB cable and it helped quite a bit.  The errors were occurring every 80 lines or so and now only every 800 lines or so.  So, I definitely have some interference issues.
I also flashed my Arduino Atmel firmware to the version listed in the link, but I haven't tested the success of this yet.  I'll report back when I do.
I want to say how helpful this has all been for me.  I would never have known I was having interference issues causing dropped commands if I had not used your software.  I've been using Universal GCode Sender and it sends every command to the console, so I would never see the errors unless I paused the job and scrolled through - which I never did until investigating this issue.  Having only the errors show in the console of cncjs has already proven invaluable to me and I didn't even know I needed this feature.
I was fighting unexpected missed steps and random job failures that I would blame on many other things - I'm sure now that this is the reason.

andrewismoody, 2017-02-19:
I started running a large job to test the success of the firmware flash, but still saw the errors being generated in the console.  The status update failures appear to be triggered by errors received from Grbl, but not every error, and not any particular one.  I wonder how much impact there is to operation if the timer sends status queries during long moves.
It still feels like a buffer issue to me, I'm going to try the suggestion of updating the serial port module to the latest beta and see if it has any impact.  I saw only one error reported until line 7098, when it started failing more.  At line 35826, I presume the command corruption sent the tool head off into the wild blue yonder, because it was jammed up against the right-hand edge of the gantry and got stuck.
I was monitoring 'top' during the run to see if there was any evident memory leakage causing the issue, but the memory usage stayed pretty consistent.

[image]

I couldn't get the console to cooperate so I could copy all of it, so here are some significant sections:
------------------------------------------------------------------------------------
One random failure at line 541, but then clean run until like 7098:
> G43 H3 D3 Z51.8 (line=10)
error:20 (Unsupported command)
> X70.057 Y72.738 (line=541)
error:1 (Expected command letter)
> ?
<Run|MPos:-156.850,-178.700,-92.682|Bf:0,55|F:951|Ov:100,100,100|A:S>
> X61.494 Y22.368 (line=7098)
error:1 (Expected command letter)
-----------------------------------------------------------------
Failures sometimes come in batches:
> X60.564 Y23.141 (line=11067)
error:1 (Expected command letter)
> X60.802 Y23.249 (line=11068)
error:1 (Expected command letter)
> X60.874 Y23.323 (line=11069)
error:2 (Bad number format)
-------------------------------------------------------------
There were a number of errors until the first status update failure, here (line 8711):
> X47.911 Y23.42 (line=8532)
error:2 (Bad number format)
> X45.138 Y19.656 (line=8711)
error:1 (Expected command letter)
> ?
<Run|MPos:-146.100,-201.975,-92.745|Bf:0,31|F:1000|WCO:-198.000,-226.500,-117.001>
> X40.272 Y30.953 (line=9788)
----------------------------------------------------------------------
Toolhead got jammed against the right-hand side:
> X61.645 Y9.477 (line=35824)
error:25 (Invalid gcode ID:25)
> X61.438 Y9.716 (line=35826)
error:2 (Bad number format)
> !

Nothing significant is reported in the output log, the status commands just stop showing up.

cheton, 2017-02-19:
A another way is to test the same Arduino without connecting any cables. That's the most simple way to clarify this issue.

cheton, 2017-02-19:
Could you also try some gcode examples listed here?
https://github.com/cncjs/cncjs/tree/master/examples/gcode

These files are usually used for testing:
* [github.gcode](https://github.com/cncjs/cncjs/raw/master/examples/gcode/github.gcode)
* [tball-pocket-laser.gcode](https://github.com/cncjs/cncjs/raw/master/examples/gcode/tball/tball-pocket-laser.gcode)

andrewismoody, 2017-02-19:
I will try the example gcodes.  I want to level-set a little bit here, though. I've been using this machine for almost 2 years now with Universal GCode Sender.  I've had a few hiccups here and there with missed steps (presumably due to the error:1 issues I saw in the UGS console), but I generally run very large jobs with no major issues.  I have ran this particular job on UGS and completed the cut successfully.  I think whatever serial port issues are occurring here are magnifying the hardware/environmental issues.

andrewismoody, 2017-02-19:
I couldn't run the github gcode because my machine setup is quite different (I work in positive coordinates - against the standard, I know).  I was able to tweak the tball pocket gcode with some simple find and replace and run it, however.  I see the same issues with this gcode.
[image]

cheton, 2017-02-19:
Could you try switching the streaming protocol to its default value (i.e. SEND_RESPONSE_PROTOCOL) by removing all the parameters.

```js
this.sender = new Sender(SP_TYPE_CHAR_COUNTING, {
    // Deduct the length of periodic commands ('$G\n', '?') to prevent from buffer overrun
    bufferSize: (128 - 8) // The default buffer size is 128 bytes
});
```

You can find the code from the production build by searching "_sender2"
```js
 this.sender = new _sender2.default(); // <--- Remove all parameters
```

andrewismoody, 2017-02-19:
I made the suggested change, but I don't know for certain whether it 'stuck'.  It doesn't appear to have made any difference.  I stopped cnc, made the change, then restarted cnc.

andrewismoody, 2017-02-19:
I rebooted the Pi after correcting the error and it didn't seem to make any difference.
It's odd that it seems to occur once or twice at the start of the job, then runs error free for a few thousand lines before it starts generating errors again.
I think the serial port upgrade is due now.  If I can figure it out.

cheton, 2017-02-19:
What's the gcc version of your wheezy?

cheton, 2017-02-19:
gcc/g++ 4.8 is required for compiling addons for Node.js v4 or higher.

https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/649#issuecomment-167933458

andrewismoody, 2017-02-19:
_node@raspberrypi ~ $ apt-cache policy gcc
gcc:
  Installed: 4:4.6.3-8
  Candidate: 4:4.6.3-8
  Version table:
 *** 4:4.6.3-8 0
        500 http://mirrordirector.raspbian.org/raspbian/ wheezy/main armhf Packages
        100 /var/lib/dpkg/status

---------
Doesn't look like I have g++ installed, but doesn't look like 4.8 is available:

------
_node@raspberrypi ~ $ apt-cache policy g++
g++:
  Installed: (none)
  Candidate: 4:4.6.3-8
  Version table:
     4:4.6.3-8 0
        500 http://mirrordirector.raspbian.org/raspbian/ wheezy/main armhf Packages

andrewismoody, 2017-02-19:
I might need to bite the bullet and get Jessie installed.

cheton, 2017-02-19:
You can also refer to the FAQ to install gcc 4.8 on wheezy

https://cnc.js.org/docs/faq/#install-native-addons-with-nodejs-v4

andrewismoody, 2017-02-19:
I don't want all this manual nonsense to interfere with troubleshooting.  I'm burning jessie onto the SD card now.  Will get it all set up and baseline before attempting to compile serial port update.

andrewismoody, 2017-02-19:
I rebuilt the Pi from Jessie Lite and set everything back up.  I had forgotten to document a few configuration settings, so it took a bit longer than expected.  I baselined with 1.9.0-alpha.3 and the behavior is the same (as expected).  Going to attempt the serial port update now.

andrewismoody, 2017-02-19:
I'm just now understanding what you meant by 'not connecting any cables' - don't power the motor drivers and troubleshoot completely in software.  This makes sense.  Still working on installing cncjs with serialport beta.  I don't want to change too many things at once, so after I complete this and do some testing, the next step will be to remove power from Grbl shield so the motors aren't energized during the test.  This will eliminate motor interference as a culprit and if it still fails, will prevent crashing the machine as well.

andrewismoody, 2017-02-19:
I was unable to compile the latest build on my Pi, it errored out after a couple of hours of running the build:
Segmentation fault

npm ERR! Linux 4.4.38-v7+
npm ERR! argv "/usr/bin/nodejs" "/home/pi/.npm-packages/bin/npm" "run" "build"
npm ERR! node v7.5.0
npm ERR! npm  v4.2.0
npm ERR! code ELIFECYCLE
npm ERR! errno 139
npm ERR! cncjs@1.9.0-alpha.3 build: `gulp prod`
npm ERR! Exit status 139

Maybe ran out of memory?
Is there another way to leverage the 5.0.0 serial port without a recompile?

cheton, 2017-02-19:
Try to downgrade your node.js version to 4 or 6, I never tested v7 before.

cheton, 2017-02-20:
Here is my best practices for the installation. I always use [Node Version Manager (nvm)](https://github.com/creationix/nvm) to easily switch between different Node.js versions on my Mac,  Raspberry Pi 2 and 3. It can help you find a proper Node.js version that matches your OS platform, and allows you to use non-root accounts to install global packages, using root account is not recommended and may cause some issues when installing native modules like serialport.

**Installation steps**
https://github.com/cncjs/cncjs#nodejs-installation

Clone the nvm repo, and check out the latest version:
```sh
$ git clone https://github.com/creationix/nvm.git ~/.nvm
$ cd ~/.nvm
$ git checkout `git describe --abbrev=0 --tags`
$ cd ..
$ . ~/.nvm/nvm.sh
$ nvm install 4
```
Add these lines to your ~/.bash_profile, ~/.bashrc, or ~/.profile file to have it automatically sourced upon login:

```sh
export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && . "$NVM_DIR/nvm.sh" # This loads nvm
```

```sh
$ nvm use 4
Now using node v4.5.0 (npm v3.10.6)
$ npm install npm@latest -g
$ npm install -g cncjs@latest
```

cheton, 2017-02-20:
One thing that I need to mention here, I personally recommend NOT to build the source code on a RPi, it will take long long time because its CPU power and disk I/O performance is very poor than laptop and desktop machines. 😞

andrewismoody, 2017-02-20:
Thanks for the tips.  I had already discovered the 'dont install as root' solution - serialport would not install otherwise.  I didn't know about nvm, though, so that's going to be very useful.  Unfortunately, I don't have another linux machine to do the build.  I'll try to downgrade node and give it another try today.

andrewismoody, 2017-02-20:
I followed your instructions to back-level Node to v4.  The compile went much quicker, but doesn't look like SerialPort v5 is backward compatible.  It did appear to compile successfully, it just doesn't seem to work.

---------
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/5.0.0-beta3/serialport-v5.0.0-beta3-node-v46-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@5.0.0-beta3 and node@4.7.3 (node-v46 ABI) (falling back to source compile with node-gyp)
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/read-poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/node_modules/serialport/build'
/home/pi/.nvm/versions/node/v4.7.3/lib

------------------
0|cnc      | 2017-02-20T13:51:49.715Z - error: uncaughtException: _serialport2.default.parsers.readline is not a function date=Mon Feb 20 2017 08:51:49 GMT-0500 (EST), pid=10699, uid=1000, gid=1000, cwd=/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/dist/cnc/app, execPath=/home/pi/.nvm/versions/node/v4.7.3/bin/node, version=v4.7.3, argv=[/home/pi/.nvm/versions/node/v4.7.3/bin/node, /home/pi/.npm-packages/lib/node_modules/pm2/lib/ProcessContainerFork.js, --watch-directory, /home/pi/CAM, -v, -v, -v], rss=62324736, heapTotal=31885292, heapUsed=29492992, loadavg=[0.66943359375, 2.0068359375, 1.75146484375], uptime=62414, trace=[column=50, file=/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/dist/cnc/app/index.js, function=new GrblController, line=7198, method=null, native=false, column=38, file=/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/dist/cnc/app/index.js, function=, line=10991, method=null, native=false, column=13, file=events.js, function=emitTwo, line=87, method=null, native=false, column=7, file=events.js, function=Socket.emit, line=172, method=emit, native=false, column=12, file=/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js, function=null, line=503, method=null, native=false, column=9, file=node.js, function=nextTickCallbackWith0Args, line=436, method=null, native=false, column=13, file=node.js, function=process._tickDomainCallback [as _tickCallback], line=406, method=_tickDomainCallback [as _tickCallback], native=false], stack=[TypeError: _serialport2.default.parsers.readline is not a function,     at new GrblController (/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/dist/cnc/app/index.js:7198:50),     at Socket.<anonymous> (/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/dist/cnc/app/index.js:10991:38),     at emitTwo (events.js:87:13),     at Socket.emit (events.js:172:7),     at /home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/node_modules/socket.io/lib/socket.js:503:12,     at nextTickCallbackWith0Args (node.js:436:9),     at process._tickDomainCallback [as _tickCallback] (node.js:406:13)]

andrewismoody, 2017-02-20:
I'm editing app/index.js directly to try and work around the incompatibility.  I commented line 7199 out, which solved one issue, then it reported that isOpen is no longer a function (line 7446), but a property, so I adjusted the code there to remove the parenthesis.  This got me past the exceptions, but it still doesn't work, and started complaining about ENOENT not being a file.  There seems to have been a significant amount of refactoring in the v5 release.  I don't think I'm going to be able to solve this on my own.

-------
            baudrate: this.options.baudrate //,
            //parser: _serialport2.default.parsers.readline('\n')

-----
        return this.serialport.isOpen;

-----
2017-02-20T14:39:16.695Z - debug: [session-file-store] will retry, error on last attempt: Error: ENOENT: no such file or directory, open 'sessions/B1KS6INkNlB1T_6WC_jlUFgaeHB_4kNJ.json'     at Object.logFn (/home/pi/.nvm/versions/node/v4.7.3/lib/node_modules/cncjs/dist/cnc/app/index.js:5447:41)

cheton, 2017-02-20:
Oops. According to your logs, the installation seems to be ok, but I just noticed that there is a breaking change between serialport 4 and 5, the interface is not compatible. I suggest you just run `npm install -g cncjs@1.9.0-alpha.3` or `sudo npm install -g --unsafe-perm cncjs@1.9.0-alpha.3` instead of creating a new build yourself.

cheton, 2017-02-20:
Set unsafe-perm flag if the root privileges is required.

https://docs.npmjs.com/misc/scripts#user
USER
If npm was invoked with root privileges, then it will change the uid to the user account or uid specified by the user config, which defaults to nobody. Set the unsafe-perm flag to run scripts with root privileges.

andrewismoody, 2017-02-20:
But I need to do this from the git clone to build with serial port v5, right?  Or is there a way to edit the package.json of an already existing installation and change the dependencies?  I'm a little lost now.

andrewismoody, 2017-02-20:
I tried running all of this in a sudo shell and got exactly the same result.

cheton, 2017-02-20:
1. The unsafe-perm flag is only necessary when you're not using NVM and you may have to use the root privilege.
2. You don't need to git clone to build cncjs with serialport, this is only necessary for local development.
3. If you have ever modified the package.json file and want to make it clean, you can run `npm uninstall -g cncjs` to uninstall previously installed package, and then run `npm install -g cncjs@1.9.0-alpha.3` to install cncjs again.

andrewismoody, 2017-02-20:
ok, I've made a giant mess of my Pi again.  I need to wipe it and start over.
What is the best, simplest way to run cncjs with serialport v5 without using nvm and without cloning and compiling?  Or is this possible?

cheton, 2017-02-20:
I believe you can continue using serialport v4 with cncjs@1.9.0-alpha.3. You previous problem seems to be an incompatible gcc version on wheezy and therefore caused unexpected result.

cheton, 2017-02-20:
If your environment allows remote connection, you can setup a account on your Raspberry Pi and drop me a mail. I can then help you setup the environment and troubleshoot what's going wrong.

andrewismoody, 2017-02-20:
I appreciate the offer, but I don't want to impose.  I've got some ferrite cores coming today to add to my USB cables.  Hoping that will help out some.
You may have seen in previous post that I already baselined with serialport v4.0.7 on Jessie and didn't see any major difference.
What is bugging me is that Universal GCode Sender seems to work more reliably with the exact same setup.  I did report earlier that there were occasional errors with UGS, but it's never caused the tool head to crash into the gantry.
I thought the serialport update looked promising, but if it's not compatible, I understand.
I really appreciate your willingness to help and all the time you've spent so far with this.
I'm going to do some rigorous troubleshooting on the USB connections and see if I can't improve things in general.  If I can rule out the connections, it might give us more to go on.
It may be that the serialport node implementation isn't as robust as the Java serial implementation that UGS uses, and is more prone to interference issues.

cheton, 2017-02-21:
Sorry for that I missed reading the post of your baseline test. To make things much easier, I can write a simple streaming script running with node-serialport v4 or v5, similar to Grbl's [python tool](https://github.com/grbl/grbl/tree/master/doc/script/), but implemented in JavaScript. That will be helpful to clarify the issue.

cheton, 2017-02-21:
Hi @AustinSaintAubin,

May I know have you ever seen unexpected Grbl errors while running on a RPi? His machine was running well with UGS, but not work with cncjs.

andrewismoody, 2017-02-21:
Ok, so, I rebuilt my Pi with Jessie-Lite, reinstalled cncjs 1.9.0-alpha.3, connected the Arduino with shielded USB with Ferrite cores (Tripp-Lite U023-006) directly to the Pi (bypassing the USB Hub it was connected to) and so far I've ran 30,000 lines of the same file with absolutely no issues at all.  Not even the 'GUI stops updating' issue.
I think I've said it before, but it needs to be said again - well done on an excellent piece of software.  I never even knew I had interference issues.  This has been a pretty wild ride through the past few days, but I think I can confidently say this issue is closed.  This was all hardware-related issues.

closed 2017-02-21 by andrewismoody

andrewismoody, 2017-02-22:
Just a follow-up confirmation on this - I've ran 4 or 5 long (~100k lines each) dry-run (no stock, spindle off) jobs  with absolutely zero errors.  I'm doing some hardware work on spindle control, but as soon as that's done, I'm going to do some more dry runs with the spindle on to make sure that doesn't cause any additional interference.
I would like to suggest we add these findings to the readme or the wiki or somewhere prominent, as it was a life saver and easily misdiagnosed as a software issue.

andrewismoody, 2017-02-22:
Happy to report that I ran a few dry runs of the same gcode files (over 100k) with the spindle running and no errors in the console.

cheton, 2017-02-23:
Thank you for your feedback. I created a new [Troubleshooting](https://github.com/cncjs/cncjs/wiki/Troubleshooting) page in the wiki. Could you help update the content about your findings? That will help a lot when people tried to troubleshoot whether it's software or hardware related issues.

Wiki page: https://github.com/cncjs/cncjs/wiki/Troubleshooting

andrewismoody, 2017-02-23:
I added a bunch of detail to the troubleshooting page, but couldn't figure out the best way to upload an image.  Going to try pasting one here and reference the URL to see how it works.

Also, I've been thinking about building a cncjs setup script to run through all the dependencies and configuration items to make rebuilding my Pi quicker and easier.  I thought this might be beneficial for other N00bs like me.  I've seen other people write scripts and host them online such that a user can paste in a single command and run the whole setup.  Let me know if you're interested.

[image]

cheton, 2017-02-23:
I did the same way to upload an image to the wiki. Once a new link has generated, just cut the link and paste it to the wiki page, and you don't need to update the original post.

@AustinSaintAubin has made some great work on updating the [Raspberry Pi Setup Guide](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide). Both of you can work collaboratively to help refine the documentation or create some scripts to minimize installation efforts.

AustinSaintAubin, 2017-03-02:
Sorry for the late response.

> Hi @AustinSaintAubin,
> 
> May I know have you ever seen unexpected Grbl errors while running on a RPi? His machine was > > running well with UGS, but no luck with cncjs.
I only had unexpected Grbl errors if I ran a command that CNCjs was not expecting a reply to. Example: '$?' or 'M3 S16000'.

I sujest a fresh Raspbian install and using the [Install Node.js via Package Manager](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide#install-nodejs-via-package-manager) method.

I just did a fresh install yesterday with no issues w/ node v7. I still need to test it with a long cnc carve though.

AustinSaintAubin, 2017-03-08:
No issues with NodeJS v7 & CNCjs.