cncjs/cncjs #134: UI stopped updating in the middle of job

liqdfire, 2017-02-18:
Running 1.8.16, Electron version
Windows 10

Running a 2 part 3d job. The roughing pass completed perfectly, though only a 10 minute job.
Part way through the 2 hour finishing job the ui stopped updating. 

Here is a video, shows the job is still running and the ui not updating: https://www.dropbox.com/s/jodkg7hvcffocwj/2017-02-18%2019.40.50.mp4?dl=0

Here are the two gcode files:
https://www.dropbox.com/s/k82m5ba359gantn/3D%20Roughing%201.gcode?dl=0
https://www.dropbox.com/s/m1ybu9ui1a1lo2a/3D%20Finish%201.gcode?dl=0

cheton, 2017-02-19:
[image]

Looks like your UI stop updating after sending 13,4xx lines. I just ran your "3D Finish 1.gcode" file using 1.8.17, and the UI is still running well. I will test it again with the electron version.

BTW, I noticed a scrolling issue in your G-code widget, I'd recommend you upgrade cncjs to [1.9.0-alpha.3](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-alpha.3). It includes significant improvements for G-code display, it can reduce much CPU usage and improve web performance.

cheton, 2017-02-19:
You can try input the '?' command in the Console widget and check if it can trigger UI updates after stopping.

[image]

cheton, 2017-02-19:
The query status command (?) will be fired again only when the controller received the status report `<Run|WPos: .....` reported by Grbl.

Related source codes:
https://github.com/cncjs/cncjs/blob/v1.9.0-alpha.3/src/app/controllers/Grbl/GrblController.js#L152
https://github.com/cncjs/cncjs/blob/v1.9.0-alpha.3/src/app/controllers/Grbl/GrblController.js#L314

However, I'm not sure for what circumstances the controller may not receive the status report line. A possible way to workaround this would be to add a timestamp to check the last time it received status report, if it has been last for 5 seconds, it will query status report (?) again to trigger updates.

andrewismoody, 2017-02-19:
I just restarted cnc, trailed the logs, opened the serial port and let it sit for a while.  The status updates flowed consistently until I received an error:1 from Grbl, then everything stopped.  I think this may be the same problem.  Something in the receive error is killing the status loop, I would think.  Poking around the code to see if I can identify the problem.
[image]

liqdfire, 2017-02-19:
@cheton  I have to run two more of these jobs today, I will run the first one and try manually sending a ? command. The second I will run in the 1.9.0-Alpha3 release.

liqdfire, 2017-02-19:
It happened on one of the jobs I ran today. Issuing a ? command did not start the refresh again.

I was able to get it to happen while I had the debugger running, though not on a live job . The ui websocket is timing out, and for some reason not re-connecting. I am still getting status messages in from the arduino, and they are being parsed correctly. When the grbl emits the status event, the listeners in the grbl controller are 0.

cheton, 2017-02-20:
Sending a ? command from the Console widget will take effect only when the server just stopped querying status, which means you cannot see request / response messages from the terminal. You can run `cnc -vvv` to see verbose output. I will try to fix this issue in 1.9 using a timestamp to check the last time it sent the ? command.

If your browser's websocket connection was stopped caused by uncertain reasons, you can try refresh the browser to connect to the server again, it won't stop a running job unless you click on the <kbd>Close</kbd> button on the Connection widget.

andrewismoody, 2017-02-21:
Based on issue #133 I would venture a guess that this is hardware related - either interference or faulty cabling/components.  I connected my Arduino directly (no USB hub) to my Raspberry Pi with a shielded ferrite-core USB cable (Tripp-Lite U023-006) and all of my similar issues went away.
I also had flashed my Arduino according to the links listed here:
https://github.com/grbl/grbl/wiki/Known-Bugs
I think some of the issues were the Atmel serial bus on the Arduino itself.

liqdfire, 2017-02-22:
@andrewismoody I am not quite sure I would agree. The symptoms between the two issues are quite different. I am not getting the gcode errors you are receiving. Even when the ui stops updating for me in the middle of a long job, the job still completes perfectly, which would mean the sender and arduino are in sync as far as the buffer goes.

The usb cable I am using has a ferrite core, and I am connected directly into the tablet I use to run cncjs.
[http://en.chuwi.com/product/items/Chuwi-Hi12.html](url)

As a side note, if you are already using an rpi to run your arduino, you might be interested in this: [http://wiki.protoneer.co.nz/Raspberry_Pi_CNC](url) I have version 2.11 of the hat, and started setting up my rpi this weekend.

andrewismoody, 2017-02-22:
I see.  Maybe I should have watched your video before making assumptions.
I'm curious - are you running the browser interface on the same device that is running cncjs?
Have you tried running the interface on another computer and connecting to the device running cncjs?
I'm wondering if it might be resource issues running both components on the same device.
What browser are you using?  I can't tell from the video.  I have had good success with Chrome on Win10.
Just trying to make some troubleshooting suggestions to narrow down the problem.  Feel free to ignore if this isn't helpful.

andrewismoody, 2017-02-22:
@liqdfire I forgot to say thanks for the suggestion on the RasPi hat.  Currently I run my CNC and 3D Printer from the RasPi, so I think I'm going to keep the Arduino separate for now.  Also, it's already set up and I'm trying not fix what isn't broken.  There are plenty of broken things that need attention. :)

cheton, 2017-02-28:
This issue should be resolved in v1.9.0-alpha.4. I will publish a new alpha release to NPM later.

liqdfire, 2017-02-28:
Great, I should be able to give it a run this weekend.

liqdfire, 2017-03-24:
I ended up switching out my windows 10 tablet for an rPi setup. I did install beta-1 and had no issues with the UI not updating.

closed 2017-03-25 by cheton