cncjs/cncjs #136: Feature Request / Inquiry for Perimeter Tracing prior to actual machining.

AZRadiohead, 2017-02-22:
Is it possible to do implement X/Y perimeter trace?

On laser systems they let you 'trace perimeter'.  Assuming starting from front left, the head would move right, back, left, forward to the start position again to show the furthest extents of the cutting area/artwork.  The other thing that my laser software does is lets me step around the perimeter so that it pauses at each corner for alignment.  ie One mode just traces the entire perimeter and another mode steps through the perimeter.  This would be great to see for the CNC and GRBL controlled machines.

cheton, 2017-02-23:
Hi @AZRadiohead,

Thank you for your query about adding support for perimeter tracing. Sorry for that I have no ideas about how it actually works on other software to probe the boundary and how to define the zero point. Will it do X and Y probing using G38.x commands? It will be great if you have figures, videos, or some generated G-code commands for reference. Thanks.

AZRadiohead, 2017-02-23:
Hi @cheton ,  My laser software calls it "Run Job Perimeter".  Basically it just would trace the outside edges of the design or in this case g-code.  I'm looking at a file I loaded in CNCJS right now and under G-code the Dimensions it shows the Min, Max and Dimensions for X, Y, and Z axis.  There wouldn't need to be any probing.  I guess you could do a separate probe, move to a safe Z distance and then trace perimeter as a separate action.  An example for use would be that I have a 1' x 2' piece of wood that I want to cut out a dish of say 10x10 inches.  The wood is clamped down but I want to maximize my use of it without hitting my clamps.  Being able to see the bit trace a box around the area that would be cut when I run the job would let me see how close I was to my clamps and if I needed to move my Zero.

A perimeter trace only uses X/Y axis and would go something like this.   You would move the head to where you want your X/Y starting location, Zero out X/Y then Click 'Trace Perimeter' (or whatever you call it).  The the machine would move the head to the X,Y Min position, then it would move right to X Max, then back to Y Max, then left X Min, and back forward to Y Min, completing a counter clockwise cycle.  This would trace a box around the X/Y plane that would show the extent of the X/Y dimensions.  After you're satisfied with the perimeter you could set your Z-Axis and Zero Out Offsets prior to running the job.

Since the software recognizes the Min, Max, and Dimensions when you load the G-code, I would think think it would just be a matter of implementation.  I would add a safety popup or something just to make sure that users have the Z-Axis at a safe distance above the surface and that the spindle is stopped and not running. 

Thanks for looking into it.

cheton, 2017-02-24:
Thank you very much for your detailed explanation. I will add it to 1.9 release backlog.

AZRadiohead, 2017-02-25:
That would be great. Thank you for putting in the time and effort, it really is appreciated.

cheton, 2017-03-22:
Added preliminary macro variables support in 1.9.0-rc.1. You can use macro variables for perimeter tracing.

Note that current version doesn't omit G0 rapid moves, so it might not be a real boundary to the workpiece. A complete solution will be added in a future release.

#### Supported Macro Variables
* [xmin]
* [xmax]
* [ymin]
* [ymax]
* [zmin]
* [zmax]

#### Example

1. Create a macro with the following content:
    ```
    ; Traverse around the boundary
    G90
    G0 Z10 ; go to z-safe
    G0 X[xmin] Y[ymin]
    G0 X[xmax]
    G0 Y[ymax]
    G0 X[xmin]
    G0 Y[ymin]
    ```

2. Once a G-code file is loaded, run the macro for perimeter tracing with respect to current G-code boundary.
[image]
[image]

closed 2017-03-22 by cheton