cncjs/cncjs #137: Grbl ALARM:3 (Abort during cycle) at end of job (low priority)

andrewismoody, 2017-02-23:
It seems that the Ctrl-X command that is being sent at the end of a Grbl job is occasionally interrupting the final move to clearance plane (maybe once every 5 jobs or so).  The machine is still in motion when the Ctrl-X is sent and Grbl detects this as an ALARM rather than just a standard reset and stops the in-progress motion before the clearance plane is reached.

I primarily notice this on my 'Home and Reposition' macro where I send a home command, then send GCode to reposition the toolhead at the corner of my clamps.  The final move is to bring the toolhead up to the clearance plane in preparation for running the actual job.  It's this final move that gets interrupted and I have to run the macro again because the exact position is lost.

I see this is being execute by line 601 of GrblController.js, but I can't seem to find which piece of code triggers this.  I think it would make sense to ensure that the machine has stopped (no changes in status or 'Idle'?) before calling reset.

Low priority, as it's fairly simple to work around.

cheton, 2017-02-23:
After gcode sending has been finished, the web interface will actively issue a stop command to the server when Grbl just entered the Idle state. If you actively stop a running program, it will hold the movement, make a 50ms delay, and then reset Grbl.

For your case, the delay should be a value of 0, you can console.log to check if it's already in the idle state. If Grbl actually reported Idle state but motor power was still in movement, maybe it is  necessary to add a long delay (e.g. 500~1000ms) to workaround it.

```js
'stop': () => {
    this.event.trigger('gcode:stop');

    this.workflow.stop();

    const activeState = _.get(this.state, 'status.activeState', '');
    let delay = 0;
    if (activeState === GRBL_ACTIVE_STATE_RUN) {
        this.write(socket, '!'); // hold
        delay = 50; // 50ms delay
    }

    setTimeout(() => {
        this.write(socket, '\x18'); // ctrl-x
    }, delay);
}
```

andrewismoody, 2017-02-23:
Are you saying that it calls the stop event on job completion?  I don't see the feed hold being sent to the console at the end of the job, so I didn't think this was the right function.  I couldn't seem to find where the event is triggered on job completion.  I assumed it was 'reset'.

cheton, 2017-02-23:
Actually you won't have chance to see the GRBL_ACTIVE_STATE_RUN state within the stop event handler while operating from UI.

I'm not sure if Grbl may report Idle state but the motor is still running its final steps. If yes, a delay might be necessary before doing a reset.

cheton, 2017-02-23:
Here is the trigger point, a dirty hack for checking finishing state but works.

src/web/widgets/Visualizer/index.jsx
[image]

andrewismoody, 2017-02-23:
Ok, so you are waiting for Grbl Idle state, but it seems that it's still moving even though it reports idle.  Honestly, I'm not sure that this step is even required.  Is there a reason why you're sending the stop at the end of the job?  It could just finish silently, couldn't it?

cheton, 2017-02-24:
The reason for sending the 'stop' event is to stop server's workflow, notify all connected clients the job has finished, and then unlock the UI.

However, I just noticed that each connected browser window will send the 'stop' event to the server, which is an undesired behavior (i.e. sending the same event multiple times). I will remove the code of `controller.command('stop')` from the Visualizer widget, and allows server to stop the workflow by itself.

I'm wondering if it's necessary to make a soft reset (^x) after a job has finished, although it is a recommended usage to make sure subsequent tasks won't be affected by previous task due to  different gcode modes.

Before making any changes, could you just try making a delay (e.g. 50-500ms) before the reset and  checks if this can resolve your problem? Thanks.

andrewismoody, 2017-02-24:
I changed the delay to 500ms and ran about 10 of the Home/Reposition macros in a row with no issues.  I think this fixed the problem.
I agree that 'completing' the job via the GUI is a bit of a hack and should be done as part of the workflow.  Your multiple browser argument makes a good case alone, but I was also thinking if there is no browser connected when the job finishes, would it successfully reset at the next connection?
I think you've got it under control, either way.  Thanks for the help.

andrewismoody, 2017-02-28:
This happened again today (only once since I made the change) even with the 500ms delay.  I noticed that it may be related to network traffic.  I had bumped up my webcam to 640x480 to get more detail.  The failure rate is much less frequent with the increased delay, but I don't think it will ever be solved completely until this is moved to the backend.

cheton, 2017-03-01:
Yes, it will be eventually moved to the backend in the 1.9 release.
I'm considering removing the call of automatically firing a "stop" event when the job finishes, and leaves the action to user. The only disadvantage is that you need to manually click the Stop button from the web interface, but it allows you to perform this action only when you see the machine has completely stopped.

andrewismoody, 2017-03-01:
I really like the automatic workflow stop at end of job.  Maybe you could add a simple machine position delta to make sure it has stopped.  
Some pseudo code below.

//during status update loop
var lastPos = new Array(0, 0, 0);
var posDelta = new Arra(0, 0 ,0);
var curPos = new Array(xPos, yPos, zPos);
var machineIdle = true;

if (lastPos[0] != 0 & lastPos[1] != 0 & lastPos[2] != 0)
{
   posDelta[0] = curPos[0] - lastPos[0];
   posDelta[1] = curPos[1] - lastPos[1];
   posDelta[2] = curPos[2] - lastPos[2];
}
else
{
   lastPos = curPos;
}

if (posDelta[0] != 0 & posDelta[1] != 0 & posDelta[2] != 0)
{
    machineIdle = false;
}

Then add a check for machineIdle in the if statement to decide whether to stop the workflow.  If Grbl is idle and the machine is idle and etc.

Just a few thoughts.

cheton, 2017-04-24:
Fixed in [1.9.3](https://github.com/cncjs/cncjs/releases/tag/v1.9.3). The workflow state is now centralized to the backend server (commit https://github.com/cncjs/cncjs/commit/7737668f9bf3d19e7997f4fae3393c5d8b882757).

closed 2017-04-24 by cheton

cheton, 2017-04-27:
I found a more simple way to do this without checking status report. https://github.com/grbl/grbl/issues/932

Send a dwell command "G4 P0" follows the last command. When the dwell is complete, Grbl will respond with an ok to indicate the machine has stopped movement.