cncjs/cncjs #138: Operating the spindle controls should send a feedhold or warning (low priority)

andrewismoody, 2017-02-23:
Operating the spindle control buttons in the Spindle Widget (M3, M4, M5) should probably send a feedhold.  Not sure on sequencing here.  At the very least, I think, turning the spindle off (M5) should trigger a warning if the job is still in progress.  Or maybe I'm not thinking about this correctly.

Low priority.

cheton, 2017-02-23:
It probably make sense to trigger a warning. I'd like to hear feedbacks on how others think about it :)

AustinSaintAubin, 2017-02-23:
Triggering a feedhold if M5 is triggered would have saved so many bits... and peaces of work. I would say it should hold the feed if Button is pressed or if spindle off is detected. However, for laser work there should be a easy way to disable feed hold and warnings... and have the setting state saved. This might be part of a bigger Laser Mode state/settings.

- Austin St. Aubin (sent from mobile)

cheton, 2017-02-24:
How about disabling these buttons in the Spindle widget when a job has started? it can be enabled again when entering either "Idle" or "Hold" state, which means you have to trigger a <kbd>Feed Hold</kbd> or click on the pause button from the UI.

andrewismoody, 2017-02-24:
Perfect!

closed 2017-02-28 by cheton

cheton, 2017-02-28:
Fixed in v1.9.0-alpha.4. I will publish an alpha release to NPM later.

commit 2f1ae1cb (2017-02-26): Disable spindle control buttons when running a gcode program (resolve #138)