cncjs/cncjs #140: When units are set to INCH the software reverts back to MM upon job completion.

AZRadiohead, 2017-02-24:
When I open the CNCJS software and change the measurement units from MM to INCH then run G-Code, upon job completion the units revert back to MM.  

Also when exporting gcode from Fusion 360 I usually tell F360 to use INCHES but CNCJS always opens as MM.  Is this by design or can it be set to always default to inches?

cheton, 2017-02-25:
Grbl defaults to mm mode (G21). You can program the startup string `$N=G20` to startup with G20 inches mode. It will execute every time you power up and soft-reset.

On job completion, cncjs will send a soft-reset to reset gcode modes, that's why you saw it always reverted to G21.

AustinSaintAubin, 2017-02-25:
Does / could CNCjs auto detect and remember the currently displayed gcodes G20 / G21 presence / known unit of measurement. 

The gcode widget that displays the max and min or XYZ could also display the gcodes G20 / G21 unit of measurement.

AZRadiohead, 2017-02-25:
@cheton Thank yoy, that solved it!  I thought I had set $N0=G20 previously but I think when I did the last GRBL 1.1f update it cleared my settings.  Still learning.  Cheers!

closed 2017-03-09 by cheton

eciramella, 2018-12-19:
Can you explain the exact steps to do this?  I'm kinda confused where to put this info.

MitchBradley, 2018-12-19:
You must interact directly with GRBL via the CNCjs Console widget.  It is on the left side of the screen, displaying white text on a black background.
https://user-images.githubusercontent.com/4861133/31747254-7be3e2a0-b407-11e7-8c1b-fbe51d04c63a.png
Click in that widget and type the line
`$N0=G20`
That will send a command directly to GRBL, which will tell GRBL to start in inch mode.

eciramella, 2018-12-19:
@MitchBradley THANK YOU!  Is this a one time thing, so I can set it once forever or is this every time I run cncjs?

(i.e. is this persisting on the grbl board or somewhere within the cncjs config)

MitchBradley, 2018-12-19:
It persists in non-volatile storage on the GRBL board, so one-time unless you upgrade the GRBL firmware, as another user reported above.

eciramella, 2018-12-29:
Hmm - seems to switch back and forth pretty abruptly even after setting once (I haven't upgraded the firmware in many moons).

Thanks, I'll keep poking, but seems like this something I need to keep an eye on.