cncjs/cncjs #142: Spindle Coolant Unexpected UI Behaviour

AustinSaintAubin, 2017-03-02:
The Coolant Buttons in the Spindle widget do not behave as expected.
I have two separate cooling / chip removal systems.
One is connected to GRBL's Flood Coolant, the other is connected to the GRBL's Mist Coolant.
I am able to enable/disable them independently or have them both on, which is good, I want the ability to have both on/off/or just one on.

[image]

**The issue is that the UI buttons do not reflect the state of the Flood Coolant, and Mist Coolant properly**.
If I enable Flood Coolant, then also enable Mist Coolant, both show on the UI as off.
They Flood Coolant and Mist Coolant should be able to be toggled independently of each other in the UI.

cheton, 2017-03-02:
@AustinSaintAubin 

May you show me the result of your $G command?

I just traced the code at https://github.com/gnea/grbl/blob/master/grbl/report.c#L321, I guess it may report M78 when both mist and flood are on.
```c
 report_util_gcode_modes_M();
  #ifdef ENABLE_M7
    if (gc_state.modal.coolant) { // Note: Multiple coolant states may be active at the same time.
      if (gc_state.modal.coolant & PL_COND_FLAG_COOLANT_MIST) { serial_write('7'); }
      if (gc_state.modal.coolant & PL_COND_FLAG_COOLANT_FLOOD) { serial_write('8'); }
    } else { serial_write('9'); }
  #else
    if (gc_state.modal.coolant) { serial_write('8'); }
    else { serial_write('9'); }
  #endif
```

cheton, 2017-03-02:
Hi @chamnit

I uncommented ENABLE_M7 from config.h and found something strange. The $G parser state reported <b>M78</b> when both M7 (Mist) and M8 (Flood) are turned on, may I check if this is a desired behavior? Thanks.

[image]

chamnit, 2017-03-02:
@cheton : It should say `M7 M8`. Thanks for letting me know. I'll fix it asap.

cheton, 2017-03-02:
It looks great. Thank you for your prompt assistance.

https://github.com/gnea/grbl/commit/849122559b647493f4deda59405f130589fcd1b6

AustinSaintAubin, 2017-03-03:
Awesome!

AustinSaintAubin, 2017-03-03:
CNCjs UI still needs work, if both M7 & M8 are on UI shows only M8 is on.
But output looks good.

```
> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S40
ok
> M7
ok
> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M7 T0 F0 S40]
ok
> M8
ok
> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M7 M8 T0 F0 S40]
ok
> M9
ok
> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S40]
ok
> M8
ok
> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M8 T0 F0 S40]
ok

cheton, 2017-03-09:
Fixed in 1.9.0-alpha.5

closed 2017-03-09 by cheton