cncjs/cncjs #143: ShuttleXpress Problems to configure Buttons

mwcnc, 2017-03-10:
I have Problems to configure the Button 5 - "Switch Jog Distance" and to configure the Jog  in the ShuttleXpress Control Panel under Windows 7. Other Buttons are ok. Can anyone tell me how to program theese keys "=" "[" "]" or can anyone export the settings for cncjs?

Thanks
Matthias

cheton, 2017-03-10:
#### Button 5
[image]

#### Jog Forward
[image]

#### Jog Backward
[image]

mwcnc, 2017-03-10:
Hello Cheton,

thanks for your help. I solved the problem. In German-Keyboard-Layout it is impossible to configure the Buttons because you have to press "Shift + 0" to get the "=" and "AltGR + 8" to get the "[", respective "AltGr + 9" to get "]". Temporary switching to US-Keyboard-Layout was an easy way to configure the Buttons.

closed 2017-03-10 by mwcnc

cheton, 2017-03-10:
Thank you for letting me know the keyboard layout issue. I will try adding customizable keyboard mapping support in a later release.

cheton, 2017-03-16:
Hi @mwcnc,

Now you can try 1.9.0-beta.1, it provides additional keyboard shortcuts for ShuttleXpress.
 
*  Select Jog Distance
   <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>cmd</kbd> + <kbd>d</kbd>
* Jog Forward
    <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>cmd</kbd> + <kbd>f</kbd>
* Jog Backward
    <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>cmd</kbd> + <kbd>b</kbd>
* Select A-axis
    <kbd>ctrl</kbd> + <kbd>alt</kbd> + <kbd>cmd</kbd> + <kbd>a</kbd>

mwcnc, 2017-03-16:
Hi Cheton,

I programmed my ShuttleXpress with the new shortcuts and
tested it with the Version 1.9.0-beta.1 and it works great.

Thank you for your Information and the great program.

Greetings from Germany
Matthias

slimline33, 2020-07-02:
for easy programming use the virtual keyboard inside Windows! Just a hint for the other ones!

commit 207d4bb8 (2017-03-13): Alias keys for Jog Lever Switch, Jog Forward, and Jog Backward (resolves #143)