cncjs/cncjs #146: 1.9.0 Beta.1 Mac  DRO not working

Jasonselinger, 2017-03-17:
Great Software. The DRO is not working at all in the newest build 1.9.0 Beta.1 for Mac OS. I can run programs and the machine responds properly, but the numbers all remain at 0. Using 1.9.0 Beta with no issues. Jut thought you should know.

cheton, 2017-03-18:
I just tried 1.9.0-beta.1 for Mac ([cncjs-1.9.0-beta.1-mac-x64.dmg](https://github.com/cncjs/cncjs/releases/download/v1.9.0-beta.1/cncjs-1.9.0-beta.1-mac-x64.dmg)), but doesn't encounter the issue you mentioned. You can try the following troubleshooting steps:

1. Open the developer tool and check if there are errors shown on the "Console" tab.

    [image]

2. Click <kbd>Reset</kbd> to reset Grbl, and perform some jog operations.
3. If all of above doesn't work, go to <strong>Settings > General</strong>, and then click <kbd>Restore Defaults</kbd> to reset configuration to system defaults.

cheton, 2017-03-22:
You can try 1.9.0-rc.1 to check whether the problem still exists. Thank you.

cheton, 2017-03-26:
Fixed by https://github.com/cncjs/cncjs/commit/7e10f4b5a43d4e9cde7880e7ea8086ac2964866b in [1.9.0-rc.3](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-rc.3)

closed 2017-03-29 by cheton