cncjs/cncjs #147: Z-Probe not working properly

hcomet, 2017-03-18:
I am using grbl 1.1 and the cncjs 1.9.0-beta.1

When using the probe widjet it always executes G10 L20 P1 Z{Touch Plate Thickness} no matter which work coordinate system is active.

Should it not be using G43.1 to set the tool length offset instead? The rough order of execution for the probe should be G49 (reset the current tool offset) then use the G38 probe command, then use G43.1 to set the new tool offset to the current workspace Z position (after probe stop) minus the touch plate offset (assuming a probe going down).

I am assuming that the intent of the probe widjet is to set the tool offset for the current work coordinate system since there is a G49 dropdown in the widjet. This is the way I was use to working with  chilipeppr for grbl. Unfortunately the Z probe for Chilipepr has also been broken in the grbl 1.1 supporting version.

cheton, 2017-03-19:
You can see past discussions at https://github.com/cncjs/cncjs/issues/53. The probe was initially using tool length offsets, but then switched to work coordinate system since G5x coordinates are persistent across power cycles and soft resets, while G43.1 settings are lost after soft reset.

In current version, the probe widget always applies offsets to G54 but not active work coordinate system, it's a kind of hard-coded issue, I will fix it in a later release. Also, the G49 dropdown should be removed as well to avoid confusion.

I recommend you use work coordinate system instead of tool length offsets for persistency. However, if you have reasons that you have to use tool length offsets, I can provide an option that allows you switch between active work coordinate system and tool length offsets, that will fulfill different needs.

hcomet, 2017-03-20:
I have read the discussion and understand the argument for using G10 to set Z0 in the workspace so once you have fixed the  Z-probe to change the current work cordinate system then the Z-probe will be usable.

For me, I am used to using the tool offset because that is how I work and how chilipepr's touchplate widjet works. I have several jigs that mount into my wasteboard and a different work cordinate space defined for each. When I work I home my machine, set the work cordinate system for the jig then rapid move to X0Y0Z0 (using the goto Zero button in Chilipepr). I keep my Z0 at a high position in the different work corodinate systems so the rapid move does not impact anything. Then I use the touch plate widjet to set Z0.  

With grbl, I like to be able to reset the Z0 easily with  grbl reset or power cycle. Grbl now saves the tool offset between jobs as long as there is no reset (M30 use to reset the tool offset but that is fixed).

I think it would be useful to be able to select which method the Z-Probe uses, G10  or G43.1 to provide flexibility to how users work. 

Thanks.

cheton, 2017-03-20:
Sure. I will add a switch control to toggle between G10 and G43.1.

There is one thing that I want to let you know, cncjs will automatically send a soft reset (^x) to Grbl after job completion each time. If this is not desired behavior, I may have to make it be a configurable option on the <b>Settings</b> page.

hcomet, 2017-03-20:
I am used to there being a soft reset becuse I only just upgraded to grbl 1,.1 with the M30 fix. But I think it would be good to have it configurable. I believe grbl fixed the M30 to make it NIST gcode compliant.

BTW: So far I really like cncjs, great job! I will probably switch over from chilipepr once the Z-probe is fixed up.  

Thanks.

cheton, 2017-03-22:
Added in [1.9.0-rc](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-rc).

[image]

Also, please let me know your test result. Thank you :)

closed 2017-03-22 by cheton

hcomet, 2017-03-22:
Hi Cheton,

I tried the change tonight. Thanks for doing it so quickly.

When the Apply the Tool Length Offset for the Z Axis button is on then it seems that both the tool length and the active work corodinate system are set. This results in the Z0 for the tool length being correct but it also modifies the Z0 for the work cordinate system for be the top of the touch plate. I don't believe having the work cordinate system Z0 set to the top of the touch plate is correct.

I had thought that the switch would toggle between the two options. If OFF then the active work cordinate Z0 is set and tool length is unchanged. If ON then the tool length is set and the active work cordinate system Z0 is not changed. 

Also, if you run the Z-Probe twice in a row it sets the tool length incorrectly. I believe you need to use G49 everytime before using G43.1 other wise the tool lenghts seem to get combined to zero.

cheton, 2017-03-23:
Oops! I forgot adding G49 to cancel TLO before running probe. I will add G49 before running G38.x probe command, which might look like so:
```
G49 # Cancel TLO
G91 # Set relative distance mode
G38.2 Z-10 F25 # Probe toward workpieces (maximum travel distance is 10mm)
G90 # Set back to absolute distance mode
G10 L20 P1 Z0 # Zero Out Work Z axis for G54
G43.1 Z-8 # Set TLO to -8mm (= the height of touch plate)
G91 # Set relative distance mode
G0 Z2 # Retract slightly from the touch plate
G90 # Set back to absolute distance mode
```

With respect to the Z0 set to the top of the touch plate, what's your expected command sequences without setting Z0?

cheton, 2017-03-23:
I made a slight modification to the previous one. Is this exactly what you would expect?
```
G49 # Cancel TLO
G91 G38.2 # Z-probe
G90
G43.1 Z[tlo] # [tlo] = <current work Z position> - <touch plate height>
G91 G0 Z2 # Retraction
G90
```

hcomet, 2017-03-23:
Yes, that looks like what I was expecting.

I am a hobbiest so when I use my CNC I may use it different than some more production oriented people. What I mean by that us that each time I use it the Z0 for my parts will be different. But since I use a jig to always hold my parts I want the X0Y0 to always be the same. When I mill something I place the material in the jig go to the jig zero then use Z-Probe. This means I expect the WCS  X0Y0Z0 to always be the same for a specific jig. Then I set the Z0 for that part using tool offset. I don't want the Z0 to change because the next time I cut something the matterial may be thicker and than the previous time. If the Z-probe of the previous piece had reset the WCS then when I go to X0Y0Z0 the bit would slam into the part.

I think if someone was doing more production work with the same piece over and over then they may prefer to have the WCS Z0 modifed by the Z-Probe as you had it before.

There are three Z-Probe options:
1 - Z-Probe sets the active WCS Z0 only
2 - Z-Probe sets the tool offset only
3 - Z-Probe sets both the active WCS Z0 and the tool offset.

Not sure if 3 is needed since it can be achieved by doing a Z-Probe using setting 1 followed by doing a Z-Probe using setting 2.

cheton, 2017-03-24:
Thank you for detailed explanation. I don't think the option 3 is necessary to use, if someone want this option he can create a macro to run Z-Probe for different need.

I'm refactoring the code to add expression parsing support like `Z[posz - 10]`, and display a preview before running Z-probe. A new release will be published in the next few days.

cheton, 2017-03-25:
Fixed in [v1.9.0-rc2](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-rc.2). 

Clicking on the <kbd>Z-Probe</kbd> button will open a modal window, you will see a preview before starting:

[image]

hcomet, 2017-03-26:
Thanks but I couldn't try it out because I had the same issues as reported in [https://github.com/cncjs/cncjs/issues/150](url)

hcomet, 2017-03-26:
Checked it with rc.3 and it works as expected. 

Thank you for modifying this.