cncjs/cncjs #148: Feature Request: Webcam Rotate

nkings10, 2017-03-19:
Hey,

I'm using CNCJS with a MJPG-Streamer and am unable to rotate the webcam feed and the way it has to be mounted is upside down.

It would be great if a simple rotate option to the wiget were added.

Currently i'm using a Chrome plugin for custom css styles where i can force a rotate of the div that holds the img element.

Thanks,

nkings10, 2017-03-19:
[image]

You can see here when i rotate it like this and hover the mouse over the element is actually upside down. But it works fine. Would just be great if the widget had this option.

cheton, 2017-03-19:
Sure. I will add two options that allows you to flip the video vertically as well as horizontally to make correction.

- [ ] Flip the video vertically 
- [ ] Flip the video horizontally

nkings10, 2017-03-19:
Perfect, that will allow for people who may be getting mirrored feeds from cameras too. Covers all basis. Thanks for all your hard work. Do you have a donation link.

cheton, 2017-03-22:
This feature is now available in [1.9.0-rc](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-rc). You can rotate left, rotate right, flip vertically, or flip horizontally.

[image]

If you'd like to donate, there is a donation link at the bottom of the README. Thank you.
https://github.com/cncjs/cncjs#donate

closed 2017-03-22 by cheton