cncjs/cncjs #149: GRBL Console some ideas to improve

xpix, 2017-03-25:
I checked cncjs and i love it, here some ideas to make it better
- sometimes the console doesn't scroll in fullscreenmode
- put the input field on the bottom side or better emulate a console screen w/o extra input textbox
- no history feature in input box, (arrow up, get the latest command)

I used windowsdesktop cncjs 1.9.0-rc.1

cheton, 2017-03-25:
Thank you for your suggestion. That's also what I would like to improve in the near future.

cheton, 2017-05-30:
xterm.js contains full feature set for terminal emulation. I will give it a try someday.
https://github.com/sourcelair/xterm.js/blob/master/demo/main.js

cheton, 2017-07-01:
Terminal history is now supported in 1.9.6

closed 2017-07-01 by cheton