cncjs/cncjs #150: v1.9.0-rc.2 keyboard and move commands unresponsive

AZRadiohead, 2017-03-26:
I updated to CNCJS v1.9.0-rc.2 today and reset the Raspberry Pi, Shapeoko 3 XL, and reloaded the CNCJS webpage.  I then Homed the cnc which worked fine but when I went to use the Move arrows they didn't work.  I tried the keyboard button and the arrow keys didn't work either.  Manual move commands like G90 X10 moved the cnc, but seemed very delayed.  Most of the options in the GRBL/More panel were unresponsive.  I reset everything a couple more times with the same results.  I ended up reverting to CNCJS v1.9.0-rc.1 and everything is back to normal with all moves working properly.  I didn't have time to re install rc.2 to see if it was a bad install but it loaded without errors and the interface looked fine.

hcomet, 2017-03-26:
I have also had the same issues with rc.2. using a Shapeoko 3 from a Mac. Homeing works and commands from the console work but nothing else. Also could not connect when running a remote browser. Used cnc -v --allow-remote-access &> error.txt to get the attached file. 
[error.txt](https://github.com/cncjs/cncjs/files/870498/error.txt)

I also tried cycling the power on my Shapeoke a couple of times to reset but it didn't help.

cheton, 2017-03-26:
@AZRadiohead @hcomet @mayhem2408

I was using Node v6 for development, however, I didn't notice that 1.9.0-rc.x had RegExp errors with Node v4. I will try resolve this issue soon.

Otherwise, did you see any errors reported from browser's developer tool?

cheton, 2017-03-26:
Fixed by https://github.com/cncjs/cncjs/commit/7e10f4b5a43d4e9cde7880e7ea8086ac2964866b in [1.9.0-rc.3](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-rc.3)

**Bad** (w/ Node.js v4 or earlier versions)
```js
const re = new RegExp(/\[[^\]]+\]/, 'g');		
```

**Good** (all Node.js versions)
```js
const re = new RegExp(/\[[^\]]+\]/g);
```

cheton, 2017-03-26:
Please let me know if [1.9.0-rc.3](https://github.com/cncjs/cncjs/releases/tag/v1.9.0-rc.3) works for you.

hcomet, 2017-03-26:
rc.3 is working for me. Node.js 4.8.1 on a Mac with grbl 1.1.

Thanks.

cheton, 2017-03-27:
@AZRadiohead 

May I know your test result with 1.9.0-rc.3? Thank you.

AZRadiohead, 2017-03-28:
@cheton Hi, sorry I lost track of this thread.  I installed 1.9.0-rc.6 this morning before I saw you comment above.  Everything seems back to normal now.
 1.9.0-rc.6
node v4.7.2
Should I update to Node v6 just to be current?

Also, Thank you so much for being on top of things.  I can't tell you how much I appreciate it.
The Perimeter trace function is exactly what I was looking for.

cheton, 2017-03-29:
Node.js v7 is fine to use if you didn't encounter any problems. Thank you for letting me know your test result.

closed 2017-03-29 by cheton