cncjs/cncjs #152: G-code widget not showing gcode commands...

hublerdesignworks, 2017-04-03:
I am using cncjs version 1.9.1 in Windows 10 (desktop) and TinyG2.

In the G-code widget, I cannot see the list of gcode commands at the bottom of the widget.

I've looked at several places on the internet on how to turn on the gcode list. Am I missing something?

Help!

mayhem2408, 2017-04-04:
@hublerdesignworks Just click on the 3d view button to turn off the 3d view and enable the gcode view

[image]

[image]

cheton, 2017-04-04:
@hublerdesignworks 

Just as @mayhem2408 said, you can click on the <kbd>3D View</kbd> button to toggle between 3D view and gcode view. So far the G-code widget doesn't have much to do, I may deprecate this widget and integrate it into the center view in 2.0 or a future version.

closed 2017-04-04 by cheton