cncjs/cncjs #153: Workspace on mobile

fringy138, 2017-04-04:
I am unable to view/use the web interface on mobile since updating to v1.9.1. It works fine on my computer, but on mobile I see "loading" for a second then only a black screen. 

Is anyone else having this problem?

AustinSaintAubin, 2017-04-04:
I have noticed this as well.

cheton, 2017-04-06:
Fixed in commit bf9f67f62cb71230ac5cd70be52a7db29baf2587

Also found several layout issues with mobile view, I'll look at it in the following days.

yfilali, 2017-04-17:
Would it be possible to have a small bug release for this mobile fix? the latest on npm doesn't have the fix above.

cheton, 2017-04-17:
Sorry for the delay. I just got back from a 7 day trip to Okinawa.

I will publish a new release today to resolve the blank screen issue on mobile. However, the mobile layout problem is not easy to resolve, I will try to fix it in a future release.

cheton, 2017-04-17:
The blank screen issue was fixed in v1.9.2.

Note that in this version, it didn't completely fix the UI layout problem on mobile devices, you may have to use an iPhone 6 Plus or a larger iPad display in the landscape mode, or you will encounter problems while scrolling.

[image]

closed 2017-05-02 by cheton

yfilali, 2017-05-03:
Sorry, forgot to reply. Yes, I tested this and it at least works very nicely on iPad now. Thank you very much!