cncjs/cncjs #155: Widgets Not Movable / Reorganise

AustinSaintAubin, 2017-04-25:
v1.9.3 & v1.9.2 
Unable to move / reorganise widgets.

cheton, 2017-04-26:
I set a delay of 300ms click & hold to prevent accidental sorting especially for touch screen scrolling. Should you be able to sort widgets after a 300ms delay?

cheton, 2017-04-26:
How about adding a sorting icon aside the title? It will be much clear this icon is intended for sorting, and it can prevent accidental sorting when scrolling with touch screen.

See below:
[image]

AustinSaintAubin, 2017-04-26:
I like that Idea a lot. Might also be good to have widget deletion confirmation or delay.

The delay seems to be acting a lot longer than 300ms... seems like 1000ms.
Using Chrome Version 57.0.2987.133 (64-bit) on Mac OS

cheton, 2017-04-26:
The delay can be removed after adding the sorting icon. I will also add a confirmation asking user if he would like to remove this widget.

cheton, 2017-05-02:
Fixed in 1.9.4

closed 2017-05-02 by cheton