cncjs/cncjs #156: Axis controls are disabled

bradymadams, 2017-04-26:
I have an open connection to GRBL. cncjs is running on a Raspberry Pi with a Protoneer CNC Shield. I can control it via the console, but the axis controls are disabled.

[image]

cheton, 2017-04-26:
It seems that CNCjs didn't receive the status report from Grbl. In general case, you should be able to see the "Idle" state or something else like below. You can try clicking on the <kbd>Reset</kbd> button on the top-right corner to reset Grbl, or reconnecting serial connection from the Connection widget.

[image]

bradymadams, 2017-04-26:
Clicking Reset enables the controls. I also tried disconnecting and re-connecting the port, but it still has the same problem. I can simply workaround by always clicking Reset after opening the port.

closed 2017-04-26 by bradymadams

cheton, 2017-04-26:
I'd also like to know what caused this issue. Could you provide me your output logs by running `cnc -vvv` from your terminal? Did you see the Grbl welcome message from the Console widget after establishing a serial connection? Thank you!

bradymadams, 2017-04-26:
Here is the log. I started 'cnc -vvv', connected to the port, and it didn't get the status report, so I disconnected, reconnected, it still didn't receive the status, so I then clicked "Reset" and it received the status.

I'm not sure if this would be related, but I had a hard time getting the correct port name. It appeared that the serialport module was returning "ttyAMA0" rather than "/dev/ttyAMA0" and so I was unable to connect. I also did not have success adding "/dev/ttyAMA0" via ~/.cncrc so I ended up manually adding it in index.js inside the CNCEngine.start function. I wonder if this might be the cause of this issue.

[cnc.log.txt](https://github.com/cncjs/cncjs/files/957136/cnc.log.txt)

cheton, 2017-04-26:
There was a similar discussion in https://github.com/cncjs/cncjs/issues/58#issuecomment-233831899. You can use a tool provided by the serialport module to check whether you can connect to Grbl correctly:
```
/home/pi/.npm/lib/node_modules/serialport/bin/serialport-terminal.js -p /dev/ttyAMA0 -b 115200
```

Otherwise, I found that the "ports" settings in `.cncrc` didn't work in current versions. I will get it fixed in 1.9.4.

cheton, 2017-04-26:
If you cannot see Grbl welcome message with serialport-terminal.js, you can try upgrading Grbl to 1.1f.  This major upgrade is not backward compatible, so you have to make a copy of your old settings ($$), and send the "%" wipe command to re-initialize Grbl after first major upgrade.

bradymadams, 2017-04-26:
I tried `serialport-terminal.js` and that is how I determined that this did not work:

`serialport-terminal.js -p ttyAMA0 -b 115200`

That would give me the same connection error that I was seeing in cncjs. However, this would work:

`serialport-terminal.js -p /dev/ttyAMA0 -b 115200`

So I needed cncjs to use `/dev/ttyAMA0`, so I ended up hard coding it. I'll look forward to the fix you referenced for 1.9.4 so I can use the config file to properly set my ports.

I'll look into upgrading Grbl to 1.1f and report back if I do so.

Thanks for the help!

commit 3e1d0d8f (2017-04-26): Fix a bug that cannot load port settings from .cncrc on current versions (resolves #156)