cncjs/cncjs #158: Multiple webcam widgets

cheton, 2017-05-03:
Some users have both the Raspberry Pi camera as well as a USB camera, and they would like to have both running and see multiple video sources on the web interface.

A better solution to this is to provide the ability to fork a widget, and the forked widget will have its own settings without interfering with original widget.

dirtdiver9, 2017-05-03:
Thank you for this!
I have the PI Camera watching the whole machine and a usb endoscope watching the bit directly.  Being able to have both on the screen at once would be great.

cheton, 2017-05-22:
It will be available soon in v1.9.5. You can fork a widget to use different configuration settings.

closed 2017-05-23 by cheton