cncjs/cncjs #159: Higher Serial Baud Rates & GUI Issue

AustinSaintAubin, 2017-05-04:
I would like to be able to increase the Baud Rate for my GRBL controller from 115200 to 250000 [which seems to be supported](https://github.com/grbl/grbl/issues/598).
I hope to be able to reach higher feed rates at a higher baud when laser engraving.
But the UI does not allow me to select a higher baud than 115200.

I would like to be able to select a higher baud rate. Like the range of baud rates that [CleanFlight GUI](https://chrome.google.com/webstore/detail/cleanflight-configurator/enacoimjcgeinfnnnpajinjgmkahmfgb?hl=en) has selectable.
[image]

-------

I also noticed that the baud drop down clashes with the terminal input field.
[image]

cheton, 2017-05-04:
It's fine to add baudrate 250000 to the list. For other baudrates, I may consider adding a "baudrates" list to `.cncrc`, then you can specify additional baudrates (e.g. 2MB) other than defaults. For example:

```json
{
    "ports": [
         {
           "comName": "/dev/ttyAMA0",
           "manufacturer": ""
       }
    ],
    "baudrates": [ 500000, 1000000, 2000000 ]    
}
```

I found a solution to the second issue, it can be fixed using the `menuContainerStyle` prop:
https://github.com/JedWatson/react-select/issues/1076#issuecomment-253629943
```js
<Select ... menuContainerStyle={{ zIndex: 5 }} />
```

AustinSaintAubin, 2017-05-05:
I think your proposed solution of having it so a user can specify additional baud rates in a list to '.cncrc' is a good idea and solves the problem.

cheton, 2017-05-23:
Fixed in v1.9.5

closed 2017-05-23 by cheton