cncjs/cncjs #161: Question: Problem GRBL, Joggmode

holle72, 2017-05-19:
Hello,
i have a China 2417 CNC Mill/USB, Windows 10 64bit. First, sorry for my bad english :)
I like this Programm and i want to use it in the future. But first i am a noob ;) and..
I have a problem with the Joggmode. When i press a Butten (no matter if i press X Y Z, spindle), i can not press once more. Only one press work, then i must press Reset and then again one press work.
But i can give gcode in the console, that works fine. Can you help me?
Thanks

cheton, 2017-05-20:
Are you running the Electron app for Windows x64? Try open the developer tools and check if there are javascript errors appeared in the console.

holle72, 2017-05-21:
Hello,
thank you for the answer. I thing the problem is much to high for me ;) After you answer i installed Electron. There was a window where i can drag the programm on. But many erros apear. When i have more time, i tried it once more and i tell you.
Greetings

AustinSaintAubin, 2017-06-03:
I actually seem to be having a similar issue that started shortly after @holle72 posted (about 2 weeks ago). My issue might or might not be related.

If i jog X,Y, or Z. after about 15 - 90 seconds CNCjs will seem to lockup. The web interface will become unresponsive. I can ping my raspberry pi, and ssh into it. I have to restart CNCjs to  resolve the issue... Ran CNCjs with -vvv, no error, log output is pauses, like the interface just died, but cncjs server is up. Chrome console, no errors...

AustinSaintAubin, 2017-06-03:
Just carved a 30 min job, no issues with the carve. But before & after the carve I had cncjs "lockup" several times.
I jogged to the workspace... had it lockup several time on the way, restarted cncjs, homed, then jogged again. Ran successful carve without any issues, then jogged the gantry and it locked up again within 15 seconds... odd and frustrating.

cheton, 2017-06-03:
@holle72 @AustinSaintAubin 

Did you see different result when comparing v1.9.5 with previous releases (e.g. v1.9.4, v1.9.3, and etc.)? So far I cannot reproduce this issue using my Mac, if you have a recorded video that will be helpful to follow your reproduction steps.

AustinSaintAubin, 2017-06-03:
I loaded v1.9.4 with ```sudo npm install -g cncjs@v1.9.4 --unsafe-perm```, issue remained.
I am going to wipe and reload my raspberry pi... will also test desktop version with my Mac.

AustinSaintAubin, 2017-06-03:
Fresh install of [CNCjs on Raspberry Pi](https://github.com/cncjs/cncjs/wiki/Raspberry-Pi-Setup-Guide). Still having issues while jogging.
Have [posted video of the issue](https://youtu.be/VuargISY7JQ).

dirtdiver9, 2017-06-03:
I've had very similar issues as well.  v1.9.3 seemed better to me, but v1.9.5 has done it a few times since using it.  I'm also on a raspberry pi and have done all the updates when I installed v1.9.5.

AustinSaintAubin, 2017-06-04:
I have checked all the logs I can, and looked throughout the raspberry pi... nothing erroring out. It looks like the CNCjs node web server might be halting.... when the freeze occurs I am unable to even load static content like ( http://xcarve/images/logo-square-256x256.png ). But pings are solid, never dropped, and ssh session is steady.

AustinSaintAubin, 2017-06-04:
Tested v1.8.15, jogged it for 5 minutes without issues.
So issue is between v1.8.15 and v1.9.3...

```
# Setting: Make Backup
mv ~/.cncrc ~/.cncrc_bak

# Remove current CNCjs version
pm2 stop cncjs
pm2 delete cncjs
sudo npm uninstall -g cncjs

# Install and run older version of CNCjs
sudo npm install -g cncjs@v1.8.15 --unsafe-perm
cnc --port 8000 -vvv

# Settings: Restore Backup
rm ~/.cncrc
mv ~/.cncrc_bak ~/.cncrc
```

AustinSaintAubin, 2017-06-04:
Testing v1.9.0 for a solid 5 min. No issues.

```
# Setting: Make Backup
mv ~/.cncrc ~/.cncrc_bak

# Remove current CNCjs version
sudo npm uninstall -g cncjs

# Install and run older version of CNCjs
sudo npm install -g cncjs@v1.9.0 --unsafe-perm
cnc --port 8000 -vvv

# Settings: Restore Backup
rm ~/.cncrc
mv ~/.cncrc_bak ~/.cncrc
```

AustinSaintAubin, 2017-06-04:
v1.9.1 No Issues.
v1.9.2 No Issues.

AustinSaintAubin, 2017-06-04:
Testing v1.9.4 again for a solid 6 min. No issues.

I think I had issues with v1.9.4 in earlier testing a few days ago because I did not uninstall v1.9.5 first. Yesterday & Today I have been uninstalling previous version first before installing the version to be tested. 

```
# Setting: Make Backup
mv ~/.cncrc ~/.cncrc_bak

# Remove current CNCjs version
sudo npm uninstall -g cncjs

# Install and run older version of CNCjs
sudo npm install -g cncjs@v1.9.0 --unsafe-perm
cnc --port 8000 -vvv

# Settings: Restore Backup
rm ~/.cncrc
mv ~/.cncrc_bak ~/.cncrc
```

--------

All this to say that it seems the issue is with something introduced in v1.9.5. Likely something introduced in one of [cncjs's many dependencies](http://npm.anvaka.com/#/view/2d/cncjs).

cheton, 2017-06-05:
@AustinSaintAubin 

Thank you very much for the recorded video. Did it happen only when turning on keypad jogging?

I saw several `Cannot read property 'toLowerCase' of undefined` errors in your video, it might not be the root cause because those errors were caused by a Chrome extension, but you can go to `chrome://extensions` to disable all extensions and try again.

Now I can reproduce similar issue by enabling keypad jogging. My Node.js server will throw an "Abort Trap" error and stop execution after 20~60 seconds of keypad jogging, as shown below:

[image]

A major difference between 1.9.4 and 1.9.5 is that I upgraded socket.io@1.7.3 to socket.io@2.0.1
https://github.com/cncjs/cncjs/blob/v1.9.4/package.json#L218
https://github.com/cncjs/cncjs/blob/v1.9.5/package.json#L220

I will rollback socket.io to 1.7.3 and run more tests.

AustinSaintAubin, 2017-06-05:
Did it happen mainly when keypad jogging. However, also happens when using the UI buttons, or the [PS3 remote pendant](https://github.com/cncjs/cncjs-pendant-ps3).

In later testing a used [chrome incognito mode](https://support.google.com/chrome/answer/95464?co=GENIE.Platform%3DDesktop&hl=en) which I had set to disable all plugins.  But that was after the video was made.

It does seem like a socket issue, so that makes sense.

How did you get your Node.js server to throw an "Abort Trap"?
I could not get any part of my stack to throw any errors...

cheton, 2017-06-06:
I just ran `cnc -vv` directly without using pm2. Everything works fine after rolling back to socket.io to 1.7.3, but it happened again after upgrading socket.io to 2.0.3.

cheton, 2017-06-06:
I will publish a new release containing the bug fix (038d870), it will show a modal to prompt user if the WebSocket connection has stopped working due to an unexpected server error.

cheton, 2017-06-08:
I'm pretty busy with my day job in this moment. I will make new updates by next week. Sorry for the delay.

cheton, 2017-07-01:
Fixed in 1.9.6

closed 2017-07-01 by cheton