cncjs/cncjs #162: Marlin Support

madmod, 2017-05-20:
Maybe this is a dumb question and its already supported. I'm using a Mini-Rambo with Marlin firmware for my MPCNC build, and I'd really like to use some proper CNC software instead of 3D printing tools. Is it currently possible (or planned) for this to support Marlin firmware? If not can someone point me in the right direction for how I can get this working? I'm willing to contribute to the project if there is some specific piece that needs to be implemented.

Thanks!

SteveHogg, 2017-05-23:
Funny, I'm trying to do the exact same thing. I've come across https://github.com/gnea/grbl-Mega to try to get GRBL on the arduino. So far I have CNCJS connected to the board and it looks like it might work, but haven't actually tried it out with the ramps connected so I can't say for sure its all working.

m0n5t3r, 2017-05-25:
Mini-Rambo seems to be equivalent to arduino mega + ramps (it has the atmega chip and the stepper drivers, etc. on the same board), so grbl-mega should work with the appropriate configuration

I, for one, found the idea of using Marlin for MPCNC weird, so I use an arduino uno clone + cnc shield + cncjs running on an Orange Pi Zero, resulting in a fully autonomous wifi-enabled machine :)

cheton, 2017-06-01:
I had a GT2560 board running Marlin 1.1.x. The communication protocol is simple, I think it's not hard to add support for Marlin firmware. I will take a look when I'm free someday.

cheton, 2017-06-28:
The preliminary Marlin support is now available in the [marlin](https://github.com/cncjs/cncjs/tree/marlin) branch. It's just a quick start, and there are some issues need to be resolved.

There is one thing I cannot find from Marlin documentation. If there is a way to query active state (i.e. Idle, Run, and etc.) and modal groups (G20/G21, G90/G91) when communicating with Marlin firmware?

m0n5t3r, 2017-06-28:
doesn't seem so; the most you can get is the current position, endstop status and temperatures, Marlin (and most other 3d printer firmwares, I guess) only implements the minimum necessary for moving things around, then adds various extensions to deal with 3d-printing specific stuff (temperature PID tuning, calibration offsets, various kinematics, etc.); the state is usually handled in the host software.

cheton, 2017-12-06:
Added Marlin support in 1.9.12

closed 2017-12-06 by cheton