cncjs/cncjs #163: Request to be able to probe and find center of tapered bits

dirtdiver9, 2017-05-30:
Would it be possible to have a probing feature, similar to Mach 3, that would allow the machine to position at the center of a bit, and in particular a tapered bit?

Here is a video of the process in action:
https://youtu.be/7W-d5fH9XrQ

The bit is put inside a circle and the probe is started.
It moves the bit in 4 directions and then calculates the center of the distances moved.

AustinSaintAubin, 2017-05-31:
I would look at the probing / touchoff featuers of [sourcerabbit-gcode-sender](https://github.com/nsiatras/sourcerabbit-gcode-sender), [chilipeppr](http://chilipeppr.com/) and [Estlcam](http://www.estlcam.com/)
Both have probing features that could be emulated for development into cncjs.

Requested Features:
- Z Touch Probe [DONE]
- [Hole Center Probing](https://www.youtube.com/watch?v=AwX43JOcHkY)
- [Edge Finding & Skew Compensation Probing](https://www.youtube.com/watch?v=bhZdbgM6S70)
- [Surface Scanning / Z-Azis Warp Compensation](https://www.youtube.com/watch?v=ApBfHW7AhsM)

saxa, 2017-05-31:
In fact this is very useful IMHO.